arendst/Tasmota #444: WS2812 troubles

Ierlandfan, 2017-05-12:
Hi all,

I connected a Ws2812 ledstrip (about 2 meters) with a 1.5A 5v adapter and data to GPIO14.
At first start it only lit up the first 30 leds. Afters etting the dimlevel (?) they all lit up. Even when I set it to 0 or any other level.

When it's on, the first 30 or so led are blinking white very rapidly and the other go into some kind of party mode. Whatever I do (http://192.168.1.30/cm?cmnd=Dimmer%2050) It's not doing what I expected.
It does something but for example, i cannot see that it actually dims, it stays on full brightness.

davidelang, 2017-05-12:
you probably need a much more powerful power supply for that many LEDs
k

jmb, 2017-05-13:
Some helpful info here: https://learn.adafruit.com/adafruit-neopixel-uberguide/power?view=all#estimating-power-requirements

kruebsam, 2017-05-13:
Do have the very same problems as described (using a 5V 10A DC power supply, thus sufficient power to e taken). With an older version (some 4.3 or 4.4) it did work very well with exactly the very same hardware been used. Thought I would be the only one having those problems with the newer versions. Is there someone that does have one of the old versions at hand in order for me to further investigate on the root cause? For me it looks (just from my impression on watching the LEDs flashing in no indentifying order) as if the timing for the WS2812 protocol is totally out of range. BTW I am using an APA104 LED strip.

arendst, 2017-05-13:
Looked into it but do not have any problems with it. I used a 24 led ws2812 ring directly connected to gpio14 and powered by an external 5V power supply.

[image]

Using firmware 5.1.0 with the PROVIDED libraries it works like a charm. Are you sure you are using the Neopixelbus library version as is available in my lib folder?

Sunrise17, 2017-05-16:
Sonoff Basic has some issue with WS2812 modules, there is no light over the strip. Strip has 147 leds and powered by 5v 12a adapter. Same system works well with Arduino as seen in attached photo, but when i move"Data and GND cables" over Arduino to Sonoff Basic's "GND and GPIO14", there is no light over the strip.

[image]

arendst, 2017-05-16:
What happens if you first disable serial logging (``seriallog 0``) and then connect your WS2812 string to GPIO01 instead of GPIO14?

As I do not have a Basic with GPIO14 exposed (one of the earlier types) I can only test on GPIO01 or GPIO03. Both are used for serial communication but I cannot get GPIO03 to work but GPIO01 works for me and my 24 led ring.

Sunrise17, 2017-05-16:
I have disabled serial logging as 0 and then i have connected ws2812 to Sonoff GPIO01(TX) and its GND, but still there is no light on the strip. My strip is ws2812b type.
[image]

arendst, 2017-05-16:
@Sunrise17 are you using the precompiled sonoff firmware or did you compile the code yourself? If you do it yourself did you use the neopixel library from my repository?

Does it help to change pixels to 30, color to 14000  and scheme to 0?

Sunrise17, 2017-05-16:
still there is no response, yes i have used Arduino IDE during uploading firmware, in my respository it has neopixel bus library but which way you can advise me to upload tasmota to Sonoff?

arendst, 2017-05-16:
Well if you have tasmota runnig just open the upload webpage and upload the precompiled version. Setting should stay the same as long as define CFG_HOLDER is equal in both precompiled version and your current version.

Precompiled versions are available in both  the api folder and via the release tab

Sunrise17, 2017-05-16:
I have already updated the Sonoff over OTA (firmware.bin), to  v5.1.0, finally this is the last precompiled version, right?

kruebsam, 2017-05-17:
Gave the binary as found within https://github.com/arendst/Sonoff-Tasmota/blob/master/api/arduino/sonoff.ino.bin a try, but received a 'Upload failed   File magic header does not start with 0xE9' during update process. The one found at https://github.com/arendst/Sonoff-Tasmota/releases/download/v5.1.0/firmware.bin was possible to upload to the Sonoff. 

Bottom-line: got it running again.

At the end, the source for my problems turned out to be that the very first LED-module within the strip had been destroyed by the Power Supply (Mean Well MDR-60-5). Probably it is not OK to have the power supply been connected to the module during power up/down of the power supply. that might be the reason why that power supply comes with a special 'DC OK' output.
For those that like to know which LED-strip had been used: https://de.aliexpress.com/item/4m-DC5V-APA-104-led-pixel-srip-waterproof-in-silicon-tube-60pcs-APA104-LED-M-with/1998535733.html

arendst, 2017-05-18:
@Sunrise17  Digged out a Sonoff RF with GPIO14 available and connected my led ring to it. No problems at all. Works like a charm.

After you connect your strip to sonoff do you power cycle your strip power supply?

Sunrise17, 2017-05-18:
Yes i am using GPIO3, now everything is ok, corners are soldered in my ws2812 led strip, may be cause of that issue, some part of ws2812 is losing its responsiveness. I will check it again. In my app, i am changing colors on rainbowcycle with producing HEX color code to Sonoff. Of course it is publishing many messages during the movement of color picker, thats why led is not giving correct colors simultaneously, hope in the future, it can have some solution. Thanks again for your quick response..

Ierlandfan, 2017-06-08:
I had al lousy (broken wire) GND connection to the Sonoff wich caused the troubles with the flickering.

closed 2017-06-08 by arendst

jhoralek, 2019-02-16:
Hi guys, I went to almost the same trouble with first 30 leds. 
I'm using binary version 6.4.1 (latest) at this time.
Only first 30 is working. So the first what I tried was cut led number 31 because I tought the this one was burned out. 
No effect at all. 
Download the source of the tasmota and went to investigating what's wrong.
In sonoff.h is defined max number of leds to 512. But in file my_user_config.h where is all stuff around wifi connection, mqtt configuration and other is defined constant WS2812_LEDS with number 30. 
So my opinion is that this is the reason why only first 30 leds is on and the rest is not working.

georgbachmann, 2019-08-11:
Is there a command to set the number of LEDs that I am using? I am also using the latest binary (6.6.0).
For me also only the first 30 are lighting up. Any my power supply is more than plenty for the 120 that I would have.

ascillato, 2019-08-11:
See pixel command in the wiki.

jeparkspr, 2020-05-28:
Solution for me was amazingly simple...
Open the Tasmota device in a browser, then go to the Console and below, enter the command "pixel #" where # is the number of pixels you want.  This is documented in the Commands section of the Tasmota Wiki under the Lights group.

[image]

elik745i, 2023-05-13:
cool solution, thanks bro