arendst/Tasmota #445: question for firmware update

mrmin24, 2017-05-12:
sorry if this is a stupid question. 

How do I update the firmware without loosing the user_config e.g. Topic

Currently I change the user_config.h of a new version of firmware after pulling from git to my wifi settings etc. Then when I do firmware upgrade I loose the settings configured on the device. like "topic" mentioned and then have to reconfigure.

Is there a way to not have to do this

davidelang, 2017-05-12:
there are two possible issues here.

in the 5.x series there have been some hiccups in retaining the config when 
upgrading from a 4.x firmware. These are just bugs.

take a look at the troubleshooting page on the wiki for the question of why your 
user_config settings are not being applied.

mrmin24, 2017-05-12:
thanks, I am still not sure that this helps though. The problem is that the user config is loaded when I actually want the device to retain the programmed config. The only real issue is the topic. I use the topic as the device ID and do not want to program them every time I update.

davidelang, 2017-05-12:
if the user_config is getting loaded, overriding the existing config, then the 
existing config was corrupted (did not have the proper CFG_HOLDER value), or 
there was a bug.

The intent is to retain everything in the config when you upgrade.

exactly what versions are you upgrading from and to? are you sure that 
CFG_HOLDER is not changing between those versions?

mrmin24, 2017-05-12:
I believe you are correct. The CFG_HOLDER did change due to some testing of other forks. I did not understand it clearly. 

Thanks for the clarification. Sorry for the bother. I will check it again at the next update

closed 2017-05-12 by mrmin24