arendst/Tasmota #447: what is MQTT fingerprint? where can I get it?

omarusman, 2017-05-13:
Hi guys,

Just want to ask how can I use SSL/TLS. I'm using CloudMQTT at the moment.

I'm just confused what is the MQTT fingerprint and where can I get it?

Thank you.

jmb, 2017-05-13:
It's your SSL certificate's fingerprint. 

I'm using the same certificate for my Home Assistant web service as my MQTT broker, so pulled it out of my browser, but you could use the openssl command (using sudo here as my cert is locked down):
`sudo openssl x509 -noout -fingerprint -sha1 -inform pem -in /path/to/cert.pem`

You can pipe that through sed to remove the colons:
`sed -e 's/:/ /g'`

zuidwijk, 2017-05-28:
What if I don't use an SSL certificate?

davidelang, 2017-05-28:
Then you ignore the fingerprint as it doesn't apply to you
k

closed 2018-05-08 by stale[bot]

ramanraja, 2024-02-22:
I am connecting to an MQTT broker whose URL is like xxxxxx.us-east-1.emqxsl.com.
I am now disabling fingerprint checking altogether, but I would like to enable it for better security.
How to get the fingerprint from this URL? Obviously the browser is not of use in my case.

barbudor, 2024-02-22:
Enable fingerprint validation
Then enter `mqttfingerprint 00`
Tasmota will restart and will automatically learn the fingerprint on mqtt connection
You will see messages in the console
```
08:34:49.020 MQT: Attempting connection...
08:34:50.055 MQT: TLS connected in 994 ms, max ThunkStack used 4468
08:34:50.057 MQT: Fingerprint learned: 3D ** ** ** ** **... ** ** ** ** 10
08:34:50.154 MQT: Connected
```

Assuming your broker will be checked against the finger print on every future reconnection.

ramanraja, 2024-02-24:
Thanks, this worked.