arendst/Tasmota #448: MQTT Connection failed, rc -4

electron2302, 2017-05-13:
Hi,

I am new to Sonoff, and i successfully flashed it.

The Web UI and the Hue emulator are working fine, but i have some Problems with MQTT.
I have mosquitto running and ather 'devices' like the MQTT App on my phone have no problem and it works flawless, but the sonoff won´t conect. I always get this Error Mesage:

`14:48:48 MQTT: Connect FAILED to 192.168.2.139:1883, rc -4.`

What means rc -4, and wehre is the problem or how can i fix it ?

Thank you!

arendst, 2017-05-13:
Wiki > Tips > MQTT Overview down the bottom

electron2302, 2017-05-13:
First of all thank you for the fast answer !

Any idea where the timeout is coming from ?

electron2302, 2017-05-21:
I just reinstalled my mqtt broker and it worked

closed 2017-05-21 by electron2302