arendst/Tasmota #449: Electrodragon stuck in upgrade loop

CommodoreWhite, 2017-05-13:
My electrodragon running 8.1.0 seems to be stuck in an upgrade loop. the following is an extract of the console log:

`21:15:16 mDNS: Query done with 0 mqtt services found
21:15:16 MQTT: Connected
21:15:16 MQTT: tele/dragon-101/LWT = Online (retained)
21:15:16 MQTT: cmnd/dragon-101/POWER = 
21:15:21 RSLT: stat/dragon-101/RESULT = {"Upgrade":"Version 5.1.0 from http://sonoff.maddox.co.uk/tasmota/sonoff.ino.bin"}
21:15:21 MQTT: stat/dragon-101/RESULT = {"Upgrade":"Version 5.1.0 from http://sonoff.maddox.co.uk/tasmota/sonoff.ino.bin"}
21:15:23 MQTT: domoticz/in = {"idx":200,"nvalue":1,"svalue":""}
21:15:23 MQTT: stat/dragon-101/RESULT = {"POWER1":"ON"}
21:15:23 MQTT: stat/dragon-101/POWER1 = ON
21:15:23 HTTP: Webserver stopped
21:15:41 HTTP: Webserver active on dragon-101-7899.local with IP address 192.168.1.101
21:15:41 MQTT: Attempting connection...
21:15:42 mDNS: Query done with 0 mqtt services found
21:15:42 MQTT: Connected
21:15:42 MQTT: tele/dragon-101/LWT = Online (retained)
21:15:42 MQTT: cmnd/dragon-101/POWER = 
21:15:42 MQTT: stat/dragon-101/RESULT = {"Upgrade":"Version 5.1.0 from http://sonoff.maddox.co.uk/tasmota/sonoff.ino.bin"}
21:15:42 MQTT: stat/dragon-101/RESULT = {"Upgrade":"Version 5.1.0 from http://sonoff.maddox.co.uk/tasmota/sonoff.ino.bin"}
21:15:44 HTTP: Webserver stopped
21:16:02 HTTP: Webserver active on dragon-101-7899.local with IP address 192.168.1.101
21:16:02 MQTT: Attempting connection...
21:16:03 mDNS: Query done with 0 mqtt services found
21:16:03 MQTT: Connected
21:16:03 MQTT: tele/dragon-101/LWT = Online (retained)
21:16:03 MQTT: cmnd/dragon-101/POWER = 
21:16:04 MQTT: stat/dragon-101/RESULT = {"Upgrade":"Version 5.1.0 from http://sonoff.maddox.co.uk/tasmota/sonoff.ino.bin"}
21:16:04 MQTT: stat/dragon-101/RESULT = {"Upgrade":"Version 5.1.0 from http://sonoff.maddox.co.uk/tasmota/sonoff.ino.bin"}

Switching off my MQTT server broke the cycle only to the extent that my dragon then started bitching that it had lost connection to its MQTT host. I noticed a restart reason catalogued a "software watchdog".

I think all my dragons are similarly afflicted. Any ideas?

arendst, 2017-05-14:
I had something like that when I accidently send the mqtt upgrade message with the retained flag set!

Executing tthe command without option 1 and retainflag stil set solved it.

CommodoreWhite, 2017-05-14:
Ok. Thanks, I'll have a go with your suggestion.

I noticed I had set ride red to monitor updates availability and so
realised my dragons were gonna get updated very often. To stop this from
happening I changed the command to"update999" not realising that tasmota
would still recognise the command as"update" with a parameter. By then the
damage was done.

Can you close the issue for me. Thanks.

closed 2017-05-14 by arendst

CommodoreWhite, 2017-05-15:
Still haven managed to stop the continuous attempts of may dragon trying to
upload new software.

I've been trying to use android Mqtt dashboard to reset the retained flag,
perhaps I should be using node-red. Sorry but I need baby steps for this
one. I've given my dragon a 127.0.0.1 address for the otaurl, at least it's
no longer chewing up broadband bandwidth. Alternatively, I'll compile from
source and do manual rest of each device.

Kind regards

Just a thought for others in my predicament might be an enhancement:

Cmnd/xxx/upgrade 0

... Which disables uploads