arendst/Tasmota #451: Get info from MQTT to Home Assistant

gbpucci, 2017-05-14:
I need to get some info like ip address, firmware version, etc, from my Sonoffs, but i only can get this information using tele/ topics !!

This Works
- platform: mqtt
    state_topic: 'tele/sonoff01/STATE'
    value_template: "{{ value_json.Wifi.RSSI }}"
    name: 'Wi-Fi Signal Sonoff01'
    unit_of_measurement: '%'

Dont show nothing
  - platform: mqtt
    state_topic: 'stat/sonoff01/STATUS2'
    value_template: "{{ value_json.StatusFWR.Program }}"
    name: 'Version Sonoff '

How i can make a sensor to get the others informations using "stat" ?

arendst, 2017-05-14:
You will have to execute the command ``status 2`` in order to get the StatusFWR message.

gbpucci, 2017-05-14:
Can you show how can a make this in home assistant sensor configuration ??
Thanks !!

emilhetty, 2017-05-15:
If you just make a automation with time trigger you could send the Status 2 command to the Sonoff based on whenever you want to request the sensor value and then you will have the sensor response.
Something like this should do the magic:
`automation:
  trigger:
    platform: time
    # Matches every hour at 5 minutes past whole
    minutes: 5
    seconds: 00
  action:
    service: mqtt.publish
    data:
      topic: 'cmnd/sonoff01/status 2' `

You may not want to send the request every hour, once a day might be often enough, see this for other time triggers: https://home-assistant.io/docs/automation/trigger/

closed 2018-05-08 by stale[bot]