arendst/Tasmota #453: New ESP8285 based 1Channel Inching/Locking 5V/12V Sonoff

ed1402, 2017-05-15:
Hi All,

I just took possession of the above new device that appears to be the new version of the old device - the ics on the Itead website has not even been updated.

This board is based on the PSF-B01 itead ESP8285 module.

I have not found any custom firmware yet that appears to work - not Tasmota, ESPEasy or Espurna seems to work.

Using Tasmota it uploads fine with success flag - but this basically puts the module in an endless switch loop with no connection to wifi as setup in config file.

Anyone have any info?

Regards,
Ed

xinghusp, 2017-05-16:
Hi,
I have in the same trouble.i just brought the sonoff touch few days ago,and i tried to flash the custom firmware and can't boot to the os anymore.and the serial port didn't have any output message.the wifi cannot connect and the led of the board don't blink anymore.

sebastianlorenzen, 2017-05-16:
The Device: https://www.itead.cc/sonoff-4ch-pro.html

ed1402, 2017-05-16:
Yes albeit not the same as in the 1 channel device photo. That indicates an ESP8266 unit. Mone is ESP8285 based.

xinghusp, 2017-05-16:
The sonoff touch of mine is based on the PSF-A01 itead ESP8285 module.

ed1402, 2017-05-16:
Some pics:
[image]
[image]

ed1402, 2017-05-17:
OK - so I have managed to flash the unit with Arduino IDE on latest Tasmota.

I can access the web interface after restart, but the device now shows odd behaviour.
As far as I can establish the actual locking/inching function is controlled externally to the ESP8285.

When in locking mode the device acts normally as a switch, but when in inching mode and the device is either trigger by button press or web toggle it goes into a 10 second flip flop (switching on and off repeatedly) and then hangs. After a restart the device is frozen still and I have to re-flash firmware to get it active again.

I have tried multiple device selections in the device setup (Basic, Slampher, 1 Channel and both 4 channel options.)

My assumption is that the "on" command trigger a gpio either high or low and that the external inching function sends a signal back to the ESP8285 to switch off the trigger gpio. With the new firmware this does not happen and as such the device "loops" and eventually somehow corrupts the ESP8285.

Open to any ideas?

davidelang, 2017-05-17:
tasmota has never supported the inching functionality of this board.
k

paco-ita, 2017-05-19:
Hello everybody!
I have received the exact same product but I can't figure out how to put it into flash mode.
Any advice?
Thanks

apouzin, 2017-06-11:
Hi,
I am also using the same device and struggling on putting the device into flash mode (not sure about the wirings nore the procedure for setting it up into flash mode). Any advice/help would be appreciated ! Thanks

jktz90, 2017-06-19:
Using Arduino (Ground to RST): on the PSFB01 you need to pull GPIO2 to Ground before Powerup and afterwards connect RX TX //  TX RX using your Arduino.

zizebra, 2017-07-09:
I can only assume this device is probably the latest effort from Itead to try and stop us from using custom firmware. the relay is seem to be driven by the stc15w204s.

levonbragg, 2017-07-11:
I was able to get this sort of working... (See attached picture) I had to remove 1 IC to stop the bouncing of the relay and bridge 2 of the empty pads.
I tried all tasmota configurations, and it would not work.
I was able to get it working using this firmware 
https://github.com/KmanOz/Sonoff-HomeAssistant
So, at least it works now, but maybe someone smarter than I can modify the tasmota firmware (which I like better) to make it work.

[image]

zizebra, 2017-07-11:
Brilliant stuff. Didn't think about that. Will try that and see if it helps.

arendst, 2017-07-11:
The devices discussed in this issue are based on the esp8285 chip while the original device used the esp8266 chip. The main difference is the way flash is addressed and loading the default sonoff firmware on these devices will probably NOT work unless you take the following steps:

- Initially load tasmota firmware file ``sonoff-touch-4ch-bnsz.bin`` or compile your own using the DOUT flash mode.
- Open the configuration webpage (or use serial commands) to select module ``12 1 Channel``
- If things are as I expect it should now work like a standard 1 channel inching device

zizebra, 2017-07-11:
Mine is definitely esp8285. That is the one I compiled and loaded on mine. I believe I tried the 1 channel and the behavior was not perfect. In my case, I noted that when using the inching method, it would work perfectly as expected but with one issue. I could turn on or off generally about 5 times in row. Thereafter, the next 2 or 3 attempts would not have effect. 
I will test again just to verify that your suggested options works on this device.

zizebra, 2017-07-11:
levonbragg, do you still have the IC number for the chip you removed.

levonbragg, 2017-07-11:
zizebra, I do not. All of mine are blank. Looks like they were rubbed off.
After digging around, it looks like a STC15F101W series 8051 micro...

zizebra, 2017-07-11:
Hope itead is not trying to revive the useless ESP8266 Black board T5

levonbragg, 2017-07-11:
arendst,
I can confirm that after adding the pullup resistor, and loading the module with the sonoff-touch-4ch firmware, and setting it to "12 1 Channel" that the board seems to be working great.
Thanks for the help!

zizebra, 2017-07-11:
arendst, on my end there is no joy. I still see the behavior where some commands will have no effect. it usually takes up to 4 commands to get a response. thereafter the next 4-5 commands works before it stops responding again. below is a trace from webconsole

[image]

arendst, 2017-07-11:
@zizebra you have a MQTT server problem. This causes delays in command processing. If you fail to connect the MQTT server execute command mqttretry 60 to make more room for command processing.

zizebra, 2017-07-11:
thank you., I didn't think that to be the problem. My sd card for my PI packed up before I received this device. I will re-install the mqtt server and test

zizebra, 2017-07-13:
Finally managed to test. It is definitely working perfectly with mqtt. One minor issue which I picked up is that after loading firmware, the relay default is activated. When I choose 1ch on configuration. The relay is changed to default on off. Occasionally when the sonoff restarts, the relay is activated momentarily before being Switch off. Does any one have a quick workaround for this. I would like the relay start default on off throughout the reboots.

jktz90, 2017-08-07:
You can simply use this wiring (if you have someone to help you, you dont neet so solder).
If you use an arduino to flash: connect GROUND with RESTART on your Arduino, select ESP12-E in Arduino IDE and you are good to go.
[image]

moonfarms, 2017-08-08:
hello jktz90, please if I connect my usb to uart like this it does overheat very fast, so instead  I am powering the sonoff from his own usb and connecting tx rx and Ground of module to my usb to uart but I get eeprom error, I also not sure that ESP goes into Flash mode

jktz90, 2017-08-08:
You probably plugged Ground/3V the wrong way around. You can definitely use your adapter to power the Sonoff.
Try switching RX with TX, the 5th pin from the button (where the red line points to) has to be connected to TX on your adapter.

Before you power up the Sonoff, you have to pull the pin out in the bottom to Ground (I don't have the official datasheet anymore) in order to force flash mode (I can't remember how exactly but the LED blinked differently to indicate the flash mode).

moonfarms, 2017-08-08:
3v is the square pin one (blue wire on your pic) isn it?

moonfarms, 2017-08-08:
thanks again I can get it into flash mode i think (the relais keeps opening and closing) (it is also possible to do it by pressing the 2 buttons simultaneously on power up. Still no luck in flushing, but i am using platformio so maybe this is the problem. Thanks alot again jktz90 for your help, really appreciated

jktz90, 2017-08-08:
Just checked it... u're in flash mode if the red led next to the relais is on permanently.

moonfarms, 2017-08-08:
can only get that red led to blink....

moonfarms, 2017-08-09:
jktz90 I managed to flash !!!!! THANK YOU IN CAPITAL LETTERS. Everything you wrote above is correct, led has to be solid red before flash. I dont know what i was doing wrong.....
Have a nice day

LurchiStromberg, 2017-08-09:
HI Guys!
I am totally new in this Scene so please be gracious to me, i am not even a Technician.

I ordered that Item from Banggood and hoped, i will receive a ESP8266 Model, what is easy to flash by soldering some pins on the Board.
Yesterday i received the Board and was surprised by its Pinout.
So i done a Google research and luckily I found this Thread here, and that its not an ESP 8266 but an ESP 8285.
The wireing @jktz90 showed to us here confused me some, so i try to figure it out myself at looking at Datasheets.

Here is the Result:

[image]

and

[image]

What i am surprised of is, that only 3,3 Volts and GND should be at the 4 Pin Connector.
Is it really like that, that RX and TX are not supported at one of the 4 or 6 Pins Connector?

What you used to get contact to the Pins of J3 and J2 ?
Soldering cables at this Pins is nearly impossible (like i said, i am not so familiar with those Stuff)

And GPIO0 must be connected to GND to put the Board into Flashmode? It is not possible with the 2 Switches on the Board?

GND to put into flashmode can be the GND next to 3,3 Volts?

I would be happy if somebody could explain all this, hopefully jktz90 a little more detailed.
Thank you in advance

PeggyFree, 2017-08-09:
I flashed 15 sonoff devices with zero solder, so you can do as well. Look
for "male dupont cable" that you can use between the ftdi cable and the
holes you show on your pic.

Esp 8285 does not really makes difference with 8266 as far as I know,
regarding the flash method.

moonfarms, 2017-08-09:
I used needles to hold on rx tx pins and somebody to help to hold the board / flash firmware

davidelang, 2017-08-09:
I use pogo pins, you can insert them into a female header (with effort) to keep 
them aligned.

https://www.adafruit.com/product/394

jktz90, 2017-08-09:
Like I said in my post, you dont need to solder. You can simply hold  the male pins to the respective pin out.

It made me go crazy tho, so I decided to simply solder it (took me 5 minutes). I had to flash a lot of times  holding 3 male pins is almost impossible. Find someone to help you and it should be easy.

LurchiStromberg, 2017-08-09:
@jktz90 The connections I posted is right ?
You also have the 1 Channel Jog inching Board?

What exact Firmware you flashed on that Board?

The Board goes in Flash Mode if I put GPIO0 to GND and connect it to the FTDI?

Sorry if i ask twice here but i like to get sure

jktz90, 2017-08-09:
Why does my connection confuse you? Its exactly what you have found in the datasheet. GPIO0 to Ground (which triggers flash mode - its enough to connect it once upon power up) and RX/TX to your FTDI adapter to send data.

After you set up your connection, go into your Arduino IDE and select ESP 8285 as your device. I for myself write my own sketches. So I control the relay by UDP packets (send by my smartphone or FHEM).
But you're free to flash firmwares like ESPeasy aswell (just go to the wiki).

LurchiStromberg, 2017-08-09:
Sorry if i maybe annoy you, but like i said i am a bloody beginner... 
And after reading hours and hours Eyes get blurry and mind freaks out ;)
So if i follow those steps like @arendst wrote before:

>Initially load tasmota firmware file sonoff-touch-4ch-bnsz.bin or compile your own using the DOUT flash mode.
Open the configuration webpage (or use serial commands) to select module 12 1 Channel
If things are as I expect it should now work like a standard 1 channel inching device

so this Download:
[https://github.com/arendst/Sonoff-Tasmota/releases/download/v5.2.4/sonoff-touch-4ch-bnsz.bin](url)

It will work on my 1 CH Board?

Lasst Question (hopefully) How the Settings need to be for the ESP8285 in Arduino?

arendst, 2017-08-09:
@LurchiStromberg Sorry to break in but that information has been superseded a while ago. That's what you get with old issues.

Forget about sonoff-touch..bin file. It is no longer available in the latest release.

Only use the latest release sonoff.bin file which has been compiled with flash mode dout. In case you want to compile yourself, ALWAYS use dout.

LurchiStromberg, 2017-08-09:
@arendst But if i use Arduino 1.8.3, File, Open and open your  sonoff.bin i also get the same Error Message from Arduino:

[image]

arendst, 2017-08-09:
Yes that's because you cannot open bin files with arduino ide...

If this is your first experience with arduino ide I suggest to read the wiki from the start how to setup your environment using the files in this repository.

LurchiStromberg, 2017-08-10:
@arendst Ok, after several hours i am now comming to a complete different Software or better Programm called esptool.py
i have that running in a Terminal now so it will do the job to flash your sonoff.bin to the ESP8285 ?

Now i also see some clearer what you Guys talking about with flashmode dout...
I was searching for that at Arduino desperatly *lol*

Can somebody give me a hint how the exact Command Line has to look like?
I am still scared to do the wrong Settings and to RIP the Board

davidelang, 2017-08-10:
On Thu, 10 Aug 2017, LurchiStromberg wrote:

> I am still scared to do the wrong Settings and to RIP the Board

The way the esp family of chips are designed, you cannot brick them with 
software. As long as you don't apply more than 3.3v to them, you can always put 
them in programming mode and at that point it doesn't matter what was flashed to 
them in the past.

arendst, 2017-08-10:
See wiki troubleshooting section for correct syntax. No need to erase.

moonfarms, 2017-08-10:
Here a pic of my setup. You can put the board in flash mode by pressing the TWO buttons simultaneously on power up, then press the outer button again and you will have the TWO led solid red. Now hold your needles on tx rx pins of  PSF-B01 and flash.
[image]

LurchiStromberg, 2017-08-10:
@moonfarms 
Hi Paul, thank you for your Picture. As i see you didnt solder the red Wire, 3,3 Volts ????

moonfarms, 2017-08-10:
No, I am powering the unit trough the usb, black wire is for ground for tx rx

LurchiStromberg, 2017-08-10:
@moonfarms 
Oh... I read that 3,3 Volts is Urgent not to damage the EPS Chip at flashing...

Its pretty confusing if some Guys here tell to read Wikis, but at the End all do different stuff *lol*

But of course i am thankful to see, that there is also a more easy Way.

One last question about the Flash Mode. If I hold the 2 Buttons and Power the Board with the mini USB nothing happens that Moment. After I release the 2 Buttons the Relais start to pulse.
At the Moment i push the outer Button, both LEDs are solid.

Is that the Way it works?

moonfarms, 2017-08-10:
The board will step down the usb 5v  to 3.3 to power the esp chip
If the TWO leds are SOLID red you can flash. I am flashing with platformio

LurchiStromberg, 2017-08-10:
@moonfarms 
Thank you Paul for the hints, make Things more easy for me 👍 
So if i dont need the 3,3 Volts from the FTDI i could also use an old USB-COM Stick?
Because i am still waiting for my Serial Converter from BG...

moonfarms, 2017-08-10:
if your usb-com does usb to uart.....

LurchiStromberg, 2017-08-10:
I use the one Attached.
I really dont know if it has UART. I used it to be able to connect a Dreambox to access Bootloader via Terminal.

[image]
[image]

jktz90, 2017-08-10:
> Its pretty confusing if some Guys here tell to read Wikis, but at the End all do different stuff lol

Because - in the end - it is a ESP8285 chip you are working with. It doesn't matter if its on a Sonoff board or Nodemcu... it is always the same. 
I flashed it months ago and I had no clue about possible button combination to force flash mode (it may also differ between 5V and 5V/12V model). So it is easier to just connect GPIO0 to Ground to force flash mode **on every ESP8285 chip**.

Maybe it is easier if you describe what u want to archieve with your Sonoff. In my case I use the Sonoff to open my garage and house door. It's important to node that your Relay will always trigger on powerup (ESPEASY Sonoff touch) which forced me to use my own sketch - so my door doesn
t accidently open on power reset.

moonfarms, 2017-08-10:
IMHO I think your usb to serial does not have a 3v output and you may fry the board

LurchiStromberg, 2017-08-10:
@arendst 
One Question to you also. Today my other sOnOffs arrived, the 90-250 V Version.
This ones are using the "normal" ESP 8266.

Your sonoff.bin is also for that Devices?

LurchiStromberg, 2017-08-10:
@moonfarms 
Well i would connect the 5V Version to USB Charger as you do on your Pic and just use the RX,TX,GND from the USB-Com Adapter. So no Power comes thru it.

arendst, 2017-08-10:
The latest sonoff.bin file is functional on ALL devices as it is compiled with Flash Mode Dout.

moonfarms, 2017-08-10:
@ lurchi IMHO the voltage of tx rx is relevant so if your usb com tx rx outputs 5v you are gambling
Let us know

LurchiStromberg, 2017-08-10:
@moonfarms 
> No, I am powering the unit trough the usb, black wire is for ground for tx rx

Uhm....it was you who told me that you flash the Board with 5V Power from USB and GND, TX, RX from FTDI.

I also asked you and what i Quote was your Answer ;)

I think its a missunderstanding, for flashing you also use the 3,3 V from FTDI, even its not connected at your Picure, right?

[90 more replies]

closed 2018-09-04 by ascillato2