arendst/Tasmota #455: POW NTP issue

rene9900, 2017-05-16:
10:27:15 MQTT: tele/POW/STATE = {"Time":"1970-01-01T10:27:14", "Uptime":11, "Vcc":3.169, "POWER":"ON", "Wifi":{"AP":1, "SSID":"Airport", "RSSI":72, "APMac":"6C:70:9F:DA:65:78"}}
10:27:15 MQTT: tele/POW/ENERGY = {"Time":"1970-01-01T10:27:15", "Total":0.178, "Yesterday":0.000, "Today":0.178, "Period":1, "Power":18, "Factor":0.50, "Voltage":238, "Current":0.149}

i tried the NTPserver command in console and added 3 ntp servers for my country, it reboots after each is added but time is still not ticking...
i assume that time is standing still is also the reason why my Yesterday use stay on zero, and the counter keeps counting for today?

davidelang, 2017-05-16:
at midnight the device takes the total for the day and moves it to the 
'yesterday' value.

If the device is not powered on at that point, it won't happen, if the device is 
rebooted during the day, the running total for the day will be lost and it will 
start over.

As for why ntp isn't working, there are a lot of possible reasons for that, 
including a dozen or more different possible things with your Internet 
connection. you would have to sniff your network and watch the Internet 
connection to see what's happening when it attempts to connect to the ntp 
servers and what response shows up.

rene9900, 2017-05-16:
funny thing is when i set it to default it works, but timezone is -1 hour wrong, as i set it to denmark its good old 1970 again 👯‍♂️

davidelang, 2017-05-16:
that sounds like you are setting the timezone wrong. what settings do you try to 
use?
k

rene9900, 2017-05-16:
i did not mess with timezone, i had the idea that ntp server told what time it is, but i need to tell it +1 i see

rene9900, 2017-05-16:
08:24:57 RSLT: Receive topic /Timezone, data size 0, data 
08:24:57 RSLT: DataCb Group 0, Index 1, Type TIMEZONE, Data  ()
08:24:57 MQTT: stat/POW/RESULT = {"Timezone":1}
08:25:11 Wifi: Checking connection...

rene9900, 2017-05-16:
Fixed it with cmd  
Timezone +2

thanks for your help davidlang

08:28:48 CMND: timezone  2
08:28:48 RSLT: Receive topic /timezone, data size 2, data  2
08:28:48 RSLT: DataCb Group 0, Index 1, Type TIMEZONE, Data  2 ( 2)
08:28:48 MQTT: stat/POW/RESULT = {"Timezone":2}
08:28:48 Config: Saved configuration (1472 bytes) to flash at FB and count 258
09:29:01 Wifi: Checking connection...
09:29:01 Wifi: Connected

arendst, 2017-05-16:
@rene9900 how do you set it to Denmark ? All you can do is change timezone from -12 to 12 and, in your case to 99 which will provide for the Daylight Saving you need in Denmark too.

rene9900, 2017-05-16:
it was the ntp i had the idea it was configurable i set it to:
NtpServer1 dk.pool.ntp.org

but since i used default from user_config.h it worked, just 1 hour off my time. so it was timezone that bugged me

davidelang, 2017-05-16:
ntp servers distribute raw zulu time (GMT), ntp servers don't care about 
timezones. It doesn't matter where in the world you are and what timeservers you 
use, they will all provide the same time.

timezone settings are purely local and adjust what time the system uses as an 
offset from the time provided by NTP.

closed 2017-05-16 by rene9900