arendst/Tasmota #456: Sonoffs are behind one hour

sebastianlorenzen, 2017-05-16:
I'm using default Settings for time zone configuration. Amsterdam and Berlin are in the same time zone.
DST and STD is also default. (+120 +60)
Regardless of these settings the Sonoff is just +1 as configured in the time zone setting. So it looks like it ignores the DST configuration.

Snip of the config:

// -- Time - Up to three NTP servers in your region
#define NTP_SERVER1            "ptbtime1.ptb.de"  // [NtpServer1] Select first NTP server by name or IP address (129.250.35.250)
#define NTP_SERVER2            "0.pool.ntp.org"   // [NtpServer2] Select second NTP server by name or IP address (5.39.184.5)
#define NTP_SERVER3            "ptbtime2.ptb.de"  // [NtpServer3] Select third NTP server by name or IP address (93.94.224.67)

// -- Time - Start Daylight Saving Time and timezone offset from UTC in minutes
#define TIME_DST               Last, Sun, Mar, 2, +120  // Last sunday in march at 02:00 +120 minutes

// -- Time - Start Standard Time and timezone offset from UTC in minutes
#define TIME_STD               Last, Sun, Oct, 3, +60   // Last sunday in october 02:00 +60 minutes

// -- Application ---------------------------------
#define APP_TIMEZONE           1                 // [Timezone] +1 hour (Amsterdam) (-12 .. 12 = hours from UTC, 99 = use TIME_DST/TIME_STD)

Status from the Sonoff itself:
{"StatusTIM":{"UTC":"Wed May 10 20:43:24 2017", "Local":"Wed May 10 21:43:24 2017", "StartDST":"Sun Mar 26 02:00:00 2017", "EndDST":"Sun Oct 29 03:00:00 2017", "Timezone":1}}

So "Local" is just +1 and not +2 as it should be.

I have also used the standard config for the time services. Same issue.

arendst, 2017-05-16:
As you rightfully copied from the user_config.h:
```
// -- Application ---------------------------------
#define APP_TIMEZONE 1 // [Timezone] +1 hour (Amsterdam) (-12 .. 12 = hours from UTC, 99 = use TIME_DST/TIME_STD)
```
It says that in order to use DST you have to use command ``timezone 99``

sebastianlorenzen, 2017-05-16:
Thanks :-)
Read first, then ask

closed 2017-05-16 by arendst