arendst/Tasmota #457: Not enough memory to upload?

TommySharpNZ, 2017-05-17:
Surely the default configuration should be optimised to fit in the standard 1MB flash?
This would make it a lot easier for newbies......

davidelang, 2017-05-17:
part of the problem is that libraries get larger, so a config that works with 
one version of libraries may not work with the next version.

We know that there are some version bumps that push this over the edge right 
now.

@arendst is working on finding ways to reduce the size, the other options are to 
shut off features or make every upgrade be a two-stage upgrade where you first 
have to flash a minimal build before flashing the next full build.

arendst, 2017-05-17:
The default configuration FITS nicely in 1M sonoff.

Try the precompiled versions or read the wiki about the 1m0 linker script to be installled in your environment to gain code space.

closed 2018-05-08 by stale[bot]