arendst/Tasmota #458: Need clarification of some hardcoded timing on PulseTime

stefanbode, 2017-05-17:
Hi Arend, hi David,

to enhance the new Wiki Page regarding pulsetimers I have some open questions about hard coded timings in the sonoff.ino.

1.  "case (STATES/10)*4" : There is a "IF-clause" checking pulsetime between 1 and 29. Does this limit the max pulsetime to 3 seconds, 30 seconds??
2.  "void stateloop()": There is a check against a value 112. Why?
3.  "void every_second()": There is a check against 111. Why?

Regarding 2 and 3 I assume you minimize checks. If time is high enough only every second. But what is the unit of pulse_timer[i]? isn't it 0.1 second? Why do you decrease only 1  (pulse_timer[i]--;) and not 10 in this case.

I'm currently using it to control a shutter (37 seconds run time) and have some "problems". Will add the answer to the wiki.

arendst, 2017-05-17:
Stefan (I suppose),

1. The test is there to save flash wear where any power change < 3 seconds induced by pulsetime is not saved to flash.
2. (and 3.) Originally the granularity was set to 0.1 seconds until someone wanted a much bigger pulsetime (in the minutes) which was unavailable in the 16 bit pulsetime variables. To stay backward compatible (one of my hobbies) I had to come up with the solution where pulsetime values below 112 would mean an increment every 0.1 second and any value from 112 would increment by 1 second. 

That's what I tried to explain in the wiki > commands:
```
PulseTime<x>     |              | Show current PulseTime of relay<x> in 0.1 seconds
PulseTime<x>     | 0 | off      | (Default) Disable use of PulseTime for relay<x>
PulseTime<x>     | 1..111       | Set PulseTime for relay<x> with 0.1 seconds increment
PulseTime<x>     | 112..64900   | Set PulseTime for relay<x> with 1 seconds increment starting with 12 
```

stefanbode, 2017-05-17:
Uhhh. This explains my problem. 1-111 == 0.1 to 11.1 seconds and then 112 seconds to 64900 seconds. This explains why 37 seconds did not work... so everything between 11.1 and 112 seconds is not possible without a code modification. I do not understand the requirement. 64900 is roughly 2h in the 0.1 intervals. More than enough. normally.

davidelang, 2017-05-17:
no, 1-111 = 0.1-11.1 seconds
112 = 12 seconds
212 = 112 seconds
(if pulsetime > 111, then delay is pulsetime - 100 seconds)

arendst, 2017-05-17:
@davidelang Totally correct!

shmrymbd, 2017-10-30:
how to reduce pulsetime to 50ms for example? i need to generate multiple pulse that is fast enough. how to do hardcode. thanks

shmrymbd, 2018-01-10:
anyone can help? i am looking for pulsetime below 50ms. i need to trigger coin machine. thank you

stefanbode, 2018-01-10:
This is not part of the implementation and please keep in mind, that the ESP is only a cheap chip that has its limitation. A pulsetime below 100ms can get difficult with different kind of logging and other processes. The ESP is not made for this. Especially a general firmware like this cannot guarantee these kinds of pulsetimes.

shmrymbd, 2018-01-10:
hi can i shift this part 

    for (byte i = 0; i < MAX_PULSETIMERS; i++) {
      if ((pulse_timer[i] > 0) && (pulse_timer[i] < 112)) {
        pulse_timer[i]--;
        if (!pulse_timer[i]) {
          ExecuteCommandPower(i +1, 0);
        }
      }
    }

to 0.05 s for my own used?

stefanbode, 2018-01-10:
You can do. but than you get a lack because 111 == 0.05s x 111 = 5.55sec. 112 is then 11.2sec. This means with this modification you cannot do anything between 5.55sec and 11.1sec.  Additionally please be prepared, that the execution is not guaranteed to lost not longer than 50ms. It will execute at 50ms or later as soon as possible.

closed 2018-05-24 by stale[bot]