arendst/Tasmota #461: Static IP Address ?

JacoFourie, 2017-05-18:
Hi. Where do I configure a static ip address ?
I can choose a new SSID but I am not able to choose a static IP there.
I also did try to add it in the config.ini file but that did not change a thing.
The unit is now in the roof controlling a bathroom extractor. It works with DHCP but I would love to keep all my sonoffs in the same ip range.

arendst, 2017-05-18:
See Wiki > commands > IPAddress1 to IPAddress4

JacoFourie, 2017-05-18:
DUDE !!!!!. Thanks so much for this amazing code. well done. This rocks.

JacoFourie, 2017-05-18:
How do I change the time zone ? I see the time is out by 1 hour in the console.

davidelang, 2017-05-18:
see the timezone command https://github.com/arendst/Sonoff-Tasmota/wiki/Commands

JacoFourie, 2017-05-18:
Aaaa I looked under the wrong section. OK thanks soo much.

closed 2017-05-18 by arendst

krtschmr, 2017-09-12:
is it enough to only set IPADDRESS1 and not change the other ones since they are identical?

htcheroportugal, 2017-09-25:
sorry but how a can change my ip for 192.168.1.109?

http://192.168.1.108/cm?cmnd=IPAddress1%192.168.1.109 this is correct?

fabricecoulon, 2018-01-12:
I ran into the same issue in the past. The trick is to read the [documentation](https://github.com/arendst/Sonoff-Tasmota/wiki/Commands)  and to be able to see the `%20` between the command and the payload:
```
http://sonoff/cm?cmnd=CMD%20PAYLOAD
```
So, in your case CMD is `IPAddress1` and the PAYLOAD is `192.168.1.109`.

Then, power cycle, and it should work.

Markkyboy, 2019-05-03:
Thank you @arendst for this feature, just what I was looking for :)