arendst/Tasmota #462: Sonoff RF going into infinite power alternating cycle

mihalski, 2017-05-19:
I got my Sonoff RF remote so I decided to finally program my modules, but I get this strange behaviour:
```
00:00:00 APP: Project sonoff Sonoff (Topic sonoff, Fallback DVES_9E20FE, GroupTopic sonoffs) Version 5.1.1
00:00:00 Wifi: Connecting to AP1 DobNet in mode 11N as sonoff-0254...
00:00:11 Wifi: Connected
00:00:11 mDNS: Initialized
00:00:11 HTTP: Webserver active on sonoff-0254.local with IP address 192.168.1.117
02:30:52 MQTT: Attempting connection...
02:30:54 mDNS: Query done with 0 mqtt services found
02:30:54 MQTT: Connected
02:30:54 MQTT: tele/sonoff/LWT = Online (retained)
02:30:54 MQTT: cmnd/sonoff/POWER = 
02:30:54 MQTT: tele/sonoff/INFO1 = {"Module":"Sonoff RF", "Version":"5.1.1", "FallbackTopic":"DVES_9E20FE", "GroupTopic":"sonoffs"}
02:30:54 MQTT: tele/sonoff/INFO2 = {"WebserverMode":"Admin", "Hostname":"sonoff-0254", "IPaddress":"192.168.1.117"}
02:30:54 MQTT: tele/sonoff/INFO3 = {"Started":"Hardware Watchdog"}
02:30:55 MQTT: stat/sonoff/RESULT = {"POWER":"OFF"}
02:30:55 MQTT: stat/sonoff/POWER = OFF
02:31:02 MQTT: tele/sonoff/STATE = {"Time":"2017-05-19T02:31:02", "Uptime":0, "Vcc":3.045, "POWER":"OFF", "Wifi":{"AP":1, "SSID":"DobNet", "RSSI":82, "APMac":"80:2A:A8:8A:21:B3"}}
02:31:03 MQTT: stat/sonoff/RESULT = {"POWER":"ON"}
02:31:03 MQTT: stat/sonoff/POWER = ON
02:31:03 MQTT: stat/sonoff/RESULT = {"POWER":"OFF"}
02:31:03 MQTT: stat/sonoff/POWER = OFF
02:31:04 MQTT: stat/sonoff/RESULT = {"POWER":"ON"}
02:31:04 MQTT: stat/sonoff/POWER = ON
02:31:05 MQTT: stat/sonoff/RESULT = {"POWER":"OFF"}
02:31:05 MQTT: stat/sonoff/POWER = OFF
02:31:06 MQTT: stat/sonoff/RESULT = {"POWER":"ON"}
02:31:06 MQTT: stat/sonoff/POWER = ON
02:31:08 MQTT: stat/sonoff/RESULT = {"POWER":"OFF"}
02:31:08 MQTT: stat/sonoff/POWER = OFF
02:31:09 MQTT: stat/sonoff/RESULT = {"POWER":"ON"}
02:31:09 MQTT: stat/sonoff/POWER = ON
02:31:10 MQTT: stat/sonoff/RESULT = {"POWER":"OFF"}
02:31:10 MQTT: stat/sonoff/POWER = OFF
02:31:11 MQTT: stat/sonoff/RESULT = {"POWER":"ON"}
```
Basically if I push either the button on the Sonoff RF or on the remote to toggle the power it just cycles back and forth until I press either of those buttons again (and sometimes it ignores even that).

Any idea what might be happening here?

davidelang, 2017-05-19:
increase the debugging level so that we can see exactly why it's switching.
k

mihalski, 2017-05-19:
Here it is powering up with NO interaction from me:
```
00:00:00 APP: Project sonoff Sonoff (Topic sonoff, F00:00:05 RSLT: stat/sonoff/RESULT = {"POWER":"OFF"}
00:00:05 RSLT: stat/sonoff/POWER = OFF
00:00:06 RSLT: stat/sonoff/RESULT = {"POWER":"ON"}
00:00:06 RSLT: stat/sonoff/POWER = ON
00:00:06 Wifi: Connected
00:00:06 mDNS: Initialized
00:00:06 HTTP: Webserver active on sonoff-0254.local with IP address 192.168.1.117
02:38:39 RSLT: stat/sonoff/RESULT = {"POWER":"OFF"}
02:38:39 RSLT: stat/sonoff/POWER = OFF
02:38:39 MQTT: Attempting connection...
02:38:41 mDNS: Query done with 1 mqtt services found
02:38:41 mDNS: Service found on homepi ip 192.168.1.64 port 1883
02:38:41 MQTT: Connected
02:38:41 MQTT: tele/sonoff/LWT = Online (retained)
02:38:41 MQTT: cmnd/sonoff/POWER = 
02:38:41 MQTT: tele/sonoff/INFO1 = {"Module":"Sonoff RF", "Version":"5.1.1", "FallbackTopic":"DVES_9E20FE", "GroupTopic":"sonoffs"}
02:38:41 MQTT: tele/sonoff/INFO2 = {"WebserverMode":"Admin", "Hostname":"sonoff-0254", "IPaddress":"192.168.1.117"}
02:38:41 MQTT: tele/sonoff/INFO3 = {"Started":"Power on"}
02:38:41 MQTT: stat/sonoff/RESULT = {"POWER":"ON"}
02:38:41 MQTT: stat/sonoff/POWER = ON
02:38:42 MQTT: stat/sonoff/RESULT = {"POWER":"OFF"}
02:38:42 MQTT: stat/sonoff/POWER = OFF
02:38:42 MQTT: stat/sonoff/RESULT = {"POWER":"OFF"}
02:38:42 MQTT: stat/sonoff/POWER = OFF
02:38:43 MQTT: stat/sonoff/RESULT = {"POWER":"ON"}
02:38:43 MQTT: stat/sonoff/POWER = ON
02:38:44 MQTT: stat/sonoff/RESULT = {"POWER":"OFF"}
02:38:44 MQTT: stat/sonoff/POWER = OFF
02:38:45 MQTT: stat/sonoff/RESULT = {"POWER":"ON"}
02:38:45 MQTT: stat/sonoff/POWER = ON
02:38:46 MQTT: stat/sonoff/RESULT = {"POWER":"OFF"}
02:38:46 MQTT: stat/sonoff/POWER = OFF
02:38:47 MQTT: stat/sonoff/RESULT = {"POWER":"ON"}
02:38:47 MQTT: stat/sonoff/POWER = ON
02:38:48 MQTT: stat/sonoff/RESULT = {"POWER":"OFF"}
02:38:48 MQTT: stat/sonoff/POWER = OFF
02:38:49 MQTT: tele/sonoff/STATE = {"Time":"2017-05-19T02:38:49", "Uptime":0, "Vcc":3.046, "POWER":"OFF", "Wifi":{"AP":1, "SSID":"DobNet", "RSSI":82, "APMac":"80:2A:A8:8A:21:B3"}}
02:38:49 MQTT: stat/sonoff/RESULT = {"POWER":"ON"}
02:38:49 MQTT: stat/sonoff/POWER = ON
02:38:50 MQTT: stat/sonoff/RESULT = {"POWER":"OFF"}
02:38:50 MQTT: stat/sonoff/POWER = OFF
02:38:51 MQTT: stat/sonoff/RESULT = {"POWER":"ON"}
02:38:51 MQTT: stat/sonoff/POWER = ON
02:38:52 MQTT: stat/sonoff/RESULT = {"POWER":"OFF"}
02:38:52 MQTT: stat/sonoff/POWER = OFF
02:38:53 MQTT: stat/sonoff/RESULT = {"POWER":"ON"}
02:38:53 MQTT: stat/sonoff/POWER = ON
02:38:54 MQTT: stat/sonoff/RESULT = {"POWER":"OFF"}
02:38:54 MQTT: stat/sonoff/POWER = OFF
02:38:55 MQTT: stat/sonoff/RESULT = {"POWER":"ON"}
02:38:55 MQTT: stat/sonoff/POWER = ON
02:38:57 MQTT: stat/sonoff/RESULT = {"POWER":"OFF"}
02:38:57 MQTT: stat/sonoff/POWER = OFF
02:38:58 MQTT: stat/sonoff/RESULT = {"POWER":"ON"}
02:38:58 MQTT: stat/sonoff/POWER = ON
02:38:59 MQTT: stat/sonoff/RESULT = {"POWER":"OFF"}
02:38:59 MQTT: stat/sonoff/POWER = OFF
02:39:00 MQTT: stat/sonoff/RESULT = {"POWER":"ON"}
02:39:00 MQTT: stat/sonoff/POWER = ON
02:39:01 MQTT: stat/sonoff/RESULT = {"POWER":"OFF"}
02:39:01 MQTT: stat/sonoff/POWER = OFF
02:39:02 MQTT: stat/sonoff/RESULT = {"POWER":"ON"}
02:39:02 MQTT: stat/sonoff/POWER = ON
02:39:03 MQTT: stat/sonoff/RESULT = {"POWER":"OFF"}
02:39:03 MQTT: stat/sonoff/POWER = OFF
02:39:04 MQTT: stat/sonoff/RESULT = {"POWER":"ON"}
02:39:04 MQTT: stat/sonoff/POWER = ON
02:39:05 MQTT: stat/sonoff/RESULT = {"POWER":"OFF"}
02:39:05 MQTT: stat/sonoff/POWER = OFF
02:39:06 MQTT: stat/sonoff/RESULT = {"POWER":"ON"}
02:39:06 MQTT: stat/sonoff/POWER = ON
02:39:08 MQTT: stat/sonoff/RESULT = {"POWER":"OFF"}
02:39:08 MQTT: stat/sonoff/POWER = OFF
```

davidelang, 2017-05-19:
see the troubleshooting page, increase the debugging level to 'more debugging' 
and see what is happening.

mihalski, 2017-05-19:
```
02:47:10 MQTT: Connected
02:47:10 MQTT: tele/sonoff/LWT = Online (retained)
02:47:10 MQTT: cmnd/sonoff/POWER = 
02:47:10 MQTT: tele/sonoff/INFO1 = {"Module":"Sonoff RF", "Version":"5.1.1", "FallbackTopic":"DVES_9E20FE", "GroupTopic":"sonoffs"}
02:47:10 MQTT: tele/sonoff/INFO2 = {"WebserverMode":"Admin", "Hostname":"sonoff-0254", "IPaddress":"192.168.1.117"}
02:47:10 MQTT: tele/sonoff/INFO3 = {"Started":"Software/System restart"}
02:47:11 MQTT: stat/sonoff/RESULT = {"POWER":"OFF"}
02:47:11 MQTT: stat/sonoff/POWER = OFF
02:47:18 MQTT: tele/sonoff/STATE = {"Time":"2017-05-19T02:47:18", "Uptime":0, "Vcc":3.053, "POWER":"OFF", "Wifi":{"AP":1, "SSID":"DobNet", "RSSI":70, "APMac":"80:2A:A8:8A:21:B3"}}
02:47:22 APP: Multipress 1
02:47:22 MQTT: stat/sonoff/RESULT = {"POWER":"ON"}
02:47:22 MQTT: stat/sonoff/POWER = ON
02:47:22 APP: Multipress 1
02:47:23 APP: Multipress 1
02:47:23 Config: Saved configuration (1492 bytes) to flash at FB and count 236
02:47:23 MQTT: stat/sonoff/RESULT = {"POWER":"OFF"}
02:47:23 MQTT: stat/sonoff/POWER = OFF
02:47:23 APP: Multipress 1
02:47:24 APP: Multipress 1
02:47:24 Config: Saved configuration (1492 bytes) to flash at FB and count 237
02:47:24 MQTT: stat/sonoff/RESULT = {"POWER":"ON"}
02:47:24 MQTT: stat/sonoff/POWER = ON
02:47:25 APP: Multipress 1
02:47:25 APP: Multipress 1
02:47:25 Config: Saved configuration (1492 bytes) to flash at FB and count 238
02:47:26 MQTT: stat/sonoff/RESULT = {"POWER":"OFF"}
02:47:26 MQTT: stat/sonoff/POWER = OFF
02:47:26 APP: Multipress 1
02:47:26 APP: Multipress 1
02:47:26 Config: Saved configuration (1492 bytes) to flash at FB and count 239
02:47:27 MQTT: stat/sonoff/RESULT = {"POWER":"ON"}
02:47:27 MQTT: stat/sonoff/POWER = ON
02:47:27 APP: Multipress 1
02:47:27 Config: Saved configuration (1492 bytes) to flash at FB and count 240
02:47:27 APP: Multipress 1
02:47:27 Wifi: Checking connection...
02:47:27 Wifi: Connected
02:47:28 MQTT: stat/sonoff/RESULT = {"POWER":"OFF"}
02:47:28 MQTT: stat/sonoff/POWER = OFF
02:47:28 APP: Multipress 1
02:47:28 Config: Saved configuration (1492 bytes) to flash at FB and count 241
02:47:28 APP: Multipress 1
02:47:29 MQTT: stat/sonoff/RESULT = {"POWER":"ON"}
02:47:29 MQTT: stat/sonoff/POWER = ON
02:47:29 APP: Multipress 1
02:47:29 Config: Saved configuration (1492 bytes) to flash at FB and count 242
02:47:29 APP: Multipress 1
02:47:30 MQTT: stat/sonoff/RESULT = {"POWER":"OFF"}
02:47:30 MQTT: stat/sonoff/POWER = OFF
02:47:30 APP: Multipress 1
02:47:30 Config: Saved configuration (1492 bytes) to flash at FB and count 243
02:47:31 APP: Multipress 1
02:47:31 MQTT: stat/sonoff/RESULT = {"POWER":"ON"}
02:47:31 MQTT: stat/sonoff/POWER = ON
02:47:31 APP: Multipress 1
02:47:31 Config: Saved configuration (1492 bytes) to flash at FB and count 244
02:47:31 APP: Multipress 2
02:47:32 MQTT: stat/sonoff/RESULT = {"POWER":"OFF"}
02:47:32 MQTT: stat/sonoff/POWER = OFF
02:47:32 Config: Saved configuration (1492 bytes) to flash at FB and count 245
```

mihalski, 2017-05-19:
And here I press the button once to turn it on and again to turn it off moments later:
```
02:49:00 APP: Multipress 1
02:49:00 APP: Multipress 1
02:49:01 MQTT: stat/sonoff/RESULT = {"POWER":"ON"}
02:49:01 MQTT: stat/sonoff/POWER = ON
02:49:01 APP: Multipress 1
02:49:01 APP: Multipress 1
02:49:01 Config: Saved configuration (1492 bytes) to flash at FB and count 246
02:49:02 MQTT: stat/sonoff/RESULT = {"POWER":"OFF"}
02:49:02 MQTT: stat/sonoff/POWER = OFF
02:49:02 APP: Multipress 1
02:49:02 Config: Saved configuration (1492 bytes) to flash at FB and count 247
02:49:02 APP: Multipress 1
02:49:03 MQTT: stat/sonoff/RESULT = {"POWER":"ON"}
02:49:03 MQTT: stat/sonoff/POWER = ON
02:49:03 APP: Multipress 1
02:49:03 Config: Saved configuration (1492 bytes) to flash at FB and count 248
02:49:03 APP: Multipress 1
02:49:04 MQTT: stat/sonoff/RESULT = {"POWER":"OFF"}
02:49:04 MQTT: stat/sonoff/POWER = OFF
02:49:04 APP: Multipress 1
02:49:04 Config: Saved configuration (1492 bytes) to flash at FB and count 249
02:49:04 APP: Multipress 1
02:49:05 MQTT: stat/sonoff/RESULT = {"POWER":"ON"}
02:49:05 MQTT: stat/sonoff/POWER = ON
02:49:05 APP: Multipress 1
02:49:05 Config: Saved configuration (1492 bytes) to flash at FB and count 250
02:49:06 MQTT: stat/sonoff/RESULT = {"POWER":"OFF"}
02:49:06 MQTT: stat/sonoff/POWER = OFF
02:49:06 APP: Multipress 1
02:49:06 APP: Multipress 1
02:49:06 Config: Saved configuration (1492 bytes) to flash at FB and count 251
02:49:07 MQTT: stat/sonoff/RESULT = {"POWER":"ON"}
02:49:07 MQTT: stat/sonoff/POWER = ON
02:49:07 APP: Multipress 1
02:49:07 Config: Saved configuration (1492 bytes) to flash at FB and count 252
02:49:07 APP: Multipress 1
02:49:07 Wifi: Checking connection...
02:49:07 Wifi: Connected
02:49:08 MQTT: stat/sonoff/RESULT = {"POWER":"OFF"}
02:49:08 MQTT: stat/sonoff/POWER = OFF
02:49:08 APP: Multipress 1
02:49:08 Config: Saved configuration (1492 bytes) to flash at FB and count 253
02:49:08 APP: Multipress 1
02:49:09 MQTT: stat/sonoff/RESULT = {"POWER":"ON"}
02:49:09 MQTT: stat/sonoff/POWER = ON
02:49:09 APP: Multipress 1
02:49:09 Config: Saved configuration (1492 bytes) to flash at FB and count 254
02:49:09 APP: Multipress 1
02:49:10 MQTT: stat/sonoff/RESULT = {"POWER":"OFF"}
02:49:10 MQTT: stat/sonoff/POWER = OFF
02:49:10 APP: Multipress 1
02:49:10 Config: Saved configuration (1492 bytes) to flash at FB and count 255
02:49:11 APP: Multipress 1
02:49:11 MQTT: stat/sonoff/RESULT = {"POWER":"ON"}
02:49:11 MQTT: stat/sonoff/POWER = ON
02:49:11 APP: Multipress 1
02:49:11 Config: Saved configuration (1492 bytes) to flash at FB and count 256
02:49:11 APP: Multipress 2
02:49:12 MQTT: stat/sonoff/RESULT = {"POWER":"OFF"}
02:49:12 MQTT: stat/sonoff/POWER = OFF
02:49:12 Config: Saved configuration (1492 bytes) to flash at FB and count 257
```

davidelang, 2017-05-19:
this seems to be showing that the button is being pressed.

mihalski, 2017-05-19:
Seems to behave normally when switched via the web interface:
```
02:51:20 HTTP: Handle root
02:51:26 MQTT: stat/sonoff/RESULT = {"POWER":"ON"}
02:51:26 MQTT: stat/sonoff/POWER = ON
02:51:26 Config: Saved configuration (1492 bytes) to flash at FB and count 258
02:51:27 Wifi: Checking connection...
02:51:27 Wifi: Connected
02:51:32 MQTT: stat/sonoff/RESULT = {"POWER":"OFF"}
02:51:32 MQTT: stat/sonoff/POWER = OFF
02:51:32 Config: Saved configuration (1492 bytes) to flash at FB and count 259
02:51:36 MQTT: stat/sonoff/RESULT = {"POWER":"ON"}
02:51:36 MQTT: stat/sonoff/POWER = ON
02:51:37 Config: Saved configuration (1492 bytes) to flash at FB and count 260
02:51:41 MQTT: stat/sonoff/RESULT = {"POWER":"OFF"}
02:51:41 MQTT: stat/sonoff/POWER = OFF
02:51:41 Config: Saved configuration (1492 bytes) to flash at FB and count 261
```

mihalski, 2017-05-19:
But I'm NOT pressing the button while that is happening.

mihalski, 2017-05-19:
2nd Sonoff RF. Only just flashed it right now so debug is off and it's not set to Sonoff RF yet. First boot:
```
00:00:00 Config: Use default configuration
00:0000:00:05 RSLT: stat/sonoff/RESULT = {"POWER":"ON"}
00:00:05 RSLT: stat/sonoff/POWER = ON
00:00:06 RSLT: stat/sonoff/RESULT = {"POWER":"OFF"}
00:00:06 RSLT: stat/sonoff/POWER = OFF
00:00:07 RSLT: stat/sonoff/RESULT = {"POWER":"ON"}
00:00:07 RSLT: stat/sonoff/POWER = ON
00:00:08 Wifi: Connected
00:00:08 mDNS: Initialized
00:00:08 HTTP: Webserver active on sonoff-7151.local with IP address 192.168.1.118
00:00:08 RSLT: stat/sonoff/RESULT = {"POWER":"OFF"}
00:00:08 RSLT: stat/sonoff/POWER = OFF
03:00:34 RSLT: stat/sonoff/RESULT = {"POWER":"ON"}
03:00:34 RSLT: stat/sonoff/POWER = ON
03:00:34 MQTT: Attempting connection...
03:00:35 mDNS: Query done with 1 mqtt services found
03:00:35 mDNS: Service found on homepi ip 192.168.1.64 port 1883
03:00:35 MQTT: Connected
03:00:35 MQTT: tele/sonoff/LWT = Online (retained)
03:00:35 MQTT: cmnd/sonoff/POWER = 
03:00:35 MQTT: tele/sonoff/INFO1 = {"Module":"Sonoff Basic", "Version":"5.1.1", "FallbackTopic":"DVES_F49BEF", "GroupTopic":"sonoffs"}
03:00:35 MQTT: tele/sonoff/INFO2 = {"WebserverMode":"Admin", "Hostname":"sonoff-7151", "IPaddress":"192.168.1.118"}
03:00:35 MQTT: tele/sonoff/INFO3 = {"Started":"Power on"}
03:00:36 MQTT: stat/sonoff/RESULT = {"POWER":"OFF"}
03:00:36 MQTT: stat/sonoff/POWER = OFF
03:00:36 MQTT: stat/sonoff/RESULT = {"POWER":"OFF"}
03:00:36 MQTT: stat/sonoff/POWER = OFF
03:00:37 MQTT: stat/sonoff/RESULT = {"POWER":"ON"}
03:00:37 MQTT: stat/sonoff/POWER = ON
03:00:38 MQTT: stat/sonoff/RESULT = {"POWER":"OFF"}
03:00:38 MQTT: stat/sonoff/POWER = OFF
03:00:39 MQTT: stat/sonoff/RESULT = {"POWER":"ON"}
03:00:39 MQTT: stat/sonoff/POWER = ON
03:00:40 MQTT: stat/sonoff/RESULT = {"POWER":"OFF"}
03:00:40 MQTT: stat/sonoff/POWER = OFF
03:00:41 MQTT: stat/sonoff/RESULT = {"POWER":"ON"}
03:00:41 MQTT: stat/sonoff/POWER = ON
03:00:42 MQTT: stat/sonoff/RESULT = {"POWER":"OFF"}
03:00:42 MQTT: stat/sonoff/POWER = OFF
03:00:43 MQTT: tele/sonoff/STATE = {"Time":"2017-05-19T03:00:43", "Uptime":0, "Vcc":3.000, "POWER":"OFF", "Wifi":{"AP":1, "SSID":"DobNet", "RSSI":76, "APMac":"80:2A:A8:8A:21:B3"}}
03:00:43 MQTT: stat/sonoff/RESULT = {"POWER":"ON"}
03:00:43 MQTT: stat/sonoff/POWER = ON
03:00:45 MQTT: stat/sonoff/RESULT = {"POWER":"OFF"}
03:00:45 MQTT: stat/sonoff/POWER = OFF
```

Here I set it to Sonoff RF, turn on debugging, press a button, discover it needs a reboot for the logging level to take, restart and DO NOT PRESS ANY BUTTONS. It goes into the cycle itself:
```
00:00:06 Wifi: Connected
00:00:06 mDNS: Initialized
00:00:06 HTTP: Webserver active on sonoff-7151.local with IP address 192.168.1.118
03:02:53 RSLT: tele/sonoff/UPTIME = {"Time":"2017-05-19T03:02:53", "Uptime":1}
03:02:53 MQTT: Attempting connection...
03:02:55 mDNS: Query done with 1 mqtt services found
03:02:55 mDNS: Service found on homepi ip 192.168.1.64 port 1883
03:02:55 MQTT: Connected
03:02:55 MQTT: tele/sonoff/LWT = Online (retained)
03:02:55 MQTT: cmnd/sonoff/POWER = 
03:02:55 MQTT: tele/sonoff/INFO1 = {"Module":"Sonoff Basic", "Version":"5.1.1", "FallbackTopic":"DVES_F49BEF", "GroupTopic":"sonoffs"}
03:02:55 MQTT: tele/sonoff/INFO2 = {"WebserverMode":"Admin", "Hostname":"sonoff-7151", "IPaddress":"192.168.1.118"}
03:02:55 MQTT: tele/sonoff/INFO3 = {"Started":"Power on"}
03:02:56 MQTT: stat/sonoff/RESULT = {"POWER":"ON"}
03:02:56 MQTT: stat/sonoff/POWER = ON
03:03:03 MQTT: tele/sonoff/STATE = {"Time":"2017-05-19T03:03:03", "Uptime":1, "Vcc":2.943, "POWER":"ON", "Wifi":{"AP":1, "SSID":"DobNet", "RSSI":88, "APMac":"80:2A:A8:8A:21:B3"}}
03:03:31 HTTP: Sonoff RF Module, GPIO1 0, GPIO3 0, GPIO4 0, GPIO14 0
03:03:32 APP: Restarting

 ets Jan  8 2013,rst cause:4, boot mode:(3,6)
[... 128 lines cut ...]
03:04:40 Config: Saved configuration (1492 bytes) to flash at FB and count 37
03:04:41 MQTT: stat/sonoff/RESULT = {"POWER":"ON"}
03:04:41 MQTT: stat/sonoff/POWER = ON
03:04:41 APP: Multipress 1
03:04:41 APP: Multipress 1
03:04:41 Config: Saved configuration (1492 bytes) to flash at FB and count 38
03:04:42 MQTT: stat/sonoff/RESULT = {"POWER":"OFF"}
03:04:42 MQTT: stat/sonoff/POWER = OFF
03:04:42 APP: Multipress 1
03:04:42 Config: Saved configuration (1492 bytes) to flash at FB and count 39
03:04:42 MQTT: stat/sonoff/RESULT = {"POWER":"ON"}
03:04:42 MQTT: stat/sonoff/POWER = ON
03:04:42 APP: Multipress 1
03:04:43 MQTT: tele/sonoff/STATE = {"Time":"2017-05-19T03:04:43", "Uptime":0, "Vcc":2.954, "POWER":"ON", "Wifi":{"AP":1, "SSID":"DobNet", "RSSI":86, "APMac":"80:2A:A8:8A:21:B3"}}
03:04:43 APP: Multipress 1
03:04:43 Config: Saved configuration (1492 bytes) to flash at FB and count 40
03:04:43 MQTT: stat/sonoff/RESULT = {"POWER":"OFF"}
03:04:43 MQTT: stat/sonoff/POWER = OFF
03:04:43 APP: Multipress 1
03:04:44 APP: Multipress 1
03:04:44 Config: Saved configuration (1492 bytes) to flash at FB and count 41
03:04:44 MQTT: stat/sonoff/RESULT = {"POWER":"ON"}
03:04:44 MQTT: stat/sonoff/POWER = ON
03:04:45 APP: Multipress 1
```

arendst, 2017-05-19:
Seeing the Multipress messages means the button is pressed all the time either by the stuck physical button on the case or the 434MHz receiver, which is tied to the same GPIO as the button is. 

As the 434MHz microcontroller is seperated from the Sonoff firmware I cannot control this behaviour. Try to unlearn the remote control button and try to learn another button again.

mihalski, 2017-05-19:
For the 2nd Sonoff RF I never paired any remote with it. and the case was not on it as I was programming it.

Could jumping R21 for the purposes of programming be causing this?

arendst, 2017-05-19:
I believe the jumper needs to be out after programming as documented in the wiki.

mihalski, 2017-05-19:
I don't see that written anywhere on the wiki and the link provided has the following from Phalox:
"The little wire stays. It doesn’t impact normal operation."

I'll try removing it tomorrow to see if it makes a difference.

arendst, 2017-05-19:
Just noticed that some moron screwed up the Sonoff RF wiki page. I'll try to restore the original one in a moment...

arendst, 2017-05-19:
Just restored page.

mihalski, 2017-05-19:
Thanks.

Vitally important information regarding the effect of having R21 shorted. Might try the jumper solution myself.

I'll update as to whether removing the short solves my problem.

mihalski, 2017-05-21:
Confirmed that R21 needs to be open circuit otherwise this problem occurs.

I'm not very impressed with the RF switch. Sometimes it ignores it for many presses at a time. Won't be getting more of these.

closed 2017-05-21 by mihalski