arendst/Tasmota #463: Latest version of sonoff basic does not enter flash mode

whhsw, 2017-05-20:
Just got 4 new sonoff basic (5 pin on the board). None of them can enter flash mode. Press program button and connect usb to computer, does not show any thing different to output console.

I also have some old sonoff 4CH. With same computer, same USB cable, same configuration, I can access flash mode on old devices. So I believe it's not the problem of cable or connection. Seems that there is some kind of protection on the newer firmware.

Any suggestion on what I should do next step for debug ? (Sorry I am a software engineer, newbie to hardware part..) 

Seen similar issues on other thread, related to different modules:
https://github.com/arendst/Sonoff-Tasmota/issues/453
https://github.com/arendst/Sonoff-Tasmota/issues/443

[image]
[image]

mihalski, 2017-05-20:
Ignore the pin further from the button. The wiki shows which pin is which. You must hold down the button as you insert the vcc pin (then you can let go). It should now be in flash mode.

whhsw, 2017-05-20:
yeap, I followed exactly what the wiki says. It does not work. 
This is the only thing serial port prints. It does not give out anymore info like the old 4ch does.

--- Miniterm on /dev/cu.usbserial  115200,8,N,1 ---
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---

This is the upload error.
Uploading .pioenvs/sonoff/firmware.bin
warning: espcomm_sync failed
error: espcomm_open failed
error: espcomm_upload_mem failed
*** [upload] Error 255

arendst, 2017-05-20:
Try to execute ``pio run -v -t upload`` from a PlatformIO serial console and analyse the output.

gdave321, 2017-05-20:
Got 2 new sonoff-basics looking exactly the same as yours (including the hardware mod, extra component soldered between D5 and C?). They flashed without issues.

Did have another problem. After successfully flashing 5.0.7 unit didn't give any sign of life anymore. Re-flashing didn't help.  After flashing a 4.0.3 backup image it starting working and then did an upgrade via web to 5.0.7 without further issues. YMMV

whhsw, 2017-05-20:
@arendst 
Thanks for guiding! Here is the the output. Is there anything helpful?

[skipped some complication message].
Auto-detected: /dev/cu.usbserial
esptool -vv -cd ck -cb 115200 -cp "/dev/cu.usbserial" -cf .pioenvs/sonoff/firmware.bin
esptool v0.4.9 - (c) 2014 Ch. Klippel <ck@atelier-klippel.de>
setting board to ck
setting baudrate from 115200 to 115200
setting port from /dev/tty.usbserial to /dev/cu.usbserial
espcomm_upload_file
espcomm_upload_mem

opening port /dev/cu.usbserial at 115200
tcgetattr
tcsetattr
serial open
opening bootloader
resetting board
trying to connect
espcomm_send_command: sending command header
espcomm_send_command: sending command payload
trying to connect
espcomm_send_command: sending command header
espcomm_send_command: sending command payload
trying to connect
espcomm_send_command: sending command header
espcomm_send_command: sending command payload
resetting board
trying to connect
espcomm_send_command: sending command header
espcomm_send_command: sending command payload
trying to connect
espcomm_send_command: sending command header
espcomm_send_command: sending command payload
trying to connect
espcomm_send_command: sending command header
espcomm_send_command: sending command payload
resetting board
trying to connect
espcomm_send_command: sending command header
espcomm_send_command: sending command payload
trying to connect
espcomm_send_command: sending command header
espcomm_send_command: sending command payload
trying to connect
espcomm_send_command: sending command header
espcomm_send_command: sending command payload
warning: espcomm_sync failed
error: espcomm_open failed
error: espcomm_upload_mem failed
*** [upload] Error 255

whhsw, 2017-05-20:
@gdave321 did your device print out anything in serial console before the flash? These ones do not print anything out..  What's the date print on the warranty label on your devices? Mine is marked as 2017/03/15 (the only date I can find out of the device).   

[image]

DarkAllMan, 2017-05-22:
I have a very similar issue. My date is 2017/3/10. I cannot flash it in any way. No terminal output.
And all flash attempts result in:
warning: espcomm_sync failed
error: espcomm_open failed
error: espcomm_upload_mem failed

mihalski, 2017-05-22:
Have you tried swapping around the TX/RX connections?

DarkAllMan, 2017-05-22:
I have tried that yes.....

whhsw, 2017-05-22:
I made some progress! Though still have no idea of why, and the way to success is not reproducible.

Out of the 4, I randomly got two worked...  A rough sequence of what I did is:  swap TX/RX, 
so the connection is TX-TX, RX-RX. hold the power button, plug in usb cable. Keep holding power button all the way, do not release. And then out of many times of try, it just uploaded !!!!

Highly suspect this is related to some kind of timing issue (sth. like input any key within this N seconds, otherwise it will reboot normally.). Since even after the succeed upload, I can't reproduce with the same device twice.  

Also another thing I noticed is, green light does not flash everytime it's plugged in. So would it be possible that this new model need 5v power supply ? Don't want to risk burning it...

DarkAllMan, 2017-05-22:
Trying..... but I am not not able to produce the same result as you did....

settenke, 2017-05-23:
The **reverse of RX / TX has solved the problem.** (Sonoff S20)
THX!

DarkAllMan, 2017-05-25:
Still not able to flash it......... no idea on how to solve it....

whhsw, 2017-05-25:
same here… I have 2 more left that still with no way of flashing..

If anyone could help pointed out which one is GPIO 0, wiring GPIO 0 with
GND might work?

DarkAllMan, 2017-06-06:
Is there an update to this issue?

khcnz, 2017-06-06:
Not sure if they have perhaps made a change.. but on my sonoff Dev I have to hold the flash button down until flashing starts (well actually I think there is a 2second window - but easiest to just hold the button down until flashing starts). 

Worth a try?

DarkAllMan, 2017-06-07:
I did try, but nope...nothing...

timothy-gibson, 2017-06-11:
I had the same issue with S20. I had to swap RX/TX, and then just kept trying to flash it following @whhsw's directions and eventually it uploaded. Appears to be some sort of timing issue with the new model?

kristoffernolgren, 2017-06-22:
The people who succeeded, what baud-rate were you using? Having the same issue with a new sonoff.

mihalski, 2017-06-22:
I have 4 basic and 4 Pow on the way. So I'll let you know.

kristoffernolgren, 2017-06-23:
@mihalski thanks, I just realized that my reseller didn't send me a basic, but rather a sonoff RF, it seems to have different demands when it comes to booting it, so I might have a different problem, (but this info will probably be useful for other people)

mihalski, 2017-06-23:
Yes. I got 2 RF units thinking they would be better.. But in fact they're a much bigger pain in the ass.

kristoffernolgren, 2017-06-23:
Have you managed to flash them? I tried shorting the resistor while holding down the button(I think it was R21, can check later) but it doesn't seem to go into programming mode. How can you tell if it is? Is the led flashing etc?

DarkAllMan, 2017-06-23:
Where can you see the difference between the basic en the RF on the board?

mihalski, 2017-06-23:
I used s dab of solder to short. Then I discovered I needed to take it off to use the devices :P wasn't too hard to remove.

But yeah.. I programmed and am using them. Was a lot more finicky than standard Sonoff/Pow/TH.

You can tell when it starts programming :P

kristoffernolgren, 2017-06-23:
@mihalski I'm trying to remove error-factors, like my FTDI, the script i'm using to program, connecting rx/tx etc, when I try to start the sonoff in programmin mode, it still connects to the wifi for example, is it suppose to do that? I just ordered a couple of new non-rf sonoffs to see if everything else works as it should...

@DarkAllMan The RF has an additional circuit attatched with an antenna, you can see it in the lower left corner on this image, upper photo, should be pretty easy to spot: https://www.itead.cc/wiki/images/thumb/d/d8/Sonoff_rf_parts.jpg/400px-Sonoff_rf_parts.jpg

mihalski, 2017-06-23:
No it should not connect o wifi in programming mode.
Just use Arduino IDE with your FTDI. We all know that this should work and it has always eventually worked for me.

IF you try 5 times and it fails.. swap tx/rx and try again.. Even when I think I'm bashing my head against the wall with these things it eventually works.. The RF units were the first to really piss me off though.

DarkAllMan, 2017-06-23:
Then mine is NOT an RF version :)
Thx. Also, I think mine is dead now….. Ordered a new one….

kristoffernolgren, 2017-06-23:
@mihalski thx! And what about the LED, should it blink, be dead etc? Can you put it in programming mode even if rx/tx is the wrong way around? Is there something else that is required for triggering programming

 I have tried to boot it with mulitple firmware, including Tasmota and espEasy, using arduino and esptool.py (osx) but it seems I can't get it to boot correctly so the IDE doesn't really seem to be the problem, atleast not yet...

mihalski, 2017-06-23:
I'll answer some of those questions when my new units arrive.. But..

Yes.. You don't have to be connected at all to put it into programming mode.. It's just that you can't program it unless it's in programming mode.. I honestly can't remember what the lights to.. It might be worthwhile looking at youtube for this.. Let me see if I can find anything useful..

Booting doesn't sound like your problem. It's programming.. It's ABUNDANTLY clear when programming is working.. So there is never any doubt.. You can see it programming in the Arduino IDE.

mihalski, 2017-06-23:
See if this helps.. https://youtu.be/6fWFnxh0EiQ?t=4m9s

kristoffernolgren, 2017-06-23:
Yeah, I seem to be failing at getting in to programming mode. He says "without programming cable and pushbutton we can start programming the device." The pushbutton is the one that sets gpio0 to zero and the programming cable is the cable used to connect the sonoff to Ftdi, he mus mean with, not without right?

mihalski, 2017-06-23:
with or cannot :P

paggard, 2017-06-30:
Just got my new sonoff wifi switches - spent about three hours and failed to put any of them to flash mode.. If anyone had any luck with that - please share :)

mihalski, 2017-06-30:
I got my 4 power/4 standard today. Won't be able to test till Sunday AEST. Will let you know.

kristoffernolgren, 2017-06-30:
@paggard what version do you have?

paggard, 2017-06-30:
@kristoffernolgren do not have them at hand now - will post exact info later... but they are very recent - bought them 10 days ago directly from ITEAD.

kristoffernolgren, 2017-06-30:
@paggard allright, but is it an RF or a basic etc?

paggard, 2017-06-30:
@kristoffernolgren - mine are not RF versions.

paggard, 2017-07-01:
@kristoffernolgren here is my board. The seal on the enclosure shows '2017 3 15'
[image]

DarkAllMan, 2017-07-04:
Received a new one with date 2017-3-15
Unable to enter flash mode.... still trying multiple times, switching RX/TX, holding pin untill upload... etc etc..
But nothing seems to be working.......

mihalski, 2017-07-04:
I will be doing this tomorrow. I'll let you know.

mihalski, 2017-07-05:
I've tried to flash a few dozen times now without success.
I've never had this much trouble before, so perhaps they have changed something.

Lennyz1988, 2017-07-06:
Itead uploaded a new schematic of the board in march, so chances are they changed something. 

https://www.itead.cc/wiki/File:Sonoff-Schematic.pdf

You should still be able to get it into flash mode by connecting GPIO0 to Ground before powering on the device.

paggard, 2017-07-06:
@Lennyz1988 I've already checked the board - the button is for sure connecting Ground and GPIO0...

mihalski, 2017-07-06:
For what it's worth you can cash the firmware OTA without needing an FTDI now. I just got it to work.. hope to create a docker container that can do it for you. I think you will at LEAST need wifi so that you can join the ITEAD access point that a Sonoff provides.

kristoffernolgren, 2017-07-06:
@mihalski that sounds way better!  Do you have any docs etc? How did you figure it out?

mihalski, 2017-07-06:
I'll try to write something basic up tonight. Also would like to create a docker container to automate it, but that's a steep learning curve as I've never built a docker container before :P

kristoffernolgren, 2017-07-07:
Cool and very interesting afaik, it wasn't possible before. Has sonoff done an update or something.

Btw, I just got 5 sonoff basic models, I could put them into programmer mode without problem.  I did not manage to flash them but got stuck half-way through. I think it's due to a crappy FTDI-module. Some people say it worked for them to change and I had a china-clone, so have ordered a new one.

Lennyz1988, 2017-07-07:
@paggard Then people should be able to flash. There is not a single ESP8266 that doesn't get into program mode with GPIO0 pulled to ground.

mihalski, 2017-07-07:
Well this is a pain. I was about to start working on some documentation when I got home from work and what happens? the most conspicuous Sonoff Pow in the house dies on me. No TV. No lights in the living area. First one to die on me.

Pro tip.. Backup your configs..

mihalski, 2017-07-07:
For what it's worth I just programmed the Sonoff Pow via FTDI on my first attempt.

Running into a problem doing it the OTA way.

tinstep, 2017-07-14:
I got 4 basic models, tried flashing 2 of them. all dated 2017-03-15. used two separate FTDI boards, (purchased a $20 one from Jaycar!!! and a FTDI basic 3.3 from Sparkfun) swapped RX, TX over many times, no luck - can flash nodeMCU without issue.

mihalski, 2017-07-14:
Search for SonOTA on Github.. It works.

chimeranzl, 2017-07-31:
This maybe the issue for those with the RF version.  I've swapped the resistor around but still have not had any luck. https://tech.scargill.net/even-more-sonoff/

naticklamb, 2017-08-05:
esptool.py ... write_flash --flash_mode dout ...  fobar.bin  ?

Maybe just need DOUT?  Worked for m.

JSCSJSCS, 2017-08-10:
I have just received two Sonoff Basics. I used arduino to upload basic blink program to blink pin 13 just as a test, with some serial output to test if it is working. It seemed to upload fine with Arduino with no errors, but it did not blink and there was no serial output from the loop.

tinstep, 2017-08-12:
So briked 2 sonoffs trying to do OTA, not Sonota fault, I made some guess as to what to do because there is no doco and got things a bit mixed up, I have posted over there, but back here now, has anyone been able to flash using ftdi on a 2017-03-15 sonoff basic?  Or even managed to get that version in to flash mode?

gnkarn, 2017-08-15:
same as JSCSJSCS  , can flash any sketch normally , but then it aint work ,
i can add , that there is output on the serial though , the problem i  that you may not see it , as the Baudrate is auto set to 74880 , and then you may see this output

ets Jan  8 2013,rst cause:1, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v3de0c112
~ld
Fatal exception (0): 
epc1=0x4021c1d4, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000000, depc=0x00000000

emanuelefanton, 2017-08-16:
I solved the problem with the sonoff basic received august  2017  setting on arduinoide the model as ESP8285. Bye

[160 more replies]

closed 2018-08-18 by stale[bot]