arendst/Tasmota #466: Not able to upload to Sonoff POW. Not booting into normal mode either

samuel235, 2017-05-20:
As the title states, i'm unable to flash Tasmota to my Sonoff POWs. I haven't attempted to upload any other variety of firmware though. The below extract is from the logging from the Arduino IDE as it attempts to upload.

`esptool v0.4.9 - (c) 2014 Ch. Klippel <ck@atelier-klippel.de>
	setting board to ck
	setting baudrate from 115200 to 115200
	setting port from COM1 to COM4
	setting address from 0x00000000 to 0x00000000
	espcomm_upload_file
	espcomm_upload_mem
	setting serial port timeouts to 1000 ms
opening bootloader
resetting board
trying to connect
	flush start
	setting serial port timeouts to 1 ms
	setting serial port timeouts to 1000 ms
	flush complete
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
	read 0, requested 1
trying to connect
	flush start
	setting serial port timeouts to 1 ms
	setting serial port timeouts to 1000 ms
	flush complete
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
	espcomm_send_command: receiving 2 bytes of data
	espcomm_send_command: receiving 2 bytes of data
	espcomm_send_command: receiving 2 bytes of data
	espcomm_send_command: receiving 2 bytes of data
	espcomm_send_command: receiving 2 bytes of data
	espcomm_send_command: receiving 2 bytes of data
	espcomm_send_command: receiving 2 bytes of data
	espcomm_send_command: receiving 2 bytes of data
Uploading 467712 bytes from C:\Users\Samuel\AppData\Local\Temp\arduino_build_672329/sonoff.ino.bin to flash at 0x00000000
	erasing flash
	size: 072300 address: 000000
	first_sector_index: 0
	total_sector_count: 115
	head_sector_count: 16
	adjusted_sector_count: 99
	erase_size: 063000
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
	setting serial port timeouts to 15000 ms
	setting serial port timeouts to 1000 ms
	espcomm_send_command: receiving 2 bytes of data
	writing flash
................................................................................ [ 17% ]
................................................................................ [ 35% ]
................................................................................ [ 52% ]
................................................................................ [ 70% ]
................................................................................ [ 87% ]
.........................................................                        [ 100% ]
starting app without reboot
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
	espcomm_send_command: receiving 2 bytes of data
closing bootloader
	flush start
	setting serial port timeouts to 1 ms
	setting serial port timeouts to 1000 ms
	flush complete`

I have successfully wiped the flash and uploaded again, nothing. When I claim that it is not uploading, i'm saying that because i get no confirmation and nothing in the serial monitor when the IDE is complete. 

If you would like any more information from me, please ask.

TRIROG, 2017-05-24:
Try opening a serial port after flashing to verify the tasmota firmware starts successfully.
If you get only garbage output at 115200 baud it is very likely a bad chip.
You can then confirm this by switching baud to 76800 (i think) and then reset the chip to see what the boot output is.

Upon flashing in serial at 115200 baud i got

Fatal exception (0): 
epc1=0x40217e08, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000000, depc=0x00000000

at 74880 baud i get

 ets Jan  8 2013,rst cause:1, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0xef
csum 0xef
csum err
ets_main.c

I replaced the chip and now it works.

samuel235, 2017-05-26:
I've managed to get one to flash properly now, it seems to be a possible power supply issue or even a bad set of ESP's. Sometimes i manage to see a serial output (obviously when the flash was successful) but then other times, nothing. Thank you for the suggestion, next time i face an issue I will be sure to refer back to these codes you're facing to recognise a possible bad chip.

closed 2017-05-26 by samuel235