arendst/Tasmota #468: Problem to switch Sonoff with Moquitto Broker: Does not toggle

canis-ferrin, 2017-05-21:
Tasmota Version:	5.1.3
Only changes from downloaded Version in user_config.h:
#define WIFI_IP_ADDRESS
#define WIFI_GATEWAY
#define WIFI_DNS
#define STA_SSID1
#define STA_PASS1
#define MQTT_HOST            "192.168.211.199"
#define MQTT_USER            "username"
#define MQTT_PASS            "password"

Test with mqtt-spy: mqtt-spy-0.5.4-jar-with-dependencies.jar
Topic is:	cmnd/DVES_3C1A6E
Data is:	Power ON
The Sonoff flashes 1 time shortly
Output is:	stat/sonoff/RESULT
Data is:	{"Command":"Unknown"}

Topic	cmnd/sonoff	is also working, the device flashes shortly
Wahat is wrong? I tried ervything without success.
Security if off in the moment, and this Broker has only this Client:
The MQTT Broker is reachable: www.wauzinet.linkpc.net:1883 by mqtt-spy.
The Sonoff web server is reachable under http://wauzinet.linkpc.net:1884
I tried also issue 340 without success.
Can anyone help? Thanks

davidelang, 2017-05-22:
see the troubleshooting page in the wiki, get more detailed logs and see exactly 
what's happening.

canis-ferrin, 2017-05-22:
I put
#define SERIAL_LOG_LEVEL       LOG_LEVEL_DEBUG    // [SerialLog]
#define WEB_LOG_LEVEL          LOG_LEVEL_DEBUG    // [WebLog]
in user_config.h and send a fresh weblog:

00:00:01 Wifi: Attempting connection...
00:00:02 Wifi: Attempting connection...
00:00:03 Wifi: Connected
00:00:03 mDNS: Initialized
00:00:03 HTTP: Webserver active on sonoff-6766.local with IP address 192.168.211.198
00:00:03 RTC: (UTC) Mon May 22 05:51:22 2017
00:00:03 RTC: (DST) Sun Mar 26 02:00:00 2017
00:00:03 RTC: (STD) Sun Oct 29 03:00:00 2017
06:51:23 HTTP: Handle root
06:51:25 MQTT: Attempting connection...
06:51:26 mDNS: Query done with 0 mqtt services found
06:51:26 MQTT: Connected
06:51:26 MQTT: tele/sonoff/LWT = Online (retained)
06:51:26 MQTT: cmnd/sonoff/POWER = 
06:51:26 MQTT: tele/sonoff/INFO1 = {"Module":"Sonoff Basic", "Version":"5.1.3", "FallbackTopic":"DVES_3C1A6E", "GroupTopic":"sonoffs"}
06:51:26 MQTT: tele/sonoff/INFO2 = {"WebserverMode":"Admin", "Hostname":"sonoff-6766", "IPaddress":"192.168.211.198"}
06:51:26 MQTT: tele/sonoff/INFO3 = {"Started":"Software/System restart"}
06:51:26 HTTP: Handle console
06:51:27 RSLT: DataCb Group 0, Index 1, Type DVES_3C1A6E, Data Power toggle
Power ON
Power toggle (POWER TOGGLE
POWER ON
POWER TOGGLE)
06:51:27 MQTT: stat/sonoff/RESULT = {"Command":"Unknown"}
06:51:28 MQTT: stat/sonoff/RESULT = {"POWER":"OFF"}
06:51:28 MQTT: stat/sonoff/POWER = OFF
06:51:35 MQTT: tele/sonoff/STATE = {"Time":"2017-05-22T06:51:35", "Uptime":0, "Vcc":3.009, "POWER":"OFF", "Wifi":{"AP":1, "SSID":"PapaNet-E", "RSSI":56, "APMac":"00:13:10:15:25:61"}}
06:53:51 RSLT: DataCb Group 0, Index 1, Type DVES_3C1A6E, Data Power on (POWER ON)
06:53:51 MQTT: stat/sonoff/RESULT = {"Command":"Unknown"}
06:54:39 RSLT: DataCb Group 0, Index 1, Type DVES_3C1A6E, Data Power off (POWER OFF)
06:54:39 MQTT: stat/sonoff/RESULT = {"Command":"Unknown"}

This in the user_config.h:

canis-ferrin, 2017-05-22:
This is the user_config.h:
/*
  user_config.h - user specific configuration for Sonoff-Tasmota

  Copyright (C) 2017  Theo Arends

  This program is free software: you can redistribute it and/or modify
  it under the terms of the GNU General Public License as published by
  the Free Software Foundation, either version 3 of the License, or
  (at your option) any later version.

  This program is distributed in the hope that it will be useful,
  but WITHOUT ANY WARRANTY; without even the implied warranty of
  MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
  GNU General Public License for more details.

  You should have received a copy of the GNU General Public License
  along with this program.  If not, see <http://www.gnu.org/licenses/>.
*/

/*********************************************************************************************\
 * ATTENTION: Changes to most PARAMETER defines will only override flash settings if you change
 *            define CFG_HOLDER.
 *            Most parameters can be changed online using commands via MQTT, WebConsole or serial
 *
 * Corresponding MQTT/Serial/Console commands in [brackets]
\*********************************************************************************************/

// -- Project -------------------------------------
#define PROJECT                "sonoff"          // PROJECT is used as the default topic delimiter and OTA file name
                                                 //   As an IDE restriction it needs to be the same as the main .ino file

#define CFG_HOLDER             0x20161209        // [Reset 1] Change this value to load following default configuration parameters
#define SAVE_DATA              1                 // [SaveData] Save changed parameters to Flash (0 = disable, 1 - 3600 seconds)
#define SAVE_STATE             1                 // [SaveState] Save changed power state to Flash (0 = disable, 1 = enable)

// -- Wifi ----------------------------------------
#define WIFI_IP_ADDRESS        "192.168.211.198"         // [IpAddress1] Set to 0.0.0.0 for using DHCP or IP address
#define WIFI_GATEWAY           "192.168.211.254"   // {IpAddress2] If not using DHCP set Gateway IP address
#define WIFI_SUBNETMASK        "255.255.255.0"   // [IpAddress3] If not using DHCP set Network mask
#define WIFI_DNS               "192.168.211.254"    // [IpAddress4] If not using DHCP set DNS IP address (might be equal to WIFI_GATEWAY)

#define STA_SSID1              "PapaNet-E"      // [Ssid1] Wifi SSID
#define STA_PASS1              "ausgext"  // [Password1] Wifi password
#define STA_SSID2              "indebuurt2"      // [Ssid2] Optional alternate AP Wifi SSID
#define STA_PASS2              "VnsqrtnrsddbrN"  // [Password2] Optional alternate AP Wifi password
#define WIFI_CONFIG_TOOL       WIFI_WPSCONFIG    // [WifiConfig] Default tool if wifi fails to connect
                                                 //   (WIFI_RESTART, WIFI_SMARTCONFIG, WIFI_MANAGER, WIFI_WPSCONFIG, WIFI_RETRY)
                                                 
// -- Syslog --------------------------------------
#define SYS_LOG_HOST           "domus1"          // [LogHost] (Linux) syslog host
#define SYS_LOG_PORT           514               // [LogPort] default syslog UDP port
#define SYS_LOG_LEVEL          LOG_LEVEL_NONE    // [SysLog]
#define SERIAL_LOG_LEVEL       LOG_LEVEL_DEBUG    // [SerialLog]
#define WEB_LOG_LEVEL          LOG_LEVEL_DEBUG    // [WebLog]

// -- Ota -----------------------------------------
#define OTA_URL                "http://domus1:80/api/arduino/" PROJECT ".ino.bin"  // [OtaUrl]

// -- MQTT ----------------------------------------
#define MQTT_USE               1                 // [Mqtt] Select default MQTT use (0 = Off, 1 = On)
// !!! TLS uses a LOT OF MEMORY (20k) so be careful to enable other options at the same time !!!
//#define USE_MQTT_TLS                             // EXPERIMENTAL Use TLS for MQTT connection (+53k code, +20k mem) - Disable by //
                                                 //   Needs Fingerprint, TLS Port, UserId and Password
#ifdef USE_MQTT_TLS
  #define MQTT_HOST            "m20.cloudmqtt.com"  // [MqttHost]
  #define MQTT_FINGERPRINT     "A5 02 FF 13 99 9F 8B 39 8E F1 83 4F 11 23 65 0B 32 36 FC 07"  // [MqttFingerprint]
  #define MQTT_PORT            20123                // [MqttPort] MQTT TLS port
  #define MQTT_USER            "cloudmqttuser"      // [MqttUser] Mandatory user
  #define MQTT_PASS            "cloudmqttpassword"  // [MqttPassword] Mandatory password
[... 96 lines cut ...]
#define USE_WS2812                               // WS2812 Led string using library NeoPixelBus (+8k code, +1k mem) - Disable by //
  #define USE_WS2812_CTYPE     1                 // WS2812 Color type (0 - RGB, 1 - GRB)
//  #define USE_WS2812_DMA                         // DMA supports only GPIO03 (= Serial RXD) (+1k mem)
                                                 //   When USE_WS2812_DMA is enabled expect Exceptions on Pow

/*********************************************************************************************\
 * Compile a minimal version if upgrade memory gets tight ONLY TO BE USED FOR UPGRADE STEP 1!
 *   To be used as step 1 during upgrade. 
 *   Step 2 is re-compile with option BE_MINIMAL commented out.
 *   !!! Needed for next release of Arduino/ESP8266 (+22k code, +2k mem) !!!
\*********************************************************************************************/

[... cut ...]
/*********************************************************************************************\
 * No user configurable items below
\*********************************************************************************************/

#if defined(USE_MQTT_TLS) && defined(USE_WEBSERVER)
  #error "Select either USE_MQTT_TLS or USE_WEBSERVER as there is just not enough memory to play with"
#endif

#if (ARDUINO < 10610)
  #error "This software is supported with Arduino IDE starting from 1.6.10 and ESP8266 Release 2.3.0"
#endif

davidelang, 2017-05-22:
see the troubleshooting page about why changes to user_config.h may not take 
effect.

do you have more than one relay configured? if so, you would need to do power1 
instead of power to switch the first relay.

What happens if you enter the command directly in the web console? do you still 
get invalid command?

canis-ferrin, 2017-05-22:
http://192.168.211.198/cm?&cmnd=Power%20Toggle is working and delivers:
08:53:15 RSLT: DataCb Group 0, Index 1, Type POWER, Data Toggle (TOGGLE)
RESULT = {"POWER":"ON"}
POWER = ON

canis-ferrin, 2017-05-22:
http://wauzinet.linkpc.net:1884/cm?&cmnd=Power%202 toggles the Poer
with mqtt-spy the command Power 2 results in short flash on the sonoff.
I have 7 Sonoss- switches installed, acting as HUE- Bridge for an Echo Dot
and i try to build up an OpenHAB2 system.

canis-ferrin, 2017-05-22:
After each flasing, i held the button more than 4 seconds, waited, and restarted by power supply that changes took affect,

canis-ferrin, 2017-05-22:
in
https://community.home-assistant.io/t/sonoff-with-embedded-mqtt/12526
I found a hint: 
You can control it with
Topic	cmnd/sonoff/POWER
DATA	ON | OFF | TOGGLE
It could be a good idea to write examples in the wiki

closed 2017-05-22 by canis-ferrin