arendst/Tasmota #469: Dimmer

macpit, 2017-05-21:
Hi arendt,

Great Project and many thank for your work! I love it and it's work perfect.
I know that is not project related but maybe you can help me.

I search a W-iFi dimmer for 230V/~400W for pulps. Sonoff is veyr cool but there is no dimmer :-(
Maybe you have a tip for me?

Regards,

Marc

davidelang, 2017-05-21:
I have not been able to find a dimmer that uses esp-8266 yet.
k

nosaj66au, 2017-05-23:
I'm currently working on a renovation and also required 240Vac dimmers. Im running trials using a push button dimmer "SAL Push Button Dimmer 350W Master Switch SDD350TCM" which has a remote switch option SDD350TCS. The remote/slave push button is just 2off, 10K resistors in series and a tactile switch. This I have replaced with a sonoff and 2 off 10K resistors. I can turn the light on off and dim. I'm currently look at way to set dim level using a sonoff POW to measure power consumption or Current but it's a bit flaky at the moment. Work is still on going.
Jaosn

Blue407, 2017-07-03:
I'm also looking for the same thing. The ability to dim 230v GU10 LED bulbs. (They are dimmable type)

jbarneson, 2017-07-03:
+1   I'm hopeful sonoff releases one soon

Amli2011, 2017-09-09:
Wishing for something like this:
http://www.apogeekits.com/advanced-electronic-kits/dc-light-dimmer-k8064.htm
Together with pwm out of a tasmota software in a sonoff package with CE and terminals for a wall switch, or even better, built into a sonoff touch panel. 
Could we all tip sonoff/itead about it? Maybe they will make it if the interest is there?

klaernie, 2017-09-29:
Has anyone thought about replacing the regular relay by a solid state relay? I think one of these fostek SSR should be fine with running smaller resistive loads in pwm. Could the esp8266 run pwm on the output steering the relay?

hesich, 2017-10-01:
Only in German, but google can translate
https://ex-store.de/2-Kanal-RS232-Dimmer-Modul-V4-fuer-Unterputzmontage-230V-3A
they have also the relay version (currently sold out)
https://ex-store.de/ESP8266-WiFi-Relay-V31

Martinvdm, 2018-01-14:
Is there still no dimmer possible with tasmota and mqtt?

davidelang, 2018-01-14:
do you have a module that has dimmer capabilities (instead of the relays that 
almost all of them have)?

Martinvdm, 2018-01-14:
https://github.com/Theb-1/ESP8266-wifi-light-dimmer

davidelang, 2018-01-15:
I was hoping that there was a commercial one available :-(

trying to do that sort of very precise timing on a system that is doing a lot of 
other things (network I/O, etc) is going to be quite a problem

On Sun, 14 Jan 2018, Martinvdm wrote:

> https://github.com/Theb-1/ESP8266-wifi-light-dimmer

jonas59075, 2018-01-15:
I have just ordered this one:

https://www.aliexpress.com/snapshot/0.html?spm=a2g0s.9042647.6.2.iGlAYq&orderId=88957336184289&productId=32562891943 

It will arrive in 10-15 Days.

Perhaps it is possible to modify the tasmota firmware to get it work.

bensuffolk, 2018-01-15:
Hi @jonas59075 that link does not work (I think its a copy of your order so only works when your logged in). 

Was it this you were linking to? [https://www.aliexpress.com/item/Best-Sales-Wallpad-White-Glass-LED-110-250V-EU-Phone-Wifi-Wireless-Remote-Controlled-Power-Dimmer/32562891943.html](https://www.aliexpress.com/item/Best-Sales-Wallpad-White-Glass-LED-110-250V-EU-Phone-Wifi-Wireless-Remote-Controlled-Power-Dimmer/32562891943.html)

Ben

jonas59075, 2018-01-15:
Oh, sorry for that. Yes ... your Link is the right one ...

Martinvdm, 2018-01-15:
This looks promising:

http://www.instructables.com/id/ARMTRONIX-WIFI-SINGLE-Dimmer-Board-V02/

madjam002, 2018-01-19:
A dimmer would still be useful for some applications. I’d like to wire up my salt lamp which has a manual dimmer at the moment, and the bulbs for a salt lamp cannot be LED.

jonathanschneider, 2018-02-15:
> A dimmer would still be useful for some applications. I’d like to wire up my salt lamp which has a manual dimmer at the moment, and the bulbs for a salt lamp cannot be LED.

The same goes for Edison light bulbs. 

> This looks promising:
> 
> http://www.instructables.com/id/ARMTRONIX-WIFI-SINGLE-Dimmer-Board-V02/

I happened to order this before stumbling upon this thread. I'll let you know about it once I received it.

checksummaster, 2018-04-06:
This can be the solution for real dimmer (not led solution), I will give a shot
https://www.banggood.com/3Pcs-RobotDyn-AC-Light-Dimmer-Module-For-PWM-Control-1-Channel-3_3V5V-Logic-AC-50hz-60hz-220V-110V-p-1255783.html?rmmds=myorder&cur_warehouse=CN

nosaj66au, 2018-04-08:
Agree let me know how it goes.

stanvv, 2018-04-13:
Any updates? I just ordered this one, since the seller told me there's an ESP8266 inside.

https://www.aliexpress.com/item/WiFi-LED-Dimmer-Switch-220V-110V-Dimming-Panel-Switch-Connected-To-Alexa-Google-Home-Voice-Control/32859257784.html?spm=2114

reeso3000, 2018-04-23:
I ordered one like Stanvv and it arrived today. This is what it looks like inside if anyone is interested. 

[image]
[image]
[image]

qingz2004, 2018-04-26:
Does it work without changing the firmware?

reeso3000, 2018-04-30:
Yes it works without changing the firmware - Uses Smart Life app that is compatible with Alexa. The results are a bit mixed - it dims ok but the lights seem to flicker, I purchased dimmable leds (Gu10) to use in my kitchen and this is the result. Going to try it in my living room with B22 bulb to see if its any bettter.

qingz2004, 2018-04-30:
I bought the same one from Aliexpress, but a US version.
[](http://www.aliexpress.com/snapshot/null.html?spm=a2g0s.issue_f040t.0.0.25194c4dSr9huA&orderId=91372679396827)
It did not response or dim to the +/- touch button on the plate, but it did (some how, but not smooth) with Smart Life app. The bulb flickers a lot once dimmed. I tried with both dimmable LED and incandescent bulb.
Even worse, after I tried a few times with phone app, the dimmer is completely dead. No response to anything. Only power cycle can fix it.
I believe the MCU STM8S103F3P6 does not work or is not properly programmed, since all the touch button signals and the cross zero interrupt signal go to the MCU, and the MCU generates the PWM.
I checked with a oscilloscope, the signals goes into MCU are perfect, but the PWM generated from the MCU is not.
ESP8266 here is to communicate with the phone, and transmit the command to the MCU through the serial port.
We might be able to bypass the MCU and use the ESP8266 to do everything.

epicality85, 2018-05-07:
I have the same one as @reeso3000. Im looking into modifying :D let us know if solve the issue.

stanvv, 2018-05-11:
Mine works fine with the standard app, but I'm trying to flash Tasmota. No success so far:

Got this schema of the ESP8266 by Hysiry, which is placed in the dimmer: https://fccid.io/2AKBPESP8266/User-Manual/User-Manual-3192780
I assume the schema at page 6 (3. Pin description) is flipped, see also page 13 (7. Minimum System Requirements).
I think I got it in programming mode by connecting ground with IO0, cause the left dimmer button lights up, while it doesn't light up when only connecting the programmer. 
I soldered 4 pins to the board as follows. When switching 3v3 and ground it didn't power up, so I assume that is connected correctly. 
[image]

But I get the 'standard' error:

> warning: espcomm_sync failed
> error: espcomm_open failed
> error: espcomm_upload_mem failed
> error: espcomm_upload_mem failed

I also tried to switch TX and RX, but that didn't work.

Anyone who can help?

qingz2004, 2018-05-11:
In my US version, there is no pin hole for Tx Rx, but anyway the serial port from ESP8266 is connected to the MCU STM8S103F3P6 which I think also drives the bus. You may have to cut the trace (at least the Rx) between ESP8266 and MCU before you can flash the ESP8266.

qingz2004, 2018-05-14:
Anybody tried this one?
https://www.amazon.com/2018Upgraded-Wireless-Assistant-Anywhere-Installation/dp/B07CL1M664
Does it use ESP8266?

ahzazou, 2018-05-19:
Im looking to buy this dimmer , any success guys !?

stanvv, 2018-05-24:
Unfortunately not.. I tried to contact the manufacturer asking where the pins (TX/RX/GND/3V3) are placed, but I don't think they understood what I meant. Anyone who can help?

ahzazou, 2018-05-24:
You need to connect those as mentioned on the board , them wile booting your esp8266 you need to connect gpio0 to ground to enable flash mode

ahzazou, 2018-05-24:
You can use the esp8266 type you have and it has all the pins you need

qingz2004, 2018-05-25:
[image]
This is how mine looks like. It's a USA version of the dimmer.
It is basically a ESP-12F. The picture show its backside.

ahzazou, 2018-05-26:
This one can be hacked easily as tx Rx vcc GND AND gpio0 is appearant on board

qingz2004, 2018-05-26:
As I said in the [previous ](https://github.com/arendst/Sonoff-Tasmota/issues/469#issuecomment-388363672) post, the serial port is used connect between esp-12F and the MCU. And the MCU also drives the port, so you have to cut the wire before you can program the esp8266.

galak, 2018-06-03:
I picked up:

https://www.amazon.com/gp/product/B07CTNSZZ8/ref=oh_aui_detailpage_o00_s00?ie=UTF8&psc=1

Its got a TYWE3S module in it, but I haven't had any luck in getting the device to flash.  I'm able to see output on the serial port.  I'll post some picture later if people are interested.

qingz2004, 2018-06-03:
Is there any other MCU like STM8 other than the ESP8285?

galak, 2018-06-03:
> Is there any other MCU like STM8 other than the ESP8285?

There might be, there are some chips that I haven't identified yet.  Why do you ask?

galak, 2018-06-03:
> There might be, there are some chips that I haven't identified yet. Why do you ask?

@qingz2004 Looks like there is a Nuvoton 8-bit micro

qingz2004, 2018-06-03:
If there is another MCU, flash ESP8285 will not get you anywhere. The other MCU is the main processor, and the ESP8285 is used for WiFi communication only.

galak, 2018-06-04:
Here are some images for the WIFI Dimmer:

[image]
[image]
[image]
[image]
[image]
[image]

galak, 2018-06-04:
> If there is another MCU, flash ESP8285 will not get you anywhere. The other MCU is the main processor, and the ESP8285 is used for WiFi communication only.

I tried disconnecting the MCU and that didn't help.

qingz2004, 2018-06-04:
[image]
TYWE3S is basically a packed ESP8266. From the picture, I can see only TDX, RDX, RST, EN, GPIO5, GND, and VCC are connected.
Before you flash, you have to ground GPIO0 and make sure Tx and Rx are not connected to anywhere else.
Make sure you backup the original stock image, or you may lose the control to the Dimmer. I guess all the logic is inside the MCU, like Zero cross detection, PWM generation, touch button response. While EPS8266 is just get the commands from WiFi and transfer them to the MCU through the Serial port.

galak, 2018-06-04:
> TYWE3S is basically a packed ESP8266. From the picture, I can see only TXD, RXD, RST, EN, GPIO15, GND, and VCC are connected.
> Before you flash, you have to ground GPIO0 and make sure Tx and Rx are not connected to anywhere else.
> Make sure you backup the original stock image, or you may lose the control to the Dimmer. I guess all the logic is inside the MCU, like Zero cross detection, PWM generation, touch button response. While EPS8266 is just get the commands from WiFi and transfer them to the MCU through the Serial port.

I'm guessing the fact that some combo of Tx/Rx seem to be connected to the cap sensor might be my issue.  I've disconnected the microcontroller board, but haven't tried w/o the cap sensor.  I've verified that my GPIO0 to GND is working - I'm in boot/uart flash mode based on the UART output:

```
 ets Jan  8 2013,rst cause:2, boot mode:(1,6)
```

vs normal boot output of:

```
 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x40100000, len 1396, room 16 
tail 4
chksum 0x89
load 0x3ffe8000, len 776, room 4 
tail 4
chksum 0xe8
load 0x3ffe8308, len 540, room 4 
tail 8
chksum 0xc0
csum 0xc0

2nd boot version : 1.4(b1)
  SPI Speed      : 40MHz
  SPI Mode       : QIO
  SPI Flash Size & Map: 8Mbit(512KB+512KB)
jump to run user1 @ 1000

OS SDK ver: 1.4.2(23fbe10) compiled @ Sep 22 2016 13:09:03
phy v
```

Guess I need to solder to the TYWE3S pins and disconnect it from the micro & cap sensor.

qingz2004, 2018-06-04:
I can't see it is connected to a cap sensor if you disconnect both ribbon cables.
It's not hard to trace the wires with a multi-meter, or even better if you have a logic analyzer to check the activities on the serial port.

qingz2004, 2018-06-09:
After cut the RXD trace, I was able to read and write the flash.
My board is [this](https://github.com/arendst/Sonoff-Tasmota/issues/469#issuecomment-391931403).
https://www.aliexpress.com/item/WiFi-Light-Dimmer-Switch-Stepless-Dimming-Touch-Panel-Dimmer-Work-with-Alexa-Google-Assistant-IFTTT-White/32857809626.html

murphyslaw05, 2018-06-11:
Why would you want to try and run it all from the esp? Zero cross is timing intensive especially with a chip that running WiFi and doing other duties.

qingz2004, 2018-06-11:
> Why would you want to try and run it all from the esp? Zero cross is timing intensive especially with a chip that running WiFi and doing other duties.

The MCU on my dimmer does not work well. Increase/Decrease buttons do not work, and the PWM does not sync with zero cross point. The bulb is flickering a lot when dimmed. The current dimmer uses smartlife app, and I need one to work with my MQTT server so I can integrate it into my home automation system.

I don't think zero cross is that timing intensive. 60Hz AC vs 80MHz system clock.
It should be doable like [this](http://www.instructables.com/id/Multichannel-Wireless-Light-Dimmer/)

chaosmaster, 2018-06-16:
I have also just received one of these and was already able to backup the original firmware, here is how I connected to the device:
[image]
[image]

qingz2004, 2018-06-17:
> I have also just received one of these and was already able to backup the original firmware, here is how I connected to the device:
That's how I backup mine. Possible to backup the MCU firmware? Since mine dimer does not work, I guess the firmware in MCU is no good.

chaosmaster, 2018-06-17:
I haven't tried backing up the MCU firmware yet. I am currently trying to get tasmota to talk to the MCU.
I could imagine it uses the same protocol mentioned here:
https://github.com/arendst/Sonoff-Tasmota/issues/595
https://github.com/enesbcs/ESPEasyPluginPlayground/blob/master/_P165_SerSwitch.ino

I assume backing up the MCU should be possible using an [STLINK v2](https://www.amazon.com/Aideepen-ST-Link-Programming-Emulator-Downloader/dp/B01J7N3RE6/ref=sr_1_1?ie=UTF8&qid=1529194633&sr=8-1&keywords=stlink+v2&dpID=41u9nyzYQoL&preST=_SY300_QL70_&dpSrc=srch)
(I have one lying around, but haven't tried yet)

chaosmaster, 2018-06-17:
I have made some progress, it is indeed the same protocol, although different data-format, since it is a dimmer.

The following code snippet switches the light on (to whatever brightness-level was previously set):
````
    Serial.write(0x55); // header 55AA
    Serial.write(0xAA);

    Serial.write(0x00); // version 00

    Serial.write(0x06); // command 06

    Serial.write(0x00); // length
    Serial.write(0x05); // length

    Serial.write(0x01); // light number
    Serial.write(0x01); // boolean
    Serial.write(0x00); // length
    Serial.write(0x01); // length
    Serial.write(0x01); // status (0 or 1)

    Serial.write(0x0E); // checksum:sum of all bytes in packet mod 256
    Serial.flush();
````

baudrate is 9600

I also have the format for setting the brightness, but need to test first

ahzazou, 2018-06-17:
Great news keep us updated i cant wait to see more updates i really appreciate your hard efforts

qingz2004, 2018-06-17:
Here are the results I got with a logic analyzer. They include transmitting command and receiving feedback.
**To turn on the light:**
Tx:
Time [s],Value,Parity Error,Framing Error
0.0s,0x55,,
0.0010419375s,0xAA,,
0.0020838125s,0x00,,
0.00312575s,0x06,,
0.004167625s,0x00,,
0.0052095625s,0x05,,
0.0062514375s,0x01,,
0.007293375s,0x01,,
0.00833525s,0x00,,
0.0093771875s,0x01,,
0.0104190625s,0x01,,
0.011461s,0x0E,,
0.05625s,0x00,,Error

Rx:
Time [s],Value,Parity Error,Framing Error
0.0126726875s,0x55,,
0.01371s,0xAA,,
0.01474725s,0x00,,
0.0157845625s,0x07,,
0.016822s,0x00,,
0.0178594375s,0x05,,
0.01889675s,0x01,,
0.0199340625s,0x01,,
0.0209714375s,0x00,,
0.022009s,0x01,,
0.0230465s,0x01,,
0.0240840625s,0x0F,,
0.0251215625s,0x55,,
0.026159125s,0xAA,,
0.0271965s,0x00,,
0.028234s,0x07,,
0.0292715s,0x00,,
0.0303088125s,0x05,,
0.0313459375s,0x01,,
0.0323830625s,0x01,,
0.03342025s,0x00,,
0.034457375s,0x01,,
0.0354945s,0x01,,
0.0365316875s,0x0F,,
0.037569s,0x55,,
0.0386064375s,0xAA,,
0.0396438125s,0x00,,
0.0406811875s,0x07,,
0.041718625s,0x00,,
0.0427559375s,0x08,,
0.04379325s,0x02,,
0.0448305s,0x02,,
0.0458678125s,0x00,,
0.046905125s,0x04,,
0.047942375s,0x00,,
0.0489796875s,0x00,,
0.0500169375s,0x00,,
0.0510541875s,0xFF,,
0.0520914375s,0x15,,
0.05625s,0x00,,Error

**To turn off light**
Tx:
Time [s],Value,Parity Error,Framing Error
0.0s,0x55,,
0.0010419375s,0xAA,,
0.0020838125s,0x00,,
0.00312575s,0x06,,
0.004167625s,0x00,,
0.0052095625s,0x05,,
0.0062514375s,0x01,,
0.007293375s,0x01,,
0.00833525s,0x00,,
0.0093771875s,0x01,,
0.0104190625s,0x00,,
0.011461s,0x0D,,
0.05625s,0x00,,Error

Rx:
Time [s],Value,Parity Error,Framing Error
0.012690375s,0x55,,
0.013727375s,0xAA,,
0.0147645s,0x00,,
0.0158015625s,0x07,,
0.016838625s,0x00,,
0.017875625s,0x05,,
0.018912625s,0x01,,
0.019949625s,0x01,,
0.0209865625s,0x00,,
0.0220235625s,0x01,,
0.023060625s,0x00,,
0.0240976875s,0x0E,,
0.02513475s,0x55,,
0.0261716875s,0xAA,,
0.02720875s,0x00,,
0.0282459375s,0x07,,
0.0292829375s,0x00,,
0.030319875s,0x05,,
0.0313568125s,0x01,,
0.0323938125s,0x01,,
0.033430875s,0x00,,
0.034467875s,0x01,,
0.035504875s,0x00,,
0.036541875s,0x0E,,
0.03757875s,0x55,,
0.03861575s,0xAA,,
0.0396528125s,0x00,,
0.0406899375s,0x07,,
0.041727s,0x00,,
0.042764s,0x08,,
0.0438010625s,0x02,,
0.044838125s,0x02,,
0.0458750625s,0x00,,
0.0469119375s,0x04,,
0.047948875s,0x00,,
0.0489858125s,0x00,,
0.0500229375s,0x00,,
0.05106s,0xFF,,
0.052097125s,0x15,,
0.05625s,0x00,,Error

**To increase the brightness**
Tx:
Time [s],Value,Parity Error,Framing Error
0.0s,0x55,,
0.001041875s,0xAA,,
0.0020838125s,0x00,,
0.0031256875s,0x06,,
0.004167625s,0x00,,
0.0052095s,0x08,,
0.0062514375s,0x02,,
0.0072933125s,0x02,,
0.00833525s,0x00,,
0.009377125s,0x04,,
0.0104190625s,0x00,,
0.0114609375s,0x00,,
0.012502875s,0x00,,
0.01354475s,0xC0,,
0.0145866875s,0xD5,,
0.05625s,0x00,,Error

Rx:
Time [s],Value,Parity Error,Framing Error
0.0158369375s,0x55,,
0.016874125s,0xAA,,
0.0179113125s,0x00,,
0.0189484375s,0x07,,
0.019985625s,0x00,,
0.02102275s,0x08,,
0.022059875s,0x02,,
0.0230969375s,0x02,,
0.024134125s,0x00,,
0.0251713125s,0x04,,
0.0262084375s,0x00,,
0.027245625s,0x00,,
0.02828275s,0x00,,
0.0293199375s,0xC0,,
0.03035725s,0xD6,,
0.0313944375s,0x55,,
0.032431625s,0xAA,,
0.0334688125s,0x00,,
0.034506125s,0x07,,
0.0355433125s,0x00,,
0.0365804375s,0x05,,
0.0376175625s,0x01,,
0.038654625s,0x01,,
0.039691625s,0x00,,
0.040728625s,0x01,,
0.041765625s,0x01,,
0.0428025s,0x0F,,
0.0438394375s,0x55,,
0.044876375s,0xAA,,
0.0459135s,0x00,,
0.046950625s,0x07,,
0.0479878125s,0x00,,
0.0490249375s,0x08,,
0.050062125s,0x02,,
0.05109925s,0x02,,
0.052136375s,0x00,,
0.0531734375s,0x04,,
0.0542105625s,0x00,,
0.055247625s,0x00,,
0.05625s,0x00,,Error

**To decrease the brightness**
Tx:
Time [s],Value,Parity Error,Framing Error
0.0s,0x55,,
0.0010419375s,0xAA,,
0.0020838125s,0x00,,
0.00312575s,0x06,,
0.004167625s,0x00,,
0.0052095625s,0x08,,
0.0062514375s,0x02,,
0.007293375s,0x02,,
0.00833525s,0x00,,
0.0093771875s,0x04,,
0.0104190625s,0x00,,
0.011461s,0x00,,
0.012502875s,0x00,,
0.0135448125s,0x62,,
0.0145866875s,0x77,,
0.05625s,0x00,,Error

Rx:
Time [s],Value,Parity Error,Framing Error
0.0159045625s,0x55,,
0.0169414375s,0xAA,,
0.017978375s,0x00,,
0.0190153125s,0x07,,
0.0200523125s,0x00,,
0.02108925s,0x08,,
0.0221261875s,0x02,,
0.0231631875s,0x02,,
0.02420025s,0x00,,
0.0252371875s,0x04,,
0.02627425s,0x00,,
0.0273113125s,0x00,,
0.0283483125s,0x00,,
0.0293854375s,0x62,,
0.0304224375s,0x78,,
0.0314594375s,0x55,,
0.0324964375s,0xAA,,
0.0335334375s,0x00,,
0.0345705625s,0x07,,
0.0356076875s,0x00,,
0.0366446875s,0x05,,
0.0376816875s,0x01,,
0.0387188125s,0x01,,
0.0397558125s,0x00,,
0.04079275s,0x01,,
0.0418298125s,0x01,,
0.042866875s,0x0F,,
0.043903875s,0x55,,
0.044940875s,0xAA,,
0.045977875s,0x00,,
0.0470146875s,0x07,,
0.0480515625s,0x00,,
0.0490884375s,0x08,,
0.050125375s,0x02,,
0.0511625s,0x02,,
0.0521995s,0x00,,
0.053236625s,0x04,,
0.05427375s,0x00,,
0.0553108125s,0x00,,Error

Actually, the MCU keep sending the light status periodically. That's why we can not read or update the FW of the esp8266 without cutting the Rx trace from the PCB.

checksummaster, 2018-06-17:
check my previous post for a zero crossing + triac... it still need a esp32
but it more cheap than build the circuit yourself

qingz2004, 2018-06-17:
> Could you test if it always sends the same values when increasing or decreasing?
Just tested. Every time it sends different values when increase or decrease.

> Maybe it's possible to keep the MCU in a reset-state and thus flash without cutting RX?
It all depends on how strong the MCU pull up or down the line.
It the MCU pulls down more strong than your programmer, you will not be able to pull it up.

qingz2004, 2018-06-17:
My MCU does not work well. Modifying the ESP8266 FW will not help.
The MCU does respond to ESP8266 commands sent from serial port, but it does not respond to the button touches (respond to on/off button, but not +/- buttons). And when the bulb dims, it flickers a lot. The PWM generated by MCU is not synced to zero crossing. I probed the PWM with an oscilloscope, half of the pulses are wrong.
I wish I can program the MCU with a good FW or totally bypass the MCU.

chaosmaster, 2018-06-17:
> It all depends on how strong the MCU pulls up or down the line.
It the MCU pulls down more strong than your programmer, you will not be able to pull it up.

I was thinking along the lines of just connecting the MCUs reset to gnd/vcc and keep it there while programming so it just doesn't boot up just how it can be done with the esp as well.

> I wish I can program the MCU with a good FW or totally bypass the MCU.

Using an STLINK v2 that should be possible, although i also saw some kind of update method in the [serial protocol](https://docs.tuya.com/en/mcu/mcu-protocol.html)  
That would also mean writing a firmware for it.

Do you think the hardware used for dimming is fine and its just an MCU issue or could it also be a hardware issue?

qingz2004, 2018-06-17:
> I was thinking along the lines of just connecting the MCUs reset to gnd/vcc and keep it there while programming so it just doesn't boot up. Just like how it can be done with the esp as well.

As long as it is not tri-stated during reset, it has some strength to pull up or down.

> Do you think the hardware used for dimming is fine and its just an MCU issue or could it also be a hardware issue?

I believe it is mcu software issue. I probed the touch button signal and cross zero detection signals which go into mcu are all good. And I bought a few of those dimmers, all have the same problem.
Mines are US version.

chaosmaster, 2018-06-18:
Since I just had to reflash the device I tried without disconnecting RX and pulling the MCUs reset low while flashing. 
This worked well, the NRST pin of the MCU is even exposed on the header (I mislabeled it before):
[image]

[509 more replies]

closed 2018-10-16 by ascillato2