arendst/Tasmota #470: cannot open linker script file eagle.flash.1m0.ld

samuel235, 2017-05-21:
I'm having a compile issue when it comes to the 1M (no SPIFFS) variant. I have followed the prerequisite page of the wiki and put the .ld file into "portable\packages\esp8266\hardware\esp8266\2.3.0\tools\sdk\ld" and replacing the boards.txt file at "portable\packages\esp8266\hardware\esp8266\2.3.0" however i get the following error:
`c:/users/samuel/desktop/arduino-1.8.2-windows/arduino-1.8.2/portable/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld.exe: cannot open linker script file eagle.flash.1m0.ld: No such file or directory`

However, i notice that inside of the boards.txt file the .ld file is named as "eagle.flash.1m0.ld" and not "esp8266.flash.1m0.ld" like it is in the git folder structure. So, I tried two changes, one was to change the .ld file name itself to eagle rather than esp8266, and the other was to change the boards.txt naming of the file but neither of these worked and gave me the same error. Obviously its referred to in other files that i'm not aware of.

If you need any more information or if i'm being an idiot then please just point it out :)

arendst, 2017-05-22:
You used the platformio linker script from the root. For arduino you need the linker script from the arduino folder.

samuel235, 2017-05-23:
You're right, i'm sorry. I think i've either partially bricked/damaged these two POWs i have or they're a bad batch as they're so sensitive to the power, i can't get them to stay powered on without them rebooting themselves. One is starting to not even turn on at all now. Powered through external 3.3v supply (which is being pulled down to about 2.2v when powering on too) Will close the issue now, thank you for pointing that out for me.

closed 2017-05-23 by samuel235

samuel235, 2017-05-24:
Theo, I have managed to get one working properly now. Do you happen to have some sort of wiki page explaining how to customise each install of Tasmota, so basically what you change for each board. Do you personally change each project name inside of the sketch as you upload them to new sonoffs? Would this automatically change the MQTT topic name too or is that something you do manually as well?

davidelang, 2017-05-24:
look in sonoff_templates.h

samuel235, 2017-05-24:
Not sure what i'm supposed to read that as, to my understanding that file is just all the options within each varient of Tasmota depending on which module you're using and various 'general use' GPIO functions too? Would I be misunderstanding?

sebringsc, 2018-02-10:
@arendst I'm having this exact same problem with the most recent download of Tasmota as well as Arduino 1.8.5.  I've followed your wiki on moving all of the folders from the sonoff and lib folders to the libraries folder in arduino however I'm not sure what you mean when you say that you need the linker script from Arduino.  I have zero experience with this or any coding so if you could put it in layman terms I'd appreciate it.  My exact error is : 

Archiving built core (caching) in: C:\Users\Cullens\AppData\Local\Temp\arduino_cache_105015\core\core_esp8266_esp8266_generic_UploadTool_esptool,CpuFrequency_80,FlashFreq_40,FlashMode_dout,UploadSpeed_115200,FlashSize_1M0,ResetMethod_ck,Debug_Disabled,DebugLevel_None_____b1cf05fec1f8df6040bb64f9db0d82de.a
//mac/home/desktop/arduino-1.8.5/portable/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld.exe: cannot open linker script file eagle.flash.1m0.ld: No such file or directory

collect2.exe: error: ld returned 1 exit status

arendst, 2018-02-10:
I suppose you are using version 2.3.0 of the ESP8266/Arduino library as the latest version, 2.4.0 already has the correct linker script installed. No problem. Just re-read the wiki again starting from:

1) Select from heading Upload Tools > Arduino IDE
2) Under heading Copy files > ESP8266 Board version before 2.4.0 read what you have to do to get the linker file in the correct position.
3) Report back your experience

sebringsc, 2018-02-10:
I was using 2.4 and received the error so I went back to 2.3 and followed the steps and was able to get it to compile and upload successfully.

Now I am not able to get the Sonoff to connect to my wifi.  The Wifi config section of the user_config.h is correct for my SSID and password so I'm at a loss.  I would check the config but since the Sonoff will not connect to my network, I am not sure how to check the config.

sebringsc, 2018-02-10:
@arendst 
I switched to platformio and was able to rebuild and upload with success however when I check the serial monitor, this is what I see:

  File "/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/threading.py", line 810, in __bootstrap_inner
    self.run()
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/threading.py", line 763, in run
    self.__target(*self.__args, **self.__kwargs)
  File "/Users/Cullens/.platformio/penv/lib/python2.7/site-packages/serial/tools/miniterm.py", line 445, in reader
    data = self.serial.read(self.serial.in_waiting or 1)
  File "/Users/Cullens/.platformio/penv/lib/python2.7/site-packages/serial/serialposix.py", line 509, in read
    raise SerialException('read failed: {}'.format(e))
SerialException: read failed: [Errno 6] Device not configure