arendst/Tasmota #471: No SENSOR topic, but there

vadimbz, 2017-05-22:
n

davidelang, 2017-05-22:
it's not clear what you are asking.

are you saying that you hooked up a sensor, configured it in the module config, 
and are not getting sensor data sent out in the tele/ messages?

we would need more details to know where to start.

vadimbz, 2017-05-22:
moment, fatfinger hit the enter button

vadimbz, 2017-05-22:
...and Home Assistant, and Mosquitto
`- platform: mqtt
  name: "Temperature"
  state_topic: "tele/sonoff2/TELEMETRY"
  value_template: "{{ value_json['DHT'].Temperature }}"
  unit_of_measurement: "°C"
- platform: mqtt
  name: "Humidity"
  state_topic: "tele/sonoff2/TELEMETRY"
  value_template: "{{ value_json['DHT'].Humidity }}"
  unit_of_measurement: "%"`

closed 2018-05-08 by stale[bot]