arendst/Tasmota #477: Update Wiki to reflect flash size setting 1M (no SPIFFS) for version 5.0.0 and up

TRIROG, 2017-05-24:
If i understand correctly, for flashing version 5.0.0 and up Flash size should be set to 1M (no SPIFFS).
Wiki should reflect that.

gajotnt, 2017-05-24:
It does...
https://github.com/arendst/Sonoff-Tasmota/wiki/Prerequisite

TRIROG, 2017-05-24:
I was refering to the section [https://github.com/arendst/Sonoff-Tasmota/wiki/Hardware-Preparation](url)
Where for example for the Touch it is stated that certain changes should be made, which could be misunderstood.
```
Sonoff Touch
As the Sonoff Touch is based on the ESP8285 using Flash Mode DOUT you will have to make some changes to the proposed Arduino IDE settings as follows:

    Tools Board Generic ESP8285 Module
    **Flash Size: 1M (64K SPIFFS)**
```

gajotnt, 2017-05-24:
There it should be updated :)

filzek, 2017-06-08:
There is a problem on Arduino IDE where it doesnt allow 1M (no SPIFFS), so how to set it?
[image],

Is there any idea?

closed 2018-05-08 by stale[bot]