arendst/Tasmota #480: Sonoff POW

agsochi, 2017-05-24:
Hi. Thank you for great firmware.
I use Sonoff POW to detect my washing machine on/off.
After washing cycle it go to standby mode and in this mode it consumes very low power.
When washing machine is in standby mode - i look at SonoffPOW main page and i see `Power` changing every 10-20sec from 0w to 3w to 0w to 6w and so on.

Feature request:
`Teleperiod` takes into account only last measurement, and if it was 0 - `Teleperiod` shows 0,
but 20 sec before power was 3w.
Can you please add `TeleperiodAverage` (or something like that) parameter, which count all `Power` measurements for `Teleperiod` period (120s) and shows average value for this period?
Result will looks like  `TeleperiodAverage=1.234W` for  `Teleperiod` period (120s) where measurents was from 0w to 6w.

davidelang, 2017-05-24:
it would be better to do this in your home automation software, or extend the 
tele period to cover the timeframe you want info for.

I'm not arendst, but remember that we are tight on space and so every feature 
added needs to be balanced against it's cost. In this case the calculation is 
trivial, but the infrastructure needed to support it (command handling, config 
space, etc) are as expensive as any other option.

agsochi, 2017-05-25:
But extending `Teleperiod` not help in this case because of `Teleperiod` takes into account only last measurement. And if it (measurement)  was 0-3-0-6-0-0-0-0-3-0-0 - `Teleperiod` shows always `0`

mihalski, 2017-05-26:
To do this in home automation software you would pretty much have to set tele period to 1 which is impractical. This does indeed look like something that would need to happen on the device.

agsochi, 2017-05-26:
> EDIT: Perhaps alternatively you could compare the energy used between the last update and the current one? I would expect that to be cumulative.

Thanks @mihalski, don't think about that.

davidelang, 2017-05-26:
Are you sure? I would expect the 'last measurement' to cover the entire last 
teleperiod.

> But extending `Teleperiod` not help in this case because of `Teleperiod` takes into account only last measurement. And if it (measurement)  was 0-3-0-6-0-0-0-0-3-0-0 - `Teleperiod` shows always `0`

mihalski, 2017-05-26:
That's what I though too.. But I'm not certain.. Lots of 0's here too.. Maybe he'd be better off monitoring power? Perhaps Watt hours just tend to very low numbers (below 1) for short periods?

Here's my fridges an example:
[image]
[image]

agsochi, 2017-05-26:
Here is my Washing machine in standby:
[image]

davidelang, 2017-05-26:
I assume this is with a 1m log interval, what is you set it longer?

agsochi, 2017-05-26:
2 minutes

mihalski, 2017-05-26:
[image]

davidelang, 2017-05-26:
if you have more than 2 points in a 10 min period, it is not set to a 5 min reporting interval. try power cycling the device to make sure it is using your new settings

agsochi, 2017-05-26:
I zoom my graph - yes it shows 2min and 20sec periods. But 5min looks same.

davidelang, 2017-05-26:
If the device is only reporting one value every 5 min, then any graph that shows 
details every minute or every two minutes cannot possibly be correct.

So if you are getting per-minute (or per two minute) data, then you have the 
tele period set for 60 or 120 seconds, not 300 seconds.

The POW only computes the power usage when it's sending a tele/ report, it 
doesn't generate it every X seconds and send the last one, it keeps a running 
total of power used and computes the number it reports when it generates a tele/ 
report.

agsochi, 2017-05-28:
Maybe you wrong, POW shows usage at main page with period every 2-3 sec.

goprojojo, 2017-06-02:
I'm looking for the same solution to receive a notification when the washing cycle is ready. Anyone success on this? If yes how should the config look like?

agsochi, 2017-06-03:
I got result - look at `Total` graph. I took `Total` value with `Teleperiod=60s` and compare it with previous value. Measurement is very accurate with resolution below 1W (compare with `Power W` from teleperiod stats). Here is grafana setting for `Total` measurement.
[image]

closed 2017-06-03 by agsochi