arendst/Tasmota #481: Hard Thermostat

off14, 2017-05-25:
Is it possible to include functionality of HARD thermostat - where no MQTT(or similar) server is available or there are no commands received.
Of course this mode will require temp sensor connected.
You need to include logic to support COOL or HEAT modes,  Hystersis setup  (0.1 - 5) together with TargetTemperature

This will make sonoff working as a thermostat with web interface, even without any server. Very useful when sonoff is installed on remote location and there are big chances that internet is not stable there

WDYT?

citrusfizz, 2017-05-25:
i like this idea.   i recently built a thermostat with a raspberry pi, some relays and Node-Red and mqtt all built in.   Trusting the PI for 24/7 operation is a bit sketchy. and if it reboots it takes a while to boot up and get all its gears turning.   Working well so far,  if you have a PI i can share my node red flows with you.

off14, 2017-05-26:
Thanks, i have mine also in node-red and in openhab rules
the main idea is to have **independent** thermostat, which not requires any server (mqtt or whatever)

CommodoreWhite, 2017-05-26:
Why not for any sensor that returns an analogue value (that is, not binary). So, for instance, a set point for humidity to control the bathroom extractor fan, a light sensor could turn security lights on and off, a proximity detector could control stair well lights. Hmm, the last one is a bad example but you get the picture. 

The underlying logic is generic, if a sensor value is exceeds some thresholds, switch a relay and or send a message.

The implementation might not be as responsive as you like because, if memory serves, sensors are polled rather than continuously monitored and why having a proximity sensor manage stair lights could have undesirable consequences.

So it's a +1 from me.

Peter

CommodoreWhite, 2017-05-26:
Oh! I forgot. It would be nice if the feature could be enabled or disabled depending on the presence or absence of an Mqtt connection. So, if the Mqtt service dies, lights still turn off and on. If Mqtt is working the control reverts to a higher authority.

AlexTransit, 2017-05-26:
I made a simple thermostat at my place. (On fast)
He has no settings

I wanted to make PID and settings.
Do not do it. :)

sonoff.ino
void every_second()
{
  char log[LOGSZ], stopic[TOPSZ], svalue[MESSZ], stime[21];
  float ctemp;

#ifdef TEMPERATURE_CONTROL_DS18x20
if (ds18x20_read(0, TEMP_CONVERSION, ctemp, true)) 
{
  if ((ctemp == 85.0) || (ctemp >= sysCfg.temp_target)){
      if (power != 0) {
      do_cmnd_power(1, 0);}
      //setRelay(0);
  } else {
      if (power != 1) {do_cmnd_power(1, 1);}
      //setRelay(1);
  }
} else { //sensor error. disable power
  if (power != 0) {do_cmnd_power(1, 0);}
}
#endif // TEMPERATURE_CONTROL_DS18x20

davidelang, 2017-05-26:
This question has come up several times.

It is currently not supported, although it would be possible to add it.

One problem is "what sensor should it look at" given that multiple sensors can 
be attached at the same time.

Another is how many such configs should it allow at once? (on a dual, should it 
be able to control both a heater and an AC with different settings?

There is similar functionality in the POW to respond locally to power 
consumption, so it's not out of the question. but someone would have to think 
through the details and code it in a way that does not require a lot of extra 
flash or ram.

AlexTransit, 2017-05-26:
All right.
I did not have a source on my boiler.
I specifically use the library 18x20
And the temperature adjustment is done taking into account the sensor address.
Also incorrect data can come from the sensor. Then the temperature will be 85.0 degrees.

If you make a thermostat, then there may be several controls.
Just a relay or SSR
Or two relays, add a cold, and add hot.
Still the sensor 18 is quite slow. (Sample 700ms)

off14, 2017-05-27:
i think it should be independent from type of sensor you are using. if you have **any sensor supplying temperature** then the "thermostat" option should be available, with dropdown list for sensors found.

of course, could be many variations, like 2 relays, pow-models etc - but i guess it will make the development complicated and even not compatible with sonoff chip (memory size)

my suggestion is to go with simplest possible setup
MODE: HEAT/COOL
HYSTERSIS: 0.1-5.0
MQTT_MSG: thermo { "mode": <mode>, "hyst": <hystersis>, "currentstate" : <state>, "enabled" : <boolean> };

pprotschka, 2017-07-26:
has anybody been able to implement this? Tasmota is great but Wifi routers and Networks fail...and i dont want my walk in freezer to be left ON forever.

i know that easyesp has "rules" that make this possible but unfortunately easyesp is a lot more tricky to  configure.

davidelang, 2017-07-26:
This has been a repeated request.

mo8Zomo0, 2017-08-19:
+1

I was wondering about the same functionality and before opening a new request I searched and found this one.

It would really be a handy feature to control heating, cooling, de-/humidifiers, or use input from other sensors like light, moisture, ...

nambuco, 2017-10-09:
+1 that would love this functionality.

To make implementation simple, should focus on Sonoff TH10/TH16 hardware (one sensor / one relay).

apathyuk, 2017-10-09:
Sounds interesting. Has anyone have a solution for controlling radiator TVRs?

PeggyFree, 2017-10-09:
It would be not far from the existing power threshold triggering, no?

vstonkus, 2017-10-10:
I building electric floors heating in bathroom. There is a simple 220v wire (https://www.infrafloor.com/images/HomeBoxes/big/InfraFloorMat-FloorHeating.jpg). I think that Sonoff TH10/16 with water proof temp sensor inside floor concrete below tiles would be a solution here.
I would like to get it connected to home assistant, and Homekit. Would be great configure it to raise tiles temperature in the morning and in the evening for a little.
In bathroom there is other unit to control - coil heater, but it should be on after shower, so it can be connected on Sonoff TH with temperature and humidity sensor. Home assistant for example could notice increased humidity and turn on coil heater or mirror heating for example. Mirror does not need hard thermostat, but coil heater and mostly floors heating does.

I agree with @nambuco. Would be great to have this feature.

nambuco, 2017-10-11:
I will try to implement a temp control for Sonoff TH, in case it works I will share it here. I may need help from someone who knows the sw structure. I will use AlexTransit code as a starting point, add PID and setpoint adjustment using MQTT (that's the complex part...).  Let's see...

AlexTransit, 2017-10-11:
I connected Sonof BASIC to the thermopot. as a relay I use SSR.
there is a small problem.
if you turn on the web. and disable the WIFI point then
the procedure "every_second" does not work until you go to the web page.
the firmware version is very old. (3.2.13)
I do not have enough programming knowledge and experience to implement a normal PID controller
but I'll find out. :) I do not know when.
still there is a problem with the OTA update when the size is more than 50%
this set back my further elaboration of PID.
the size of the firmware is great. and update the cable is not possible.

spacefolder, 2017-11-01:
Hi guys! Any news on this request? I have my sonoffs running my own (buggy, sloppy and slow) firmware, but having a TASMOTA Thermostat would be great, reaaly great! 

+1

nambuco, 2017-11-07:
Hello Guys, I finally got my USB/TTL converter and flashed my first Sonoff TH for testing. I forked this repository and will start coding a "hard thermostat".

Every tip / information will be welcome.

I will implement an on-off controller because my application (refrigerator) does not require a PID control. 

I noticed that temp measures are done on every telemetry cycle defined by TelePeriod parameter. My idea is to run the thermostat function at same frequency.

I want to use some of the free setting bytes for a **SetPointTemp**, **Histeresys** and **ThermostatMode**  - On (or Auto) and Off .

I will post here the evolution of this change.

Nambuco

davidelang, 2017-11-07:
look at how the power limit functions are done, they are probably a good model

mo8Zomo0, 2017-11-14:
It would be good if the TH1x temperature sensors would be supported and even cooler of also a remote temperature sensor could be supported.

@nambuco Btw, regards flashing, check out: https://github.com/mirko/SonOTA

nambuco, 2017-11-15:
@davidelang, thanks for the tip, it helped me to implement the first beta version available on https://github.com/nambuco/Sonoff-Tasmota for testing.

 @mo8Zomo0, I am using the OTA in the WebServer interface, works fine here. I implemented for DS18B20, I do not have the TH1x sensor to test, but should be easy to adapt (replace Ds18b20ReadTemperature() by ShtReadTempHum() in the thermostat.ino file).

sgttrs, 2017-11-19:
@nambuco Can you make in the firmware a simple logic for switching the relay?
Example:
if Temperature<30 - Relay OFF
if Temperature>30 - Relay ON

nambuco, 2017-11-19:
Hi @sgttrs ,

If you change the constant on thermostat.ino file

#define OFF_DELAY 90  // minimal off time in seconds
#define HYSTERESIS 2  // Hysteresys in 0.1 Degrees

both to zero, you have the effect you want. I just warn you that turning off and on before 60 seconds can damage a compressor (if this is your application).

Important point, set tele_period to 10 to allow a fast response

sgttrs, 2017-11-19:
@nambuco Thanks for the answer. I want to use Sonoff in the heating system to start the circulation pump. Is it suitable for this application?

nambuco, 2017-11-20:
Hi @sgttrs , if the logic is like you describe (temp above setpoint turns on the relay) it will work as it is. Please be aware that I did not test in all possible conditions, You have to download and help me test in your setup.

I am planning to add a second mode where the relay will turn on if temp is below setpoint.

sgttrs, 2017-11-21:
Hi @nambuco, I'm testing the firmware with the settings you suggested. When I activate the thermostat and enter SETPOINT 30, the thermostat turns on the relay at any temperature, even if it is less than 30 ° C. I also noticed that when I connect Sonof to AC, it turns on and off the relay several times. Is there any way to avoid this?

nambuco, 2017-11-21:
Hi  @sgttrs, the SetPoint is defined in 0,1 steps, this means, to get 30C you have to input 300 in SetPoint.
Concerning the on / off during boot, that odd, I never faced it, but I will test in different conditions.

pixidixi88, 2017-11-26:
Hi
I would like to use SONOFF TH16 to floor heating control. I have a AM2301 sensors. I need such rules:
If temperature is > wished temp - relay ON
If temperature is < wished temp - relay OFF
It seems like "cooling" because it will close valves when temperature is higher then wished.

Is it possible to put wished temperature via MQTT? I would like to add this as "thermostat device" to my Homebridge.

I am beginner  so please explain me as clear as possible. :)

sgttrs, 2017-11-26:
Hi @nambuco. I just tried to change the temperature with the `setpoint` command via mqtt and via console. The result of processing the command:
```
15:41:25 CMD: setpoint 500
15:41:25 MQT: stat/sonoff/RESULT = {"SetPoint":300}
```
Rebooting the module does not help.

Hi @pixidixi88. The firmware is still under development. Various errors in work are possible. If you are ready for this, [here](https://github.com/arendst/Sonoff-Tasmota/wiki) is the instruction on how to install the firmware on the module. Also read the posts of **nambuco** in this thread.
I use Home Assistant to control my devices. To set the temperature value through mqtt, send the topic: `cmnd/sonoff/setpoint`. In the content of the topic, set the desired temperature value, multiplied by 10.

nambuco, 2017-11-27:
Hi @sgttrs, the setpoint is limited to -500 to +499. It is easy to change that and I will put it in a constant next release. Please check if it works for 499.

Hi @pixidixi88, I implemented this feature only for DS18b20 sensors, I have to check how AM2301 is implemented to see if it is easy to add it to thermostat function. If someone has any tip on AM2301 it will be helpful.

pixidixi88, 2017-11-27:
Hi @sgttrs 
I have installed Tasmota software and I am very glad of use this, but what I really need is thermostat function. I am not afraid about errors because I understand that it is more less "homemade" solution.
Hi @nambuco 
I have tried adjust code from DS18B20 to AM2301 by myself but I failed :( so I really need your help. If you need any data, please let me know.

sgttrs, 2017-11-27:
Hi @nambuco. Yes. At values up to 499, everything works fine. Thanks.
Is it possible to configure the firmware to work in this mode? Example:
Setpoint 400. Hysteresis 50.
T > 450 - Relay ON
T < 350 - Relay OFF
T 350:450 - Saving the previous state of the relay

jtamasi, 2017-12-01:
Hi @nambuco
First of all, I'd like to thank you for the work you put into this mod.
I think I'm not alone when I'm saying that AM2301 compatibility is very important because that is the most obvious choice for a thermostat with Sonoff TH. Could you waste some time and try to add it to the supported sensors
Cheers

skorc99, 2017-12-09:
Hello,

this works for cooling.

How to change for heating?

T<20 relay ON
T>20 relay OFF
ON/OFF delay x seconds

This should keep setpoint temperature.
U

Stevehans, 2018-04-02:
Any chance of a pull request for this code, would be great to see this in mainstream?

Stevehans, 2018-04-04:
Would also be good to extend the timer functionality to have on/off/auto (min,max)

IceStuff, 2018-04-08:
Hello All,

I extended on the Hard Thermostat with webpage controls, heating and cooling modes and supports all temperature sensors.

Check it out and provide feedback would be great.  Humidity control is coming.

https://github.com/IceStuff/Sonoff-Tasmota/releases/tag/v5.12.1

Thanks,
IceStuff

Highcooley, 2018-04-15:
Hi IceStuff

I did the German translation and gave it a try with a TH10. However, somehow I cannot find the Thermostat webpage controls, no matter what language I flash the module with. Do I have to select another module type (I already tried standard and dual next to TH)? Also with the MQTT request STAUTUS:10, I don't get a temperature in return anymore. 

Cheers,
Andy

wasi54, 2018-04-16:
Hi IceStuff
I loaded [5.12.1_beta.bin](https://github.com/IceStuff/Sonoff-Tasmota/releases/download/v5.12.1/sonoff-GB-thermostat-5.12.1_beta.bin) to one of my TH10 via sonoff-minimal and OTA. It is working verry well.
Next step was compiling from source with PlattformIO (PlatformIO 3.5.3a9 and all updates are installed). Compiling was successfull without any issues.
Unfortunately the TH10 was not visible on the WiFi any more after OTA upgrade. Button 4 short presses will show the WiFi Manager and i was able to set the SID and password. But no WiFi access after reboot and not pingable. Toggle the Button will toggle the Relais and the LED's. Furthermore the generated firmware.bin is 2k smaller than the beta binarie.
Any idea or help is highly appreciated.
Thanks

wasi54, 2018-04-16:
@Highcooley
I guess MQTT_HOST set to "" and the library #define MQTT_LIBRARY_TYPE MQTT_PUBSUBCLIENT will cause the webserver to hang ...
The solution for me was:
#define MQTT_LIBRARY_TYPE MQTT_TASMOTAMQTT

wasi54, 2018-04-16:
Thermstat module setting:
[image]

Highcooley, 2018-04-16:
@wasi54 got it, setting MQTT_HOST helped and the sonoff didn't hang. My other issue (no thermostat menue) *facepalm*, reflashing removed the sensor setting which switches off this menue. 
Now everything is working splendid. 
@IceStuff the German translation works a treat as well. Do you have your 
My next step is to implement TM1637 (4 digit LED Display) support including two buttons to set the thermostat temperature at GPIO 2 and 4 or 5.

IceStuff, 2018-04-17:
Sorry about that in my last post.  I mentioned that as soon as Sonoff TH was selected the thermostat settings menu would appear.  I forgot that you also need to have a proper temperature sensor on GPIO 14.

I removed the MQTT_HOST during testing because it was flooding the console with errors not being able to find the default MQTT host and it was causing my debug messages to roll off screen too quickly.  I noticed that the first time I flashed it with the MQTT host blank that it hung but once I rebooted it came back online.  I chalked it up to my mistake.

@Highcooley do you have the German translation you can send to me so I can implement it?

You may have noticed some of the default includes in the config file were disabled in my code.  This was to keep it under 500k so I could upload the changes via the webpage.

I am not sure why the code would be 2k smaller when you compiled it unless there was an update.  My Platform IO is Home 0.9.5·Core 3.5.3a9 with Visual Studio Code Version 1.21.1
Commit 79b44aa704ce542d8ca4a3cc44cfca566e7720f1
Date 2018-03-14T14:46:47.128Z
Shell 1.7.9
Renderer 58.0.3029.110
Node 7.9.0
Architecture x64

Are you using VS Code or Atom?

Glad to hear you have this working.  I was thinking to implement the display and rotary encoder via the RS232 programming header since they also have power on the header.

IceStuff

AlexTransit, 2018-04-17:
good solution (use pid)
https://github.com/arendst/Sonoff-Tasmota/issues/2425

wasi54, 2018-04-17:
@IceStuff Atom 1.25.1x64 on Linux with PlatformIO 3.5.3a9 is currently the newest version available.

IceStuff, 2018-04-19:
@Highcooley thanks for the translation.  I merged it with my repository.  I also created a pull request back to the main code, hoping they will implement along with their main code.

@AlexTransit I was debating using PID.  I figured it would be too much on and off when using relay control.  If we were using PWM or analog output then it would definitely make sense.

AlexTransit, 2018-04-19:
@IceStuff  I figured it would be too much on and off when using relay control. 
all is correct. if the PID time is short, frequent operation will quickly break relay.
can make SSR in GIPO settings?

there is a defect in the current branch. because of it I have not yet checked the firmware on my boiler.
I'll try to fix it.
If there is no data from the sensor, or data with an error, it is necessary to disconnect the load.
I assembled the test version.
At low temperature, the load is switched on. I physically disconnect the sensor and everything continues to work. It is not safe.

arendst, 2018-04-20:
My two cents:

How about a rule like:
```
{"Rule":"ON","Once":"ON","Rules":"on ds18b20#temperature>24 do power off endon on ds18b20#temperature<23 do power on endon"}
```
With variations in warming/cooling or other sensor type. Once the rule is configured the unit will perform like a thermostat standalone.

aenertia, 2018-04-28:
Without going the ssr route why not set a timeout since last change of state. Something like a tunable and if the state changed X seconds ago then don't trigger the relay again until the timeout. This combined with a range +/- sensor threshold would solve frequent flipping of the relay problem.

dweston, 2018-04-29:
@IceStuff I've been using your v5.120k binary for the past 4-5 days and the whole thing has run impeccably. For a non-programmer the UI was straight forward and very effective. I'm running it on Sonoff TH16 controlling a fan in a cavity behind a wine fridge. Wasn't sure what parameters best suited the situation. Didn't even know what temp settings to work with. Set up a nodered UI chart and then kept tweaking your settings on the device UI directly - and within 48 hrs adjusting temp & hysteresis had the whole situation controlled as well as I believe will ever be. Fantastic.

IceStuff, 2018-04-29:
@AlexTransit I fixed the output staying on if temperature sensor fails.  Please check out the latest release and provide feedback: https://github.com/IceStuff/Sonoff-Tasmota/releases/tag/v5.12.2
It will have to wait until the sensor read times out then will shut off relay.  It will also go through the regular cycle to make sure the minimum off time is met.

@dweston I really appreciate your comments.  I am glad that you found the hard thermostat helpful.  I recommend upgrading to the latest release to implement the fix in case the temperature sensor fails.
https://github.com/IceStuff/Sonoff-Tasmota/releases/tag/v5.12.2

IceStuff

IceStuff, 2018-04-30:
@arendst Sorry I missed your comment above.  Can you give me a link that would explain the rule you posted?  Why would you include the sensor type?  The current firmware will work with any of the sensors.

Thanks,
IceStuff

arendst, 2018-04-30:
See Wiki - Rules.

The rule allows any sensor, current and future as you just have to enter your local sensor name as seen in the JSON teleperiod message.

The main advantage of a rule is that here is no coding needed from my side; it's all in the hands of the user allowing me to continue to the next level...

spacefolder, 2018-05-21:
Hi!

I have a Sonoff TH16. Was working flawlessly with tasmota 5.10.0.
Decided to try the ThermoWeb. Flashed it ok OTA.

Now I can't see the sonoff in my network.
Serial Monitor shows that the device boots up just fine, 
but it's manually self assigning the IP to 192.168.1.1 (which is my routers IP).

If I unplug my router, I can access the web interface (I have a dedicated AP for home automation), but it's asking to login in, with user & pass, which I don't know.

[image]

[image]

Any ideas?
Cheers!

dweston, 2018-05-21:
@IceStuff I have been running with v5.12.2 now for 20 days and it has worked flawlessly.

ascillato, 2018-05-21:
Hi,

In the wiki, now is available an example of a Simple Thermostat using RULE feature (https://github.com/arendst/Sonoff-Tasmota/wiki/Rules#7-simple-thermostat-example)

Please, check it to see if that helps you.

spacefolder, 2018-05-21:
@dweston , did you go from "regular" tasmota  to ThermoWeb, uploading **sonoff-GB-thermostat-5.12.2_beta.bin** OTA through the web interface, or did you flash the .bin using the serial port?

dweston, 2018-05-21:
@spacefolder I downloaded the .bin file off Github and flashed it OTA within my local network.

dweston, 2018-05-21:
@ascillato I would not say that is simple, but I can certainly see the powerful flexibility built into the 'rule' system. When I can set aside the time I will give it a go using this example. In the meantime for my present purposes v5.12.2 is working perfectly well for me and is considerably easier to work with.

[11 more replies]

closed 2018-12-22 by ascillato2