arendst/Tasmota #482: Receive this error while compiling

claudioita, 2017-05-26:
What can this be?

Arduino:1.8.2 (Windows 10), Scheda:"Generic ESP8266 Module, 80 MHz, 40MHz, DIO, 115200, 1M (64K SPIFFS), ck, Disabled, None"

C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\Claudio\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\Claudio\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\Claudio\Documents\Arduino\libraries -fqbn=esp8266:esp8266:generic:CpuFrequency=80,FlashFreq=40,FlashMode=dio,UploadSpeed=115200,FlashSize=1M64,ResetMethod=ck,Debug=Disabled,DebugLevel=None____ -ide-version=10802 -build-path C:\Users\Claudio\AppData\Local\Temp\arduino_build_299844 -warnings=none -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.esptool.path=C:\Users\Claudio\AppData\Local\Arduino15\packages\esp8266\tools\esptool\0.4.9 -prefs=runtime.tools.xtensa-lx106-elf-gcc.path=C:\Users\Claudio\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2 -prefs=runtime.tools.mkspiffs.path=C:\Users\Claudio\AppData\Local\Arduino15\packages\esp8266\tools\mkspiffs\0.1.2 -verbose C:\Users\Claudio\Downloads\Sonoff-Tasmota-master (1)\Sonoff-Tasmota-master\sonoff\sonoff.ino
C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\Claudio\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\Claudio\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\Claudio\Documents\Arduino\libraries -fqbn=esp8266:esp8266:generic:CpuFrequency=80,FlashFreq=40,FlashMode=dio,UploadSpeed=115200,FlashSize=1M64,ResetMethod=ck,Debug=Disabled,DebugLevel=None____ -ide-version=10802 -build-path C:\Users\Claudio\AppData\Local\Temp\arduino_build_299844 -warnings=none -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.esptool.path=C:\Users\Claudio\AppData\Local\Arduino15\packages\esp8266\tools\esptool\0.4.9 -prefs=runtime.tools.xtensa-lx106-elf-gcc.path=C:\Users\Claudio\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2 -prefs=runtime.tools.mkspiffs.path=C:\Users\Claudio\AppData\Local\Arduino15\packages\esp8266\tools\mkspiffs\0.1.2 -verbose C:\Users\Claudio\Downloads\Sonoff-Tasmota-master (1)\Sonoff-Tasmota-master\sonoff\sonoff.ino
Using board 'generic' from platform in folder: C:\Users\Claudio\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0
Using core 'esp8266' from platform in folder: C:\Users\Claudio\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0
Detecting libraries used...
"C:\Users\Claudio\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\Users\Claudio\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0/tools/sdk/include" "-IC:\Users\Claudio\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0/tools/sdk/lwip/include" "-IC:\Users\Claudio\AppData\Local\Temp\arduino_build_299844/core" -c -w -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11  -ffunction-sections -fdata-sections -w -x c++ -E -CC -DF_CPU=80000000L -DLWIP_OPEN_SRC   -DARDUINO=10802 -DARDUINO_ESP8266_ESP01 -DARDUINO_ARCH_ESP8266 -DARDUINO_BOARD="ESP8266_ESP01"  -DESP8266 "-IC:\Users\Claudio\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266" "-IC:\Users\Claudio\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\variants\generic" "C:\Users\Claudio\AppData\Local\Temp\arduino_build_299844\sketch\sonoff.ino.cpp" -o "nul"
"C:\Users\Claudio\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\Users\Claudio\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0/tools/sdk/include" "-IC:\Users\Claudio\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0/tools/sdk/lwip/include" "-IC:\Users\Claudio\AppData\Local\Temp\arduino_build_299844/core" -c -w -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11  -ffunction-sections -fdata-sections -w -x c++ -E -CC -DF_CPU=80000000L -DLWIP_OPEN_SRC   -DARDUINO=10802 -DARDUINO_ESP8266_ESP01 -DARDUINO_ARCH_ESP8266 -DARDUINO_BOARD="ESP8266_ESP01"  -DESP8266 "-IC:\Users\Claudio\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266" "-IC:\Users\Claudio\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\variants\generic" "-IC:\Users\Claudio\Documents\Arduino\libraries\PubSubClient" "C:\Users\Claudio\AppData\Local\Temp\arduino_build_299844\sketch\sonoff.ino.cpp" -o "nul"
"C:\Users\Claudio\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\Users\Claudio\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0/tools/sdk/include" "-IC:\Users\Claudio\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0/tools/sdk/lwip/include" "-IC:\Users\Claudio\AppData\Local\Temp\arduino_build_299844/core" -c -w -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11  -ffunction-sections -fdata-sections -w -x c++ -E -CC -DF_CPU=80000000L -DLWIP_OPEN_SRC   -DARDUINO=10802 -DARDUINO_ESP8266_ESP01 -DARDUINO_ARCH_ESP8266 -DARDUINO_BOARD="ESP8266_ESP01"  -DESP8266 "-IC:\Users\Claudio\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266" "-IC:\Users\Claudio\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\variants\generic" "-IC:\Users\Claudio\Documents\Arduino\libraries\PubSubClient" "-IC:\Users\Claudio\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\Ticker" "C:\Users\Claudio\AppData\Local\Temp\arduino_build_299844\sketch\sonoff.ino.cpp" -o "nul"
"C:\Users\Claudio\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\Users\Claudio\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0/tools/sdk/include" "-IC:\Users\Claudio\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0/tools/sdk/lwip/include" "-IC:\Users\Claudio\AppData\Local\Temp\arduino_build_299844/core" -c -w -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11  -ffunction-sections -fdata-sections -w -x c++ -E -CC -DF_CPU=80000000L -DLWIP_OPEN_SRC   -DARDUINO=10802 -DARDUINO_ESP8266_ESP01 -DARDUINO_ARCH_ESP8266 -DARDUINO_BOARD="ESP8266_ESP01"  -DESP8266 "-IC:\Users\Claudio\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266" "-IC:\Users\Claudio\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\variants\generic" "-IC:\Users\Claudio\Documents\Arduino\libraries\PubSubClient" "-IC:\Users\Claudio\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\Ticker" "-IC:\Users\Claudio\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266WiFi\src" "C:\Users\Claudio\AppData\Local\Temp\arduino_build_299844\sketch\sonoff.ino.cpp" -o "nul"
"C:\Users\Claudio\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\Users\Claudio\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0/tools/sdk/include" "-IC:\Users\Claudio\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0/tools/sdk/lwip/include" "-IC:\Users\Claudio\AppData\Local\Temp\arduino_build_299844/core" -c -w -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11  -ffunction-sections -fdata-sections -w -x c++ -E -CC -DF_CPU=80000000L -DLWIP_OPEN_SRC   -DARDUINO=10802 -DARDUINO_ESP8266_ESP01 -DARDUINO_ARCH_ESP8266 -DARDUINO_BOARD="ESP8266_ESP01"  -DESP8266 "-IC:\Users\Claudio\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266" "-IC:\Users\Claudio\AppData\Local\Arduino15\ [... cut ...]
Uso la libreria ESP8266WebServer alla versione 1.0 nella cartella: C:\Users\Claudio\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266WebServer 
Uso la libreria DNSServer alla versione 1.1.0 nella cartella: C:\Users\Claudio\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\DNSServer 
Uso la libreria ESP8266mDNS nella cartella: C:\Users\Claudio\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266mDNS (legacy)
exit status 1
Errore durante la compilazione per la scheda Generic ESP8266 Module.

joecotton, 2017-05-26:
Did you install the external libraries as described in the prerequisites instructions: https://github.com/arendst/Sonoff-Tasmota/wiki/Prerequisite#installing-external-libraries

claudioita, 2017-05-26:
Yes I did install pubsubclient and arduinojson (the other I put // as per Bruh video)

davidelang, 2017-05-26:
well, since it's complaining about not finding the pubsubclient, it looks like 
something went wrong when installing them.

Try installing them again, and tell us exactly what you are doing and what 
errors you get when you do so.

claudioita, 2017-05-26:
after your message I renamed the directory PubSubClient in pubsubclient (I am in windows). DOes that influence?

claudioita, 2017-05-26:
once is installed it does not let me install again (not even uninstalling)

claudioita, 2017-05-26:
I managed to install 2.5.0 and then update to 2.6.0

BTW I also have PubSubCLient (imroy) also installed, maybe I have to uninstall it? How?

claudioita, 2017-05-26:
LOG

Arduino:1.8.2 (Windows 10), Scheda:"Generic ESP8266 Module, 80 MHz, 40MHz, DIO, 115200, 1M (64K SPIFFS), ck, Disabled, None"

C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\Claudio\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\Claudio\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\Claudio\Documents\Arduino\libraries -fqbn=esp8266:esp8266:generic:CpuFrequency=80,FlashFreq=40,FlashMode=dio,UploadSpeed=115200,FlashSize=1M64,ResetMethod=ck,Debug=Disabled,DebugLevel=None____ -ide-version=10802 -build-path C:\Users\Claudio\AppData\Local\Temp\arduino_build_757631 -warnings=none -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.esptool.path=C:\Users\Claudio\AppData\Local\Arduino15\packages\esp8266\tools\esptool\0.4.9 -prefs=runtime.tools.mkspiffs.path=C:\Users\Claudio\AppData\Local\Arduino15\packages\esp8266\tools\mkspiffs\0.1.2 -prefs=runtime.tools.xtensa-lx106-elf-gcc.path=C:\Users\Claudio\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2 -verbose C:\Users\Claudio\Downloads\Sonoff-Tasmota-master (1)\Sonoff-Tasmota-master\sonoff\sonoff.ino
C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\Claudio\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\Claudio\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\Claudio\Documents\Arduino\libraries -fqbn=esp8266:esp8266:generic:CpuFrequency=80,FlashFreq=40,FlashMode=dio,UploadSpeed=115200,FlashSize=1M64,ResetMethod=ck,Debug=Disabled,DebugLevel=None____ -ide-version=10802 -build-path C:\Users\Claudio\AppData\Local\Temp\arduino_build_757631 -warnings=none -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.esptool.path=C:\Users\Claudio\AppData\Local\Arduino15\packages\esp8266\tools\esptool\0.4.9 -prefs=runtime.tools.mkspiffs.path=C:\Users\Claudio\AppData\Local\Arduino15\packages\esp8266\tools\mkspiffs\0.1.2 -prefs=runtime.tools.xtensa-lx106-elf-gcc.path=C:\Users\Claudio\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2 -verbose C:\Users\Claudio\Downloads\Sonoff-Tasmota-master (1)\Sonoff-Tasmota-master\sonoff\sonoff.ino
Using board 'generic' from platform in folder: C:\Users\Claudio\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0
Using core 'esp8266' from platform in folder: C:\Users\Claudio\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0
Detecting libraries used...
"C:\Users\Claudio\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\Users\Claudio\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0/tools/sdk/include" "-IC:\Users\Claudio\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0/tools/sdk/lwip/include" "-IC:\Users\Claudio\AppData\Local\Temp\arduino_build_757631/core" -c -w -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11  -ffunction-sections -fdata-sections -w -x c++ -E -CC -DF_CPU=80000000L -DLWIP_OPEN_SRC   -DARDUINO=10802 -DARDUINO_ESP8266_ESP01 -DARDUINO_ARCH_ESP8266 -DARDUINO_BOARD="ESP8266_ESP01"  -DESP8266 "-IC:\Users\Claudio\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266" "-IC:\Users\Claudio\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\variants\generic" "C:\Users\Claudio\AppData\Local\Temp\arduino_build_757631\sketch\sonoff.ino.cpp" -o "nul"
"C:\Users\Claudio\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\Users\Claudio\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0/tools/sdk/include" "-IC:\Users\Claudio\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0/tools/sdk/lwip/include" "-IC:\Users\Claudio\AppData\Local\Temp\arduino_build_757631/core" -c -w -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11  -ffunction-sections -fdata-sections -w -x c++ -E -CC -DF_CPU=80000000L -DLWIP_OPEN_SRC   -DARDUINO=10802 -DARDUINO_ESP8266_ESP01 -DARDUINO_ARCH_ESP8266 -DARDUINO_BOARD="ESP8266_ESP01"  -DESP8266 "-IC:\Users\Claudio\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266" "-IC:\Users\Claudio\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\variants\generic" "-IC:\Users\Claudio\Documents\Arduino\libraries\PubSubClient" "C:\Users\Claudio\AppData\Local\Temp\arduino_build_757631\sketch\sonoff.ino.cpp" -o "nul"
"C:\Users\Claudio\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\Users\Claudio\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0/tools/sdk/include" "-IC:\Users\Claudio\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0/tools/sdk/lwip/include" "-IC:\Users\Claudio\AppData\Local\Temp\arduino_build_757 [... cut ...]
Uso la libreria ESP8266WebServer alla versione 1.0 nella cartella: C:\Users\Claudio\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266WebServer 
Uso la libreria DNSServer alla versione 1.1.0 nella cartella: C:\Users\Claudio\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\DNSServer 
Uso la libreria ESP8266mDNS nella cartella: C:\Users\Claudio\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266mDNS (legacy)
exit status 1
Errore durante la compilazione per la scheda Generic ESP8266 Module.

davidelang, 2017-05-26:
C:\Users\Claudio\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/WString.h:260: 
undefined reference to `PubSubClient::publish(char const*, char const*, unsigned 
char)'

This says that it still isn't seeing the PubSubClient library

claudioita, 2017-05-27:
I deleted the directory, somehow removed the library (can't remember how), and reinstalled it. Now it compiled  (did not flash though, because of other errors)

closed 2018-05-08 by stale[bot]