arendst/Tasmota #483: Orvibo S20

dontpanic21, 2017-05-27:
Hi, just wondering if anyone has worked out the pins to program the Orvibo S20, I assume its a very similar device and I think it was around before the sonoff. There are a couple of variants, one has a large pin header already soldered and the other has 3 unsoldered sets of holes.

smadds, 2017-05-29:
Not sure they are that similar inside, @dontpanic21 . 
https://blog.slange.co.uk/orvibo-s20-wifi-power-socket/

dontpanic21, 2017-06-03:
Thanks for the reply, maybe they aren't the same. The difficulty is that the Orvibo has the external pins soldered to the board and the board cannot be pulled out to see the underside, so I can't confirm.

bruce34, 2017-06-09:
Is it possible you have the Orvibo B25? I ordered more replacement S20s but got B25s instead. If so, I've put pictures/pinouts on https://blog.slange.co.uk/orvibo-b25-wifi-power-socket/ and the sonoff_template.h works apart from the LED not being fully functional

```
  {
     "Orvibo S20",      // Orvibo S20 v2 (ESP8266)
     GPIO_USER,        // GPIO00 (3L) Marked "100" on the 3.3V RX TX 100 GND pads
     GPIO_USER,        // GPIO01 (3B) Serial RXD and Optional sensor
     GPIO_USER,        // GPIO02 (4L)
     GPIO_USER,        // GPIO03 (5B) Serial TXD and Optional sensor
     GPIO_LED1_INV,    // GPIO04 (2L) LED - TODO (can be red/blue dep on direction)
     GPIO_REL1,        // GPIO05 Relay
     0,                // GPIO06 (SD_CLK   Flash)
     0,                // GPIO07 (SD_DATA0 Flash QIO/DIO/DOUT)
     0,                // GPIO08 (SD_DATA1 Flash QIO/DIO)
     0,                // GPIO09 (SD_DATA2 Flash QIO)
     0,                // GPIO10 (SD_DATA3 Flash QIO)
     0,                // GPIO11 (SD_CMD   Flash)
     GPIO_LED1,        // GPIO12 (3R) - TODO (can be red/blue dep on direction)
     GPIO_USER,        // GPIO13 (1B)
     GPIO_KEY1,        // GPIO14 Switch
     GPIO_USER,        // GPIO15 (5L)
     0,                // GPIO16
     0                 // ADC0 Analog input
  }
```

closed 2018-05-08 by stale[bot]