arendst/Tasmota #484: Flashing/Plusing when pin14 is connected to ground

aidancrane, 2017-05-27:
I have just flashed and installed Sonoff-Tasmota in a lightswitch in the house. I also attached pin14 to the old light switch to also have the ability to use the old light switch to turn it on and off.

From the console, or MQTT the bug is not present, I can turn the switch on and off no trouble when the wall switch is off.

However, when I turn on or off the light from the wall switch, it pulses for less than a couple seconds all the time, and it never ends, I've tried disabling MQTT.

```
12:04:09 MQTT: stat/sonoff/RESULT = {"POWER":"ON"}
12:04:09 MQTT: stat/sonoff/POWER = ON
12:04:10 Config: Saved configuration (1492 bytes) to flash at FB and count 157
12:04:16 Wifi: Checking connection...
12:04:16 Wifi: Connected
12:04:16 MQTT: stat/sonoff/RESULT = {"POWER":"OFF"}
12:04:16 MQTT: stat/sonoff/POWER = OFF
12:04:16 MQTT: stat/sonoff/RESULT = {"POWER":"ON"}
12:04:16 MQTT: stat/sonoff/POWER = ON
12:04:20 MQTT: stat/sonoff/RESULT = {"POWER":"OFF"}
12:04:20 MQTT: stat/sonoff/POWER = OFF
12:04:20 MQTT: stat/sonoff/RESULT = {"POWER":"ON"}
12:04:20 MQTT: stat/sonoff/POWER = ON
```
12:04:09 is when I flipped the wall switch from off to on, the rest is of the switches own accord, does anyone know why this is happening?

khcnz, 2017-05-27:
I have some units setup in the same fashion so it should work - I assume your light swtich is a proper toggle switch not a momentary on/off unit?

Can you confirm
1.  Switch was setup as a "Switch" (not a button) 
2. What switchmode you are running in (should be toggle)? 
3. Confirm you haven't set a pulsetime

You can do this from the console on the webportal by issuing the commands
1. GPIO
2. SwitchMode1
3. PulseTime1

aidancrane, 2017-05-28:
Thank you for your reply, the sonoff basic module is setup as follows.
[image]

This is the result from the commands you asked me to run. Since I posted the issue, I have re-flashed the sonoff to try to rectify the issue, but It still remains but the interval between flashing is much longer now sometimes minutes apart.
```
09:50:53 CMND: GPIO
09:50:53 RSLT: stat/sonoff/RESULT = {"GPIO1":0 (None), "GPIO3":0 (None), "GPIO4":0 (None), "GPIO14":9 (Switch1)}
09:51:09 CMND: SwitchMode1
09:51:09 RSLT: stat/sonoff/RESULT = {"SwitchMode1":0}
09:51:10 UPnP: Multicast (re)joined
09:51:10 CMND: PulseTime1
09:51:10 RSLT: stat/sonoff/RESULT = {"PulseTime1":0}
```

The Switch is a standard household switch that latches and follows this schematic, that I obtained from [here](https://www.reddit.com/r/homeautomation/comments/64jpht/sonoff_basic_and_onewire_switches_eg_uk/). The setup is a little different as I live in the UK, where no neutral is present at the switch but follows the same principles as the UK.

[image]
The sonoff is mounted above a ceiling rose in the loft, and then loops back into the room.
[image]

davidelang, 2017-05-28:
switchmode 0 tells it that you want to toggle when it sees a connection, you probably want switchmode set to follow (1) see https://github.com/arendst/Sonoff-Tasmota/wiki/Commands

khcnz, 2017-05-28:
@davidelang depends if you are likely to 'remotely' turn on/off often - as then toggle makes more sense (which makes it work similar to a 2 way switch) - a flick of the switch will always change the state - but up or down on the actual wall switch doesn't mean anything. At least that is what i have done.

@aidancrane Your diagram seems to make sense to me and rest of setup looks ok. I can only imagine here that there is some cross talk/induction due to the length of cable from the rose to the switch (how long is this?). Could mean that it floats of GND sometimes even when switch closed. Also just want to check if you disable switch one it doesn't turn on/off by itself?

khcnz, 2017-05-28:
Presuming you eliminate any possibility something other than the switch is causing the on/off then suggest reading this https://electronics.stackexchange.com/questions/49824/microcontroller-with-a-long-wire-for-digital-input/49825

aidancrane, 2017-05-28:
Ok, I have been up into the ceiling and tried a dead short across the pins, low and behold the issue is gone, therefore it must be that the problem exists due to voltage drop from the switch as you had said. The distance to the switch is quite short, only 6-7 meters round trip, but seems to be enough to cause the flickering.

Now this is more of an electronics question so may not be the best place to ask, but I don't really want to re-wire the switch and would rather use the existing wire. After reading the stackexchange question, would my best bet be to run a boost converter in line with pin14? say 3v to 5v?

khcnz, 2017-05-28:
No, I think you want to follow the idea of trying a capacitor first (which is mentioned in link above and also [here ](https://raspberrypi.stackexchange.com/questions/24895/i-am-getting-ghosting-bouncing-on-my-digital-input)and [here](https://www.raspberrypi.org/forums/viewtopic.php?t=53548)). I am more of a s/w guy than a h/w guy so if that doesn't work you may need more help than i can provide.

aidancrane, 2017-05-28:
Ok, thank you, I will try a capacitor and if it fails I will try other means.

closed 2017-05-28 by aidancrane

davidelang, 2017-05-28:
add a small bypass capacitor to the line (between gpio14 and ground), that will 
reduce the noise on the line. You may also want to attach a pull-up resistor on 
the line to make it more solidly non-zero when the switch is open.

adding a 5v converter would not help, because when the switch is closed, it's 
not seeing 3.3v, it's connecting to ground.

DNA99, 2017-07-08:
[image]
@davidelang  Hi, sorry I need some visuals to get this right. Is the attached diagram correct for the placement of the capacitor and resistor? What size capacitor and resistor do I use?

DNA99, 2017-07-08:
[image]
@davidelang @khcnz Here is my diagram. Please help!

davidelang, 2017-07-08:
that looks correct, the value of the resistor and capacitor don't really matter much

the resistor needs to be large enough that it doesn't eat a lot of power.

the capacitor needs to be small enough that it's charge time isn't large enough to be noticable.

DNA99, 2017-07-09:
@davidelang what puzzles me is that this happens only with this circuit with a 2 way switch. I have another 3 on a single gang switch and they seem to work fine. Do you think it is necessary to put the resistor and capacitor on the other sonoffs as well?

davidelang, 2017-07-09:
the resistor is needed when using pin 14 (st least most of the time)

the capacitor is needed when the attached wire gets 'too long' or 'too near 
sources of itnerference', so you don't need it a lot of the time, but the longer 
the wire, the more the chances are that you will need it.

DNA99, 2017-07-09:
@davidelang could you give me the specification of the capacitor i need to get? I tried looking for one and there is a ton of them out there and they have voltage as well. The smallest voltage capacitors i could find was 4v and 40uF and i got a couple of 0.25w 10k om resistors. Does that sound right to you?

davidelang, 2017-07-09:
On Sun, 9 Jul 2017, DNA99 wrote:

> @davidelang could you give me the specification of the capacitor i need to
> get? I tried looking for one and there is a ton of them out there and they
> have voltage as well. The smallest voltage [...]

those will probably do, the ESP-8266 is a 3.3v system so the 4v cap will 
probably work, 10k resistors should be just fine.

DNA99, 2017-07-09:
Thanks pal! Now to remove the sonoffs and back to the soldering iron.

DNA99, 2017-07-09:
[image]
@davidelang  I haven't got the capacitors as they were out of stock. and I just tried soldering a 10k resistor between GPIO14 and 3.3v as in the pic and it seems to be working fine now! Do you know if it makes any difference if you use 10k resistors and say 7k? Thank you for the help!

davidelang, 2017-07-09:
I've seen people use anything from about 3K up past 10K, the value is not 
critical. The smaller the resistor, the more current will be pulled from the pin 
when the switch is closed

v=ir

v=3.3v
if r=1k then i=0.033 amps, I believe the pins are ratd at 0.02A, so this would 
be too small, but if r=2k, that would drop it to 0,017A, which would not burn 
out the chip, but would be quite a bit of draw.

David Lang