arendst/Tasmota #485: Sonoff Touch compiles but is to large to flash.

chuxxsss, 2017-05-30:
Hi,
    My problem is the Sonoff Touch cannot save flash program as it is to large.

Error: The program size (479877 bytes) is greater than maximum allowed (434160 bytes)
Error Took 27.37 seconds
=========================================================
Enviroment sonoff                                       [ERROR]
Enviroment sonoff-touch-4ch                     [ERROR]
Enviroment esp8285                                    [ERROR]

================[ERROR} Took 113.74 seconds ======================

I used Atom/Platformio to do this as well. 
https://www.youtube.com/watch?v=FoJu-UNPrJU

Anyway of shorting the program would be helpful.  I use Node-Red as my MQTT server so don't need the other stuff.  Thank you for any help in advance, and it look like a great program.

queuetue, 2017-05-30:
@chuxxsss There are 6 places in user_config.h that you can make the code smaller - search for "Disable by"

arendst, 2017-05-30:
No need to disable anything as you do not use the correct platform io environment. 

Install patformio.ini and linker script from this repository and read the wiki how to setup your platformio enironment.

Bottomline is to never use esp8285 as board.

chuxxsss, 2017-05-31:
Hi @arendst OK I understand Install Platformio.ini. Where do I find the linker script?  So is the wiki here or somewhere else?   So on the board the ESP8285, you are saying use the ESP8266 how do I set DOUT= Data out?  Thank you for your help really appreciate it mate.

davidelang, 2017-05-31:
the wiki is on github (go to the top of the page), I believe the linker script 
is in prerequisits (or something close to tehre)

chuxxsss, 2017-05-31:
Had lots of problem with MAC OS X. I have installed on Linux Mint now. Cannot find the location on the file Platformio.ini . Did find the program after installing Atom and installing clang "sudo apt-get install clang" .
The main program is under the home directory ~/.platformio . Any help finding the file would be great. I did think it would be under root etc/ as a script file but no joy. Could the script file be in the main zip file please.
Thank you for all the help Chris

chuxxsss, 2017-06-01:
Ok, The file is in the zip file. Still fails upload with Error 255 plus 
Warning! Please install '99-platformio-udev-rules and check that your board's PID and VID are listed in the rules.
https://raw.githubusercontent.com/platformio/platformio/develop/scripts/99-platformio-dev.rules
Looking for upload port...
Auto-detected: /dev/ttyUSB0
Uploading .pioenvs/sonoff-touch-4ch/firmware.bin
error: espcomm_sync failed
error: espcomm_open failed
error: espcomm_upload-mem failed
upload Error 255
Error Took 22.17 seconds
Environment sonoff           [ERROR]
Environment sonoff-touch-4ch  [ERROR]

ERROR took 47.66 seconds
So could this be the data lines or something else.

Please any help would be gratefully appreciated.

chuxxsss, 2017-06-06:
I think the problem is the esptool.py still working on it.

Ghostbird, 2017-08-12:
I just uploaded Sonoff-Tasmota without issues using the PlatformIO extension for Visual Studio Code on Debian 9. I had no issues at all. I'll update the wiki article a bit to clarify some things. I spend some time looking for a correct way to upload it, since the chip is a PSF-A85 (ESP-8285 based) instead of an ESP-8266, but the Upload script for PlatformIO automatically detects and compensates for that.

davidelang, 2017-08-13:
make sure you have the flash size set correctly (1m no spiffs), this will 
support images up to about 512k

If the image you are compiling is larger than this, you need to turn off some 
options to make it fit.

beyond this, we would need more info.

chuxxsss, 2017-08-16:
I think I used spiffs me bad. Will try without and get back to you. Thank
you for your time and efforts. Have a wonderful day.

closed 2018-05-07 by stale[bot]