arendst/Tasmota #486: Wiki: Split up Hardware Preparation

ThomDietrich, 2017-05-30:
Hey @arendst,
seems like the [Hardware Preparation](https://github.com/arendst/Sonoff-Tasmota/wiki/Hardware-Preparation) page is growing out of it's size while the device specific articles (currently titled "Tips") do only contain little content. If you agree I'd move all device specific preparation content into the specific articles under a title "Devices".

arendst, 2017-05-30:
Yes please do.

ThomDietrich, 2017-05-30:
Sonoff Dual done. https://github.com/arendst/Sonoff-Tasmota/wiki/Sonoff-Dual (you might want to copy the second image over to your media folder)

ThomDietrich, 2017-05-31:
Sonoff Pow done. I've also tuned the first part of the Hardware Preparation article.

Whoever is reading this, feel free to follow along with any of the other modules.

ThomDietrich, 2017-06-01:
Sonoff Touch done.

ThomDietrich, 2017-06-23:
Sonoff 4CH done.

Next up Sonoff (basic) probably the most important.

ThomDietrich, 2017-07-20:
@arendst regarding: https://github.com/arendst/Sonoff-Tasmota/wiki/Sonoff-Pow

Sorry I removed this warning you've readded. However I'm not sure why it is needed. The warning is no different for all other modules and I've added it to the general Hardware Preparations article long ago. All module specific articles are referencing this article and hence are already linked to this warning.

arendst, 2017-07-20:
The problem with a pow is that ANY connection, even gnd only could blow your PC as AC mignt blow it's power supply. That's why only opto coupling is safe.

For all other sonoffs serial connection is safe as long the Vcc is not connected to your PC.

closed 2018-05-07 by stale[bot]