arendst/Tasmota #487: Sonoff POW v 2.0 has 4MB Flash

danilogd, 2017-05-30:
I bought a bunch of Sonoff POW a week ago and all have a 32Mbit Flash chip instead of the usual 8Mbit.
What changes are needed to adapt to this new flash capacity?

Here are two photos of the new chips:

[image]
[image]

Thank you in advance.

davidelang, 2017-05-30:
If the only difference is the flash size, the existing builds should work just 
fine. Some of the other devices have 4MB flash and normally we just ignore the 
extra capacity.

khcnz, 2017-05-31:
This is the standard sonoff pow build - but since the desire is to run the same build on all sonoffs the normal advice is just to use the 1mb mode for all devices - some more info here https://github.com/arendst/Sonoff-Tasmota/issues/437

closed 2017-05-31 by danilogd

dimitrisA, 2018-01-04:
Hello. Could you please share where you bought these 4 Mb Sonoff POWs ? Thanks! :)