arendst/Tasmota #491: Twitter and IFTTT

stoehrmark, 2017-06-02:
Would there be any way to integrate Twitter? Maybe with IFTTT? Have the device send out a tweet when its status changes. Here is what I want to do. I want to use a Sonoff RF to send a Tweet when the button is pushed. The button could be used as a “help Button” or request button. 
Here is an example of come code but I don’t know how to add it.
https://bigjungle.net/blog/2015/6/26/connect-maker-to-anything-ifttt

Killobite, 2017-06-03:
Yes there is, but it will take some extra "gear".  If you are using a Raspberry Pi to do anything around the house, say the MQTT broker, HA Host or anything else there is something called 'Node Red' already ready and waiting on the Pi.

There is an MQTT node and a Twitter node that you can combine and have it send out a tweet when the state changes on the Sonoff.  I use Node Red to poll and send e-mail updates to control some of my home automation items using Alexa and IFTTT.

stoehrmark, 2017-06-04:
Yes MQTT and Node-RED are a good way to do this but i was hoping for a simple one device solution. I have MQTT and Node-RED setup at my home and it works great for my home automation needs. And i know that it can tweet, email, etc, etc  and do lots of other great things. I want to make this device for my mother at her home as a  "I've fallen and I can't get up!" assistance button. I don't want to complicate thing by adding a Pi. If the Sonoff-Tasmota code could directly talk to IFTTT then the Tweet could simply be sent from one device. Looks like the Sonoff-Tasmota code would only need the following code below  (i'm guessing with some modified) and some GUI additions. Adding a direct connection to IFTTT i'm sure would open up a whole new level of functionality to this already excellent code. There are endless possibility with IFTTT. (I have nothing to do with IFTTT. Just think it would be good.)

`void send_event(const char *event)
{
  // set the LED on whle we are sending the event
  digitalWrite(ledPin, HIGH);

  Serial.print("Connecting to ");
  Serial.println(host);
  
  // Use WiFiClient class to create TCP connections
  WiFiClient client;
  const int httpPort = 80;
  if (!client.connect(host, httpPort)) {
    Serial.println("Connection failed");
    return;
  }
  
  // We now create a URI for the request
  String url = "/trigger/";
  url += event;
  url += "/with/key/";
  url += privateKey;
  
  Serial.print("Requesting URL: ");
  Serial.println(url);
  
  // This will send the request to the server
  client.print(String("GET ") + url + " HTTP/1.1\r\n" +
               "Host: " + host + "\r\n" + 
               "Connection: close\r\n\r\n");

  // Read all the lines of the reply from server and print them to Serial,
  // the connection will close when the server has sent all the data.
  while(client.connected())
  {
    if(client.available())
    {
      String line = client.readStringUntil('\r');
      Serial.print(line);
    } else {
      // No data yet, wait a bit
      delay(50);
    };
  }
  
  // All done
  Serial.println();
  Serial.println("closing connection");

  client.stop();
  
  // Finished sending the message, turn off the LED
  digitalWrite(ledPin, LOW);
}`

stoehrmark, 2017-06-05:
Here is some more info on the IFTTT connection. Seems it is as simple as one HTTP request. I used the Maker Webhooks service. They provide a KEY and you generate a "event" name (my example test was  button_pushed) So my test URL was 

https://maker.ifttt.com/trigger/button_pushed/with/key/XXXXXXXXkeyXXXXXXX

**Below taken from IFTTT website**
Your key is: XXXXXXXXkeyXXXXXXX
To trigger an Event
Make a POST or GET web request to:

https://maker.ifttt.com/trigger/{event}/with/key/XXXXXXXXkeyXXXXXXX

With an optional JSON body of:
{ "value1" : "button_pushed", "value2" : "", "value3" : "" }
The data is completely optional, and you can also pass value1, value2, and value3 as query parameters or form variables. This content will be passed on to the Action in your Recipe.
You can also try it with curl from a command line.
curl -X POST -H "Content-Type: application/json" -d '{"value1":"button_pushed"}' https://maker.ifttt.com/trigger/{event}/with/key/XXXXXXXXkeyXXXXXXX

sebastianlorenzen, 2017-06-14:
if there is still space for this piece of code on the sonoffs that would be a cool feature

MaciejTarmas, 2017-06-20:
Take a look at scripting in ESP Easy. You can make it send an HTTP request when something is triggered:

https://www.letscontrolit.com/wiki/index.php/Tutorial_Rules#SendToHTTP

Would be nice if Tasmota had similar funcionality.

closed 2018-05-08 by stale[bot]