arendst/Tasmota #492: sonoff pow uses.

terryrow, 2017-06-02:
Can the sonoff pow switch a remote sonoff th 16 using the "max power" variable?
This is not available using current Itead firmware/app.
My aim is to charge my electric car via a socket, switched on by sonoff 16, when the solar panel power exceeds a threshold Kw level then off when Kw drops to a lower level with some sort of hysteresis delay to avoid too frequent switching.

davidelang, 2017-06-03:
no, the max power type of stuff only works on the local device. You can 
implement this sort of logic using your home automation software.
k

terryrow, 2017-06-03:
Thanks, these micro processors are a totally new area for me. The code I saw seemed to have the watts , amps, max power as variables in an If statement so I assumed it would be possible to send a wireless "switch on" to a relay if watts>123.. 
I originally looked to use a current loop, transistor amp and solid state relay but the cost of parts made it a none starter compared to sonoff items. The current loop option is still preferred to breaking into the solar cable, especially as the sonoff connectors are so miserable.
Do you have any detailed suggestions as to how I can achieve monitoring the power and wirelessly switching on a relay? Please bear in mind I have no knowledge of these systems.
Thanks againTerry Rowland.

davidelang, 2017-06-03:
http://docs.openhab.org/tutorials/beginner/rules.html
http://www.homeautomationforgeeks.com/openhab_rules.shtml

the second shows sending an e-mail when it gets hot, you can just as easily send 
any command and test the power rather than temp.

terryrow, 2017-06-07:
Thanks David, I will go through those soon.
Cheers Terry Rowland. 

 

 http://docs.openhab.org/tutorials/beginner/rules.html
http://www.homeautomationforgeeks.com/openhab_rules.shtml

closed 2018-05-08 by stale[bot]