arendst/Tasmota #493: Sonoff Basic ignoring commands

dontpanic21, 2017-06-03:
Hi, I have a few sonoff basics that I have flashed up successfully and they all have the same symptom which is that I have to send the power command several times before the power will switch. Not all the time however, sometimes they switch on the first command sent.

I have them connected to openhab, but the symptom is the same whether I use openhab or mosquitto_pub to send the power command, they are equally unreliable. Whether powered from PC USB or A/C the result is the same.

The button works pretty much all the time.

I have the serial monitor in Arduino connected, and at the same time I have mosquitto_sub receiving tele updates and mosquitto_sub receives every "online" tele update that I see in the serial monitor, so I have no reason to suspect the wifi. 

With the serial monitor I can see that only about 1 in 5 commands are acknowledged. I have tried setting qos level on the command publish to 1 and 2 with no change.

Any ideas? Is there some extra debugging I can do?

thanks

davidelang, 2017-06-03:
see the troublshooting page on the wiki

dontpanic21, 2017-06-03:
Thanks, I have read the trouble shooting and the only thing that appears relevant is to do some higher level debugging, the other issues described there don't apply, as far as I can see. I have enabled LOG_LEVEL_ALL on the serial port and below is the log. during the time period where the tele data is being received I have sent the same mosquitto_pub command several times from 02:32:39, but it is not until 02:32:57 that the sonoff seems to acknowledge reception of the command and changes state

02:32:39 MQTT: cmnd/SONOFF_AV02/POWER = 
02:32:40 MQTT: Attempting connection...
02:32:41 mDNS: Query done with 0 mqtt services found
02:32:41 MQTT: Connected
02:32:41 MQTT: tele/SONOFF_AV02/LWT = Online (retained)
02:32:41 MQTT: cmnd/SONOFF_AV02/POWER = 
02:32:43 MQTT: Attempting connection...
02:32:44 mDNS: Query done with 0 mqtt services found
02:32:44 MQTT: Connected
02:32:44 MQTT: tele/SONOFF_AV02/LWT = Online (retained)
02:32:44 MQTT: cmnd/SONOFF_AV02/POWER = 
02:32:45 MQTT: Attempting connection...
02:32:46 mDNS: Query done with 0 mqtt services found
02:32:46 MQTT: Connected
02:32:46 MQTT: tele/SONOFF_AV02/LWT = Online (retained)
02:32:46 MQTT: cmnd/SONOFF_AV02/POWER = 
02:32:48 MQTT: Attempting connection...
02:32:49 mDNS: Query done with 0 mqtt services found
02:32:49 MQTT: Connected
02:32:49 MQTT: tele/SONOFF_AV02/LWT = Online (retained)
02:32:49 MQTT: cmnd/SONOFF_AV02/POWER = 
02:32:50 Wifi: Checking connection...
02:32:50 Wifi: Connected
02:32:50 MQTT: Attempting connection...
02:32:51 mDNS: Query done with 0 mqtt services found
02:32:51 MQTT: Connected
02:32:51 MQTT: tele/SONOFF_AV02/LWT = Online (retained)
02:32:51 MQTT: cmnd/SONOFF_AV02/POWER = 
02:32:53 MQTT: Attempting connection...
02:32:54 mDNS: Query done with 0 mqtt services found
02:32:54 MQTT: Connected
02:32:54 MQTT: tele/SONOFF_AV02/LWT = Online (retained)
02:32:54 MQTT: cmnd/SONOFF_AV02/POWER = 
02:32:55 MQTT: Attempting connection...
02:32:56 mDNS: Query done with 0 mqtt services found
02:32:56 MQTT: Connected
02:32:56 MQTT: tele/SONOFF_AV02/LWT = Online (retained)
02:32:56 MQTT: cmnd/SONOFF_AV02/POWER = 
02:32:57 RSLT: Receive topic cmnd/SONOFF_AV02/POWER, data size 2, data ON
02:32:57 RSLT: DataCb Topic SONOFF_AV02, Group 0, Index 1, Type POWER, Data ON (ON)
02:32:57 MQTT: stat/SONOFF_AV02/RESULT = {"POWER":"ON"}
02:32:57 MQTT: stat/SONOFF_AV02/POWER = ON
02:32:57 Config: Saved configuration (1364 bytes) to flash at F7 and count 4
02:32:58 MQTT: Attempting connection...

davidelang, 2017-06-03:
during the timeframe when it's attempting a connection to the mqtt server, it's 
not able to respond at all (due to a bug in the arduino TCP library), so all 
those "attempting connection" messages indicate that it's having trouble 
connecting to mosquitto. does anything show up on the mosquitto side in it's 
logs?

dontpanic21, 2017-06-03:
below is the mosquitto log. It appears I have something odd going on in my router. The router has 2 dhcp devices on the same mac address. I'm try to sort that out, if I can't I'll set the address in the device. I'll let you know how I go.

I'm assuming its this odd routing behaviour thats causing the MQTT reconnects seen in the serial log?

1496455344: New client connected from 192.168.88.4 as SONOFF (c1, k15, u'SONOFF_USER').
1496455346: New connection from 192.168.88.12 on port 1883.
1496455346: Client SONOFF already connected, closing old connection.
1496455346: Client SONOFF disconnected.
1496455346: New client connected from 192.168.88.12 as SONOFF (c1, k15, u'SONOFF_USER').
1496455347: New connection from 192.168.88.4 on port 1883.
1496455347: Client SONOFF already connected, closing old connection.
1496455347: Client SONOFF disconnected.
1496455347: New client connected from 192.168.88.4 as SONOFF (c1, k15, u'SONOFF_USER').
1496455349: New connection from 192.168.88.12 on port 1883.
1496455349: Client SONOFF already connected, closing old connection.
1496455349: Client SONOFF disconnected.
1496455349: New client connected from 192.168.88.12 as SONOFF (c1, k15, u'SONOFF_USER').
1496455350: New connection from 192.168.88.4 on port 1883.
1496455350: Client SONOFF already connected, closing old connection.
1496455350: Client SONOFF disconnected.
1496455350: New client connected from 192.168.88.4 as SONOFF (c1, k15, u'SONOFF_USER').
1496455352: New connection from 192.168.88.12 on port 1883.
1496455352: Client SONOFF already connected, closing old connection.
1496455352: Client SONOFF disconnected.
1496455352: New client connected from 192.168.88.12 as SONOFF (c1, k15, u'SONOFF_USER').
1496455353: New connection from 192.168.88.4 on port 1883.
1496455353: Client SONOFF already connected, closing old connection.
1496455353: Client SONOFF disconnected.
1496455353: New client connected from 192.168.88.4 as SONOFF (c1, k15, u'SONOFF_USER').
1496455355: New connection from 192.168.88.12 on port 1883.

davidelang, 2017-06-03:
this is saying that you have two devices using the name SONOFF and each time one 
connects, mosquitto disconnects the other one.

make sure each one has a unique name.
k

dontpanic21, 2017-06-03:
yes, thats what I am trying to work out. I definitely only have one sonoff connected, however in the router DHCP information there are 2 devices, and they are actually on different mac addresses and they have different names AV01 and AV02. If I power off the sonoff those mosquitto log messages stop completely, when I power it on both the .4 and .12 messages start appearing. its almost as if i have 2 sonoffs in one!

dontpanic21, 2017-06-03:
I just realised there are 2 devices connected. In the mosquitto log message it says new client connected .... as SONOFF. is that the client_id thats been duplicated? the two devices are named AV01 and AV02 (those are the host names)

davidelang, 2017-06-03:
change the name on your sonoff and see what happens.

k

davidelang, 2017-06-03:
exactly, the client_id is the same, it's not the name in DNS that matters, it's 
what the device self-identifies itself as.

dontpanic21, 2017-06-03:
Fixed, thanks :) silly mistake, I thought it was just the DNS name that had to be unique.

davidelang, 2017-06-03:
add a note to the troubleshooting page to explain this problem and close this 
issue if you don't mind.

dontpanic21, 2017-06-03:
I've added, feel free to edit if I've used incorrect terminology or anything else. thanks for the help.

closed 2017-06-03 by dontpanic21