arendst/Tasmota #494: [Sonoff Dual] External switches

dariosm, 2017-06-03:
Hi, I've been searching here if there is any way to make use of the pin header available OTB, labeled as GND, BTN1, BTN2, GND. It seems like it is not supported by this firmware's project, and I'm wondering if it is planned to be, since I flashed several Sonoff single and I pretend to use the same fw in each module, even in the dual.

In any case, I also wonder what is the original purpose of the mentioned pin header. What can it be used for?

Thanks in advance

davidelang, 2017-06-03:
it looks like nobody has taken the time to track down what those header pins are 
connected to. If you can trace them to the chip, we can figure out what gpio 
they are.

arendst, 2017-06-03:
Well you could take a look at the schematics of the Sonoff Dual to find out what it means. 

Deducing the fact that the Single button on Sonoff Dual is connected as Button 3 chances are that Grounding pin BTN1 might toggle relay0 and grounding pin BTN2 might toggle relay1. It will also send a signal to the esp8266 chip which translates it as a toggle of the Single button and will toggle the relay back. So no real functionality here without major esp8266 code change.

Both relays and buttons are under control of a seperate chip which makes integration with Tasmota a bit of a hassle.

dariosm, 2017-06-03:
I've made some tests with a toggle button connected like BTN1-GND and
BTN0-GND switches.

I first performed a single toggle test
<https://www.youtube.com/watch?v=0EoBw_3xNZs> in each pair of pins, and as
both triggered the relay1, I did the same test but with double toggle
<https://www.youtube.com/watch?v=GskJBrgpWjY> events. Please check the
videos and the logs, hope it would be useful.

BUTTON 1 - Single toggle
-----------------------------------------------------------------------------
Jun  3 16:52:24 sonoff3-host ESP-APP: Button code 0402
Jun  3 16:52:24 sonoff3-host ESP-APP: Multipress 1
Jun  3 16:52:25 sonoff3-host ESP-MQTT: stat/sonoff3/RESULT = {"POWER1":"ON"}
Jun  3 16:52:25 sonoff3-host ESP-MQTT: stat/sonoff3/POWER1 = ON

Jun  3 16:52:27 sonoff3-host ESP-APP: Button code 0403
Jun  3 16:52:27 sonoff3-host ESP-APP: Multipress 1
Jun  3 16:52:28 sonoff3-host ESP-MQTT: stat/sonoff3/RESULT =
{"POWER1":"OFF"}
Jun  3 16:52:28 sonoff3-host ESP-MQTT: stat/sonoff3/POWER1 = OFF
-----------------------------------------------------------------------------

BUTTON 0 - Single toggle
-----------------------------------------------------------------------------
Jun  3 16:52:44 sonoff3-host ESP-APP: Button code 0401
Jun  3 16:52:44 sonoff3-host ESP-APP: Multipress 1
Jun  3 16:52:44 sonoff3-host ESP-MQTT: stat/sonoff3/RESULT = {"POWER1":"ON"}
Jun  3 16:52:44 sonoff3-host ESP-MQTT: stat/sonoff3/POWER1 = ON

Jun  3 16:52:47 sonoff3-host ESP-APP: Button code 0400
Jun  3 16:52:47 sonoff3-host ESP-APP: Multipress 1
Jun  3 16:52:47 sonoff3-host ESP-MQTT: stat/sonoff3/RESULT =
{"POWER1":"OFF"}
Jun  3 16:52:47 sonoff3-host ESP-MQTT: stat/sonoff3/POWER1 = OFF
-----------------------------------------------------------------------------

BUTTON 1 - Double toggle
-----------------------------------------------------------------------------
Jun  3 16:56:19 sonoff3-host ESP-APP: Button code 0402
Jun  3 16:56:19 sonoff3-host ESP-APP: Multipress 1
Jun  3 16:56:19 sonoff3-host ESP-APP: Button code 0400
Jun  3 16:56:19 sonoff3-host ESP-APP: Multipress 2
Jun  3 16:56:19 sonoff3-host ESP-MQTT: stat/sonoff3/RESULT = {"POWER2":"ON"}
Jun  3 16:56:19 sonoff3-host ESP-MQTT: stat/sonoff3/POWER2 = ON

Jun  3 16:56:21 sonoff3-host ESP-APP: Button code 0400
Jun  3 16:56:21 sonoff3-host ESP-APP: Multipress 1
Jun  3 16:56:21 sonoff3-host ESP-APP: Button code 0402
Jun  3 16:56:21 sonoff3-host ESP-APP: Multipress 2
Jun  3 16:56:22 sonoff3-host ESP-MQTT: stat/sonoff3/RESULT =
{"POWER2":"OFF"}
Jun  3 16:56:22 sonoff3-host ESP-MQTT: stat/sonoff3/POWER2 = OFF
-----------------------------------------------------------------------------

BUTTON 0 - Double toggle
-----------------------------------------------------------------------------
Jun  3 16:56:39 sonoff3-host ESP-APP: Button code 0401
Jun  3 16:56:39 sonoff3-host ESP-APP: Multipress 1
Jun  3 16:56:39 sonoff3-host ESP-APP: Button code 0400
Jun  3 16:56:39 sonoff3-host ESP-APP: Multipress 2
Jun  3 16:56:39 sonoff3-host ESP-MQTT: stat/sonoff3/RESULT = {"POWER2":"ON"}
Jun  3 16:56:39 sonoff3-host ESP-MQTT: stat/sonoff3/POWER2 = ON

Jun  3 16:56:41 sonoff3-host ESP-APP: Button code 0403
Jun  3 16:56:41 sonoff3-host ESP-APP: Multipress 1
Jun  3 16:56:41 sonoff3-host ESP-APP: Button code 0402
Jun  3 16:56:41 sonoff3-host ESP-APP: Multipress 2
Jun  3 16:56:41 sonoff3-host ESP-MQTT: stat/sonoff3/RESULT =
{"POWER2":"OFF"}
Jun  3 16:56:41 sonoff3-host ESP-MQTT: stat/sonoff3/POWER2 = OFF
-----------------------------------------------------------------------------

---
Lic. Darío Silva Morán

dariosm, 2017-06-12:
Hi guys, did you have the time to check the logs and videos I sent?

---
Lic. Darío Silva Morán

lino-silva, 2017-06-12:
I've done a fair bit of testing with the Sonoff-Dual as welll, and had the exact same results as @dariosm - the firmware just isn't ready for having two external switches controlling each relay independently. Will try to take a look into the code, but I think I'm way over my head on this one.

mydakota, 2017-07-19:
So if the Tasmota code won't work for duel switchs is it possible using alternative FW to get two relays and two switches to work . If so could you recommend the best FW to use like espeasy or something . It's a shame because my single sonoffs work perfectly with Tasmota ;( . Any home any ideas for getting this going please help thanks

Game01Master, 2017-11-03:
Sonoff Dual with original firmware and buttons. Buttons are conected to button header( gnd, button0, button1,gnd). Push switch must be open type.

https://youtu.be/VX_eV_HKb3s

Joeyhza, 2017-11-08:
I've had OTA issues flashing my Duals, in retrospect I'm glad i did for now

I have 2 of my duals installed behind wall light switches each controlling 2 lights on Ewelink as well as each physical wall switch connected to But 0 and But 1 on the Sonoff. Using the original firmware im able to turn on and off each light from ewelink and the physical switch.

Only problem with the wall switch is i have to flip it up and then down to turn the light on and off. So 2 flips to turn on and then 2 flips to turn off.

I have a few other sonoff basics running lights for other parts of the house where a dual switch wasn't needed. These are all Tasmota flashed with the physical wall switch for each running to GPIO14. Works flawlessly.

I'm still not entirely sure why the custom firmware can't be made to work the same as the orginal firmware for the dual when it comes to attaching physical switches to But 1 and But 2.
So.....for now I'm running some of my house on ewelink and someof my house on Domoticz using MQQT. If custom firmware can make the dual behave like the orginal when it comes to wiring up a switch to But 1 and But 2 - I'll be the first to flash Tasmota :)

Alexa integration working flawlessly regardless of which switch controls what

Probably better - not having all my eggs in one basket and all :)

Thanks for all the great dev work people!!! Love Tasmota

enesbcs, 2017-12-02:
Already exists custom firmware, that know button0 and button1 :)
[#1176](https://github.com/arendst/Sonoff-Tasmota/issues/1176)

ramgopal5, 2018-01-27:
@Game01Master can you share how you connected and the buttons u purchased.
I want to do exactly that without playing with firmware.

marcinbauer85, 2018-01-31:
@Game01Master Yes please share! I want to do exactly the same!

closed 2018-06-15 by stale[bot]