arendst/Tasmota #495: Itead based 3 gang wall switch [ESP8285][Sonoff Touch alternative]

ferbulous, 2017-06-03:
Hello, so I got this 3 gang switch that's using PSF-B85 (ESP8285) from [here](https://item.taobao.com/item.htm?spm=a312a.7700846.0.0.BIIMLl&id=546758441258&_u=32avq0dofdfc)

https://www.itead.cc/psf-b85.html
https://www.itead.cc/wiki/PSF-B85

I was hoping if anyone  could point out which is the correct serial pins for tx,rx,gnd & 3v3

[image]
[image]
[image]
[image]
[image]
[image]

Tasmota firmware is still compatible for this, yes?
Should I follow the exact same steps for sonoff touch flashing guide?

https://github.com/arendst/Sonoff-Tasmota/wiki/Sonoff-Touch

davidelang, 2017-06-03:
well, the wiki link you point at https://www.itead.cc/wiki/PSF-B85 includes a diagram that shows what pins are what on the idead module, see the pins definition section.

There's no way of knowing that each of these pins are wired to without tracing things on the circuit board, you will have to do that work. Once you figure that out, you can flash it and define it as a wemos module and define each of the GPIOs appropriately.

But it's going to take some digging on your part to figure out which GPIO is which. Please add the item to the wiki once you track this down.

ferbulous, 2017-06-03:
Thanks david
Is there a guideline for tracing on the circuit board? Or any examples from any sonoff boards?
I've never done this before

davidelang, 2017-06-03:
not really, you may be able to see the traces with a bright light, you may be 
able to track things down with a multimeter (but with many components on the 
board, this is non-trivial as you will see connectivity through other 
components)

some people will destroy a unit by removing all the components to make this 
easier.

Sometimes, if you buy an extra unit and have it shipped to an interested hacker, 
they will be willing to do the tracing.
k

ferbulous, 2017-06-03:
I see, that might be my other option if I couldn't figure it by myself
Do you know of any interested hacker maybe?
Btw, any other tools that I might need other than multimeter

I've only flashed the firmware on sonoff products that are listed on itead. 
Recently I've noticed there's 3rd party manufacturers that are releasing up to 4 gang wall switch based on itead chipsets. That got me interested since the sonoff touch is only available for 1 gang.

davidelang, 2017-06-03:
see if you can get any support from the manufacturers of the switches. Will they 
tell you what is hooked to what GPIO.

you could also experiment with the various GPIOs, but there is a (small) risk of 
burning something out by doing so.

If you can run the device with it's stock firmware, you may be able to figure 
some of it out by watching what voltages there are on the different pins when 
you turn things on and off

ferbulous, 2017-06-03:
Not to sure about getting support from the manufacturer

It's running on the stock itead firmware that's controlled by ewelink app.
For the voltages, what should I expect from each pin?
GND - 
Rx -
Tx -
3v3 -
GPIO -

davidelang, 2017-06-03:
well, you don't have to puzzle out the tx/rx pins, the ground pin will be 
ground, the 3.3v pin will be 3.3v

the gpio pins will be either 3.3v or ground, depending on what level they are 
at, what you will need to do is turn on one relay and see if one of the gpio 
pins goes from ground to 3.3v (or vice-versa), and then turn the relay off and 
see if it flips. repeat for all relays and you can then have a good idea what's 
what.

This gets more complicated when you have LEDs that turn on at the same time as 
the relays.

you also will need to try and figure out what gpio lines go high when an input 
is touched.
kk

ferbulous, 2017-06-05:
Hi David, if I need to test for rx,tx,gnd pins using multimeter with the switch turned on
What's the best way to power the switch without having to hook it up to the wall?

davidelang, 2017-06-05:
you can read the pinout on the wiki to find the 3.3v and ground pins on the 
module, and then apply power across them.

ferbulous, 2017-06-05:
Ok, I compared the pinout and and tried to pinpoint the location on the board
Still waiting for my multimeter to be delivered
Apparently the other 2 pin holes are not utilized? (doesn't lead to anywhere)
What do you think?

[image]
[image]

davidelang, 2017-06-05:
well, it looks like the one futhest to the right in your bottom picture is ground. I have no way of telling what the other pin is. It doesn't look like it's connected to anything on the top or bottom of the board, I can't tell if it's only a 2-layer board of if there is a third layer in the middle (shining a bright light through the board may show something)

you should also look to see what you can trace from the connector and the relays

ferbulous, 2017-06-07:
I'll get right on that 
For flashing the firmware, do I need to modify anything else on the sketch?
Since sonoff touch is one gang and this is 3 gang switch.
Will toggles for all 3 switch show up automatically?

davidelang, 2017-06-07:
you will have to set the module type to wemos and then configure the gpios for 
each swtich and for each relay (which is why it's important to track those 
connections down)

don't think of this as a variation of a touch, think of it as a completely new 
type of module.

ferbulous, 2017-06-07:
I see, this just got more complicated than I expected
Is there a wiki that I can read about starting with a new type of module and configure gpio for switch & relay?

khcnz, 2017-06-07:
Are you sure the ESP is connected to the relay directly? With some of the other multi-relay devices they communicate with another chip with serial commands see some info here https://github.com/arendst/Sonoff-Tasmota/wiki/Sonoff-Dual

davidelang, 2017-06-07:
yes, see "wiring additional things to a module"

it's pretty simple once you know how things are hooked up, but you first have to 
trace the wires to see how they are hooked up.

You've done well so far, you just need to keep going

filzek, 2017-06-08:
@ferbulous can you tell me what are the IC make/model at the right side of the PSF-B85? Also if you take a closer look at the upside of the PCB the rails come from the Relay activate coil, so, this PIN has to be connect to a GPIO level. Also the Female Header that connects to the touch panel has to have a GPIO pin too, so, it is a must have 3 GPIO to Touch Panel and 3 GPIO to the Relay. Its possible to do not use a HSPI interface.... so, if the IC is not a HSPI/SPI i have no clue what is it for.

ferbulous, 2017-06-08:
@filzek Here's a few more pics up close

[image]
[image]
[image]
[image]

You lost me there at HSPI, I don't know much about electronics yet, still learning here

bitdivision, 2017-06-09:
If you're in the UK I'd be happy to trace this out for you

ferbulous, 2017-06-11:
@bitdivision unfortunately no, and it's probably gonna cost me twice the price of the switch to ship it to your location. I'm in malaysia btw

ferbulous, 2017-06-11:
@davidelang when you mentioned wemos, does it have to be setup similar to this guide for sonoff 4ch?

http://www.themakersworkbench.com/tutorial/how-use-amazon-alexa-control-linknode-r4-esp8266-4-channel-relay-board

bitdivision, 2017-06-11:
@ferbulous Fair enough. I'll probably buy one at some point, but it won't be for a while.

If you can work out which gpios are connected to the relays then that will be most of the work.

One thing we need to work out is what that 16 pin chip beneath the ESP is, I can't read it in your photo.

I'm not hugely familiar with the Sonoff line but I believe that a lot of them use a Silicon Labs F330 microcontroller to expand the GPIO pins. It looks like that might be what's going on here but with a different chip (I think). 

I would suggest finding the control pins of all the relays, putting your multimeter in continuity mode, put one probe on the relay pin and touch the other probe to the pins on the esp and the chip beneath it until you find the correct pins. Obviously, don't do this while the unit is connected to power.

ferbulous, 2017-06-11:
@bitdivision also, there's another 4 gang switch that just arrived which I got from [here](https://item.taobao.com/item.htm?spm=a1z09.2.0.0.hUb7tn&id=540194306845&_u=b2avq0do0df1)

[image]

You can see more pics here https://goo.gl/photos/xzGnCyVKFUs284H89
I will add as another issue just for this switch since the pcb is a bit different once I'm done with this 3 gang switch. Not to sure if it's easier or more difficult to spot the pins for this one

From what I can see, the name for the 16 pin chip is:
16DN8SW
ULN2003A g4

Ok, so technically I don't have to hook up anything or even power the switch when using the multimeter. How would it show up on the multimeter display if I got the correct pins?

bitdivision, 2017-06-11:
If you look up ULN2003A you'll find: http://www.ti.com/lit/ds/symlink/ulq2003a.pdf

So that chip is an array of transistors. They're likely using it to drive the relay pins. The collectors (1C, 2C etc) of the transistors should be connected to the relay pins. If so, you'll need to find which GPIO on the esp the corresponding Base pin (1B, 2B etc.) is connected to.

Your multimeter should have a continuity mode, in that mode, when the probes are touched together you will hear a beep. So if you touch two pins that are connected, then the multimeter should beep. If you don't have a continuity mode then you'll have to use resistance mode. In that case, if two points are connected you'd expect to see a resistance of < 1 Ohm.

davidelang, 2017-06-11:
the wemos module setting lets you specify the use of every GPIO in the software 
setttings, nothing is pre-set. So when you have an unknown esp8266 based device, 
that module setting will let you use it

On Sun, 11 Jun 2017, ferbulous wrote:

> @davidelang when you mentioned wemos, does it have to be setup similar to this guide for sonoff 4ch?
>
> http://www.themakersworkbench.com/tutorial/how-use-amazon-alexa-control-linknode-r4-esp8266-4-channel-relay-board
>
>
>

ferbulous, 2017-06-12:
@bitdivision @davidelang 
So I finally got my hands on a multimeter
If I'm reading the ULN2003A diagram right, this should be the pins for the 3 relay pins
What should be the next step?

[image]
[image]

bitdivision, 2017-06-12:
I'm slightly confused as to why the outputs of that chip appear to be connected to the ESP, not the relays... Are the pins on that unpopulated header definitely connected to the relays? Anyone else have any ideas? @filzek maybe?

In any case, next thing to do is probably to flash it with Tasmota (for 8285) set to Wemos mode and try to set those GPIOs up as relays. If those GPIOs are incorrect though, it's possible you may break some things on the board.

bitdivision, 2017-06-12:
In fact @ferbulous I would hold off flashing anything until we work out what's going on there.

davidelang, 2017-06-12:
@bitdivision it could be that that chip is doing something with the touch signals.

@ferbulous did you ever trace the pins from the connector? we will need to do that to have any chance of responding to the touch sensor

also, now that you have a multimeter, try tracing the 4th pin of each relay and see if it connect to the module. in your lat picture it would be the top right pin of the left two relays and bottom right pin of the last relay.

ferbulous, 2017-06-13:
@bitdivision for the chip connection, do you think it's similar to this board (tantech)?
This guy flashed espurna firmware on it

https://www.letscontrolit.com/forum/viewtopic.php?f=5&t=2721

How should I modify the sketch to include the extra pins/relay once set to wemos mode? Any examples for it?

I'll hold off the flashing for now

@davidelang I haven't  trace the pins from the connector yet. 
How should I do that? Do I need attach jumper cables from touch board connector to the PSF-B25 board?

Do you mean for these relay pins?

[image]

bitdivision, 2017-06-13:
@davidelang Ah, that makes much more sense.

@ferbulous It does look similar to that other board, but it's not identical.

Yes, those are the relay pins that you need to trace.

ferbulous, 2017-06-14:
@davidelang @bitdivision 
Just tested the pins, but I'm not sure which ESP pins lead to which relay pin since they're all giving me the same value

[image]
[image]

And I'm still not sure what's the best way to test the connector, could need some instructions here

davidelang, 2017-06-14:
the pins you circled in the last post are not the ones to test, they are ground (as you can tell by tracxing the circuit from the two left ones). the ones you pointed arrows at a couple posts back are the ones you need to trace. they could go to any pin, not just the 0, 9,10 pins (in fact, since those were previously traced to the other chip, it's probably not them) These could also go to the other chip instead of the esp chip.

davidelang, 2017-06-14:
you would test the connector the same way you test everything else, put one 
probe of your meter on a pin of the connector, and try every pin of the esp 
module to see if there is a connection.

ferbulous, 2017-06-14:
@davidelang Oh, so I'm not really limited to those 3 pins, I thought it was odd how I'm not getting any reading from the pins you mentioned before. Will test it again today

davidelang, 2017-06-14:
those pins on the relays are connected to something, you need to track down what 
they connect to.

anhxco, 2017-06-14:
I think that similar sonoff 4 chanels

ferbulous, 2017-06-15:
@davidelang what values should I expect? Tried tracking the esp pins & 4th pin from each relay pins but I'm getting varied values (no 1 ohm)

ferbulous, 2017-06-15:
@anhxco what do you think about this [one](https://goo.gl/photos/xzGnCyVKFUs284H89)?

davidelang, 2017-06-15:
when you find the right place that the pin is connected, you will see near zero resistance.

check all the chips, they could be connected to the ULN2003A or they could be connected to the three transistors (the small chips with three connections). The only way to know is to either trace the circuit board or probe with the meter.

anhxco, 2017-06-16:
@ferbulous i had this one, i will try to find out the pins, but i saw that
product work unstable. below is sonoff 4ch schematic, it's using esp8285:
https://www.itead.cc/wiki/images/d/d3/Sonoff_4CH.SCHMATIC.pdf
I think and hope that similar.
BR

davidelang, 2017-06-16:
no, there is no reason to believe that the schematic is going to be different. 
The only thing that you can count on is the pinout of the module, once you move 
off of that, all bets are off, any functionality can be connected to any pin.

things produced by a single company tend to use similar design patterns for 
similar products. But when you start talking about products from different 
companies, you can't make any assumptions about what pins are used for what 
purposes.

ferbulous, 2017-06-16:
@davidelang one more thing, should I use the positve or negative probe pins on the relay pins
Seems like I can different readings when I use either one, so which should be the correct probe for testing?

ferbulous, 2017-06-16:
@anhxco so you have the same 4 gang switch too?
What do you mean when you mentioned it can be quite unstable? Does it like not turn/on when you're using the app?

davidelang, 2017-06-16:
when you find the correct pins it won't matter, both directions will give you 
near zero readings.

ferbulous, 2017-06-16:
@davidelang ok, it's just that I notice LED on the bottom lights up when I was using the negative probe pin on relay pin & getting 0 ohm on the pin 9 of ULN2003. But not with positive probe pin
I'll test again with both negative/positive probes till I get zero readings.

davidelang, 2017-06-16:
There are all sorts of reasons why you can get different results in different 
directions, diods (including LED) being a common one. the voltage from you meter 
can power parts of the circuit. But you are looking for cases where there is 
nothing between your probes but wire, and that will show near zero resistance 
and not power anything up.

ferbulous, 2017-06-16:
@davidelang These seem to the only points where I got zero resistance

[image]
[image]
[image]

Didn't get any readings like that when I tested the connectors & the esp pins

bitdivision, 2017-06-16:
@ferbulous Well done, you're most of the way there now. What that shows is that each of the relays is connected to a transistor (the three pin device you showed in your third picture). The top right pin of each of those transistors is connected to the ESP through a resistor. I can see that the first two are connected to GPIO2 and GPIO12 respectively. You'll need to work out what the rightmost transistor is connected to since I can't see that from the picture.

bitdivision, 2017-06-16:
The bigger 2 pin black devices you pointed out in your first two pictures are probably diodes, they will connect to ground and can be ignored.

bitdivision, 2017-06-16:
So just to be clear, you need to figure out what the via circled in orange is connected to @ferbulous 

[image]

bitdivision, 2017-06-16:
I'm going to take a wild guess and say GPIO5. Check that one first.

ferbulous, 2017-06-17:
@bitdivision 
 I'm posting this ESP if it was viewed on the front of the board, should make it less confusing on which direction it is

[image]

If I'm tracing by looking at the back of the board (and if I'm doing this right) 
The orange one goes to pin GPIO12

[image]
[image]
[image]

davidelang, 2017-06-17:
Ok, you are continuing to make progress, you have mapped out how the relays are connected. Now you just need to track down the inputs from the second board. There have to be pins on the connector that somehow get to the ESP module, they may go through other components first, so you will need to track them down.

anhxco, 2017-06-17:
I have 3 gang switch, and below are the pins connection for switch(SW) and
relay (RL), it's same sonoff 4ch:
[image: Inline image 1]

ferbulous, 2017-06-17:
@anhxco could you upload the pictures again? can't seem to view it on your post
also, have you tried flashing tasmota firmware on it?

anhxco, 2017-06-17:
reup image

anhxco, 2017-06-17:
[image: Inline image 1]

bitdivision, 2017-06-17:
@ferbulous Ah, sorry, I didn't check the orientation of the chip. You need to check the second relay (you marked GPIO11) connection, I don't think that via is actually connected to the chip there (see my edit above), looks like it goes to the other side and then back to GPIO4

In terms of the switch inputs, I can't find any information on the capacitive touch controller they're using (XW05A), but I would take a guess and say that the three resistors above the chip are the switched lines. I think those resistors are pullups.

If you can work out whether those connect to the ESP through the connector, then I think you're good to go

https://i.imgur.com/uLOsgMF.jpg

davidelang, 2017-06-17:
@bitdivision my thinking is that you don't need to try to decode anything on the touch board. You know that it interfaces with the main board through the connector. So if you figure out where the pins of the connector are connected on the ESP board, you don't need to worry about how they are setup on the other board.

There will be power, ground, three pins to send the touch signal to the ESP, and possibly three (or more) LED signals going back out (which may be connected to the ESP or may be connected directly to the relay outputs. That can account for up to 11 of the 14 pins, so it's a bit incomplete.

But there really isn't a need to decipher the touch board any more unless we find that there is something really odd going on (a serial interface for example)

but simply tracing the pins of the connector will tell us what GPIOs to play with.

[36 more replies]

closed 2018-05-07 by stale[bot]