arendst/Tasmota #496: Differences between compiled binaries are not clear for me

terba, 2017-06-03:
Hello,

I wanted to flash a Touch with Tasmota. But was not able to understand (after reading the wiki), what are, or are there any differences between the compiled binaries under .pioenvs directory. And how to set the the one I want to upload to the Touch?

The same question stands for OTA. Do I need to put two binaries on my webserver, one for basic Sonoffs and an another for Touches (and set different Ota url on them)?

I read somewhere (maybe on the wiki), that uploading the first firmware to the touch has to be made carefully, because subsequent uploads would break if it is done wrong.

So I commented out other builds from platformio.ini so it uploaded the correct (only) one. This is the right method?

I think this should be made clear in the wiki. Thanks!

arendst, 2017-06-03:
If you compile the source using Arduino IDE it should be clear what settings to change in the IDE to select either Sonoff-Touch or Sonoff-4CH and other Sonoffs as you noticed.

Using platformio is more challenging as the default platformio.ini now spits out four compiled versions meaning:
- **sonoff.bin** = the default firmware for all but initial Sonoff-Touch and Sonoff-4CH devices
- **sonoff-touch-4ch.bin** = the inital firmware for the Sonoff-Touch and Sonoff-4CH ONLY and is only used at initial serial upload to these devices. Future (OTA and web) uploads can use the **sonoff.bin** version as this version will be patched during (OTA and web) uploading to satisfy the esp8285 chip and flash layout.
- **sonoff-minimal.bin** = is interim firmware to be used when the above firmware images become too big to fit as OTA or web upload; installing this one first and THEN uploading the desired **sonoff.bin** allows for future firmware size growth over the OTA file limit of 1/2 flash size.
- **sonoff-ds18x20.bin** = is a version of **sonoff.bin** with the USE_DS18X20 define enabled and a larger MQTT buffer size to be used by people having more than 4 ds18x20 sensors connected.

closed 2017-06-05 by arendst

terba, 2017-06-05:
Thanks! I added this to the Upload wiki page.