arendst/Tasmota #498: H801 still supported? Flash OK, FW start NOK

amiraoul, 2017-06-03:
Hi,

I have received several H801 and tried to flash v4.20 and 5.14. Flashing works okay but the FW is not starting. Serial Log after restart:

␀;l␀d��|␀�$�|␃␄␌␄�␌$�␄c|ǃ␂�␛�{�c�␌b��og�l'o���␌#␜x␌dc␏l{$p�'�␘␃␄␌�␄l␄��␄␌␄c␄o�<␃l�␄␌�c��gn�␀d��d`␃�␛␛'o␌d`␂␎␃gs�ۓo␌␄#␄�␎$␏;��g␄␌c␄�␇l�␃$␌���x�␌�␇d��g�␃
 ets Jan  8 2013,rst cause:2, boot mode:(3,0)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v09826c6d
~ld

 ets Jan  8 2013,rst cause:2, boot mode:(3,0)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v09826c6d
~ld

 ets Jan  8 2013,rst cause:2, boot mode:(3,0)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v09826c6d
~ld

 ets Jan  8 2013,rst cause:2, boot mode:(3,0)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v09826c6d
~ld

 ets Jan  8 2013,rst cause:2, boot mode:(3,0)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v09826c6d
~ld

.... and so on

I tried flashing espurna and also ESPEasy and these 2 worked fine on the same board.
I also tried flashing with Arduino IDE, PlatformIO and also direct file upload with esptool.
All give the same serial log and do not work.

Since the device is not listed in supported device list, i'm asking myself if the support has been removed?

Anybody else was recenty flashing a H801? If yes, did it work?

BR,

amiraoul

Staars, 2017-06-03:
Hmm, I'm on 5.0.0 and this works.
Can you try to revert back to 5.0.0 (maybe you need to step back to 4.x before that). I only have one H801 here and I need it in my living room. So I don't want to test 5.1.4 for now.

amiraoul, 2017-06-03:
Well as mentionned i tried 4.20 and the latest 5.14! I will give 5.0.0 a try but i already doubt that it will work since 4.2.0 is next to 5.0.0!
Anyone knows what the log means?

Staars, 2017-06-03:
Yes, and i can safely say that 5.0.0 works. 
I have a second h801 on monday and then i will play around with it. But maybe in the meantime you can figure out, if a certain versions breaks the device.
As i wrote in another post it is possible to brick your device with firmware 5.x where you must reflash with 4.x, then make a device reset, then re-flash 5.x.

amiraoul, 2017-06-04:
Thanks for your answer. I found the post you mentioned about the bricking and also tried the process without any luck --> already blocked at first step to access UI because 4.1.3 not booting ang still showing the error above.
Tried also erasing the complete flash memory (esptool.py -p /dev/USBPORT erase_flash) without any change when flashing Tasmota 4.1.3 back to the H801 over pio/atom.
Reflashed ESPEasy which is working again instantly :-/. 

@Staars really curious if you will run in any trouble flashing your H801

Staars, 2017-06-05:
Okay, it still works with my H801.
I updated via web-interface from 4.2 to 5.0, then from 5.0 to 5.1.4. The web-updater did not want to take 5.1.4 over 4.2, that's why the two-step-update.
Everything seems to work correctly.

amiraoul, 2017-06-05:
Crap... ;-)
Thanks for letting me know. I ordered 5 new H801 from AliExp and let's hope these will work fine for me. 
So i will also first update to 4.2 before going to 5.xx

BR, Raoul

closed 2017-06-05 by amiraoul

mirko, 2017-07-22:
I'm having the exact same issue with my H801 and wonder whether it's, the device oder something/someone else. Did you figure out what it was? Indeed HW issues?

amiraoul, 2017-07-22:
Never tried flashing Tasmota to an H801 again, using them now with ESPEasy... Sorry

mirko, 2017-07-22:
But did your weirdly behaving ones worked with ESPTool eventually?

amiraoul, 2017-07-22:
ESPTool?

I tried flashing them with espeasy and also espurna and both worked fine, only tasmot didn‘t work!

mirko, 2017-07-23:
Sorry, I meant espeasy, not esptool. Thanks for the feedback!