arendst/Tasmota #500: Sonoff touch diconnect

phdemartin, 2017-06-04:
I've flashed successfully my sonoff touch 3 days ago, every thing working great for 3 day, and suddenly, it disappeared.
Trying to scanning the port, i discovered that the port 80 is closed but a 1716 port is open ??
Since it's a "touch" (I can not disconnect it without unscrewing it)I tried to cut the light circuit in my house to reboot it, but it don't come back online.

Any idea ?

phdemartin, 2017-06-05:
So, diving into the command, I made a
mosquitto_pub -u xxxxxx -P xxxxxx -t cmnd/sonoff/main_light/OtaUrl -m http://sonoff.maddox.co.uk/tasmota/sonoff.ino.bin
and
 mosquitto_pub -uxxxxxx -P xxxxxx -t cmnd/sonoff/main_light/Upgrade -m 1
It restarted and now it's working as intended...
Still don't know what is happening

phdemartin, 2017-06-06:
It still dont work properly.
After a few minute it die.
The wifi still connected, I can ping the sonoff touch
but the mqtt can't find the mosquitto broker er any more,  and after a few more minutes the web server die also.
I've tried to deactivate the mqtt, and run my automation just with web querry, but it don't last more time up, I also tried to deactivate the web server and run just the mqtt protocol , same thing :(
I've got another sonoff (basic) on the same broker, with the same firmware and it run flawlessly.
Please help me...
Serial Log:

00:00:00 APP: Project sonoff Sonoff (Topic sonoff, Fallback main_light, GroupTopic sonoffs) Version 5.1.3
00:00:00 Config: Saved configuration (1492 bytes) to flash at FB and count 24
00:00:00 Wifi: Checking connection...
00:00:00 Wifi: Attempting connection...
00:00:00 Wifi: Patch issue 2186
00:00:00 Wifi: Connecting to AP1 Demartin_Bureau in mode 11N as main_light...
00:00:01 Wifi: Checking connection...
00:00:01 Wifi: Attempting connection...
00:00:02 Wifi: Checking connection...
00:00:02 Wifi: Attempting connection...
00:00:03 Wifi: Checking connection...
00:00:03 Wifi: Attempting connection...
00:00:04 Wifi: Checking connection...
00:00:04 Wifi: Connected
00:00:04 mDNS: Initialized
00:00:04 HTTP: Webserver active on main_light.local with IP address 192.168.0.28
00:00:04 RTC: (UTC) Tue Jun 06 15:38:26 2017
00:00:04 RTC: (DST) Sun Mar 26 02:00:00 2017
00:00:04 RTC: (STD) Sun Oct 29 03:00:00 2017
16:38:27 MQTT: Attempting connection...
16:38:28 mDNS: Query done with 0 mqtt services found
16:38:28 MQTT: Connected
16:38:28 MQTT: tele/sonoff/main_light/LWT = Online (retained)
16:38:28 MQTT: cmnd/sonoff/main_light/POWER = 
16:38:28 RSLT: Receive topic cmnd/sonoff/main_light/POWER, data size 2, data on
16:38:28 RSLT: DataCb Group 0, Index 1, Type POWER, Data on (ON)
16:38:28 MQTT: stat/sonoff/main_light/RESULT = {"POWER":"ON"}
16:38:28 MQTT: stat/sonoff/main_light/POWER = ON
16:38:28 MQTT: tele/sonoff/main_light/INFO1 = {"Module":"Sonoff Basic", "Version":"5.1.3", "FallbackTopic":"main_light", "GroupTopic":"sonoffs"}
16:38:28 MQTT: tele/sonoff/main_light/INFO2 = {"WebserverMode":"Admin", "Hostname":"main_light", "IPaddress":"192.168.0.28"}
16:38:28 MQTT: tele/sonoff/main_light/INFO3 = {"Started":"Power on"}
16:38:29 Config: Saved configuration (1492 bytes) to flash at FB and count 25
16:38:30 MQTT: stat/sonoff/main_light/RESULT = {"POWER":"ON"}
16:38:30 MQTT: stat/sonoff/main_light/POWER = ON
16:38:37 MQTT: tele/sonoff/main_light/STATE = {"Time":"2017-06-06T16:38:37", "Uptime":0, "Vcc":3.054, "POWER":"ON", "Wifi":{"AP":1, "SSID":"Demartin_Bureau", "RSSI":100, "APMac":"14:CC:20:72:D0:DE"}}
16:38:47 Wifi: Checking connection...
16:38:47 Wifi: Connected
16:39:12 Wifi: Checking connection...
16:39:12 Wifi: Connected
16:39:14 MQTT: Attempting connection...
16:39:15 mDNS: Query done with 0 mqtt services found
16:39:22 MQTT: Connect FAILED to 192.168.0.26:1883, rc -2. Retry in 10 seconds
16:39:32 MQTT: Attempting connection...
16:39:33 mDNS: Query done with 0 mqtt services found
16:39:40 MQTT: Connect FAILED to 192.168.0.26:1883, rc -2. Retry in 10 seconds

davidelang, 2017-06-06:
check for logs on the mosquitto side. We had someone else with a similar issue 
that turned out to be the fact that multiple devices were connecting to 
mosquitto at the same time, when the second connected, mosquitto decided the 
device had changed it's IP and dropped the connection from the first device.
k

phdemartin, 2017-06-06:
Ok, I will check this out
Thank you

______________________________________________________________
Philippe Demartin

Mail:           phdemartin@gmail.com
Web:            www.demartinenchile.com
tel:            +56 32 277 20 53
Cel:            +56 9 9 278 28 68
Skype:          phdemart

phdemartin, 2017-06-07:
Ok, I found the problem looking at the mosquitto verbose messages ...
It was really silly, I had an ip address conflict from my tablet and the sonoff touch. :=(
The sonoff have a fixed IP, but the tablet was requesting an IP from the DHCP server, and to add to the problem, I have a dashboard installed on this device who also submit mqtt data about battery and sensors.
For a mysterious reason, the router gave the same IP to the tablet ???
So, I simply fixed the problem assigning an different IP to the tablet in the router network config.
Excuse me for the buzz, I was despaired

davidelang, 2017-06-07:
you need to either set your static IPs outside of the range of your DHCP server, 
or set them as fixed allocations in your DHCP server (they can still be outside 
of the dynamic range)

dup IPs happen to everyone at some point.

closed 2018-05-08 by stale[bot]