arendst/Tasmota #502: Pow LED Indication

mr-const, 2017-06-05:
I've got a POW device with 5.0.7 firmware, attached to my boiler and it seems a little bit confusing when both LEDs (red and blue) turn on/off simultaneously with relay, so, when relay is off - there is no visual information whether device is up and running, or no.
Are there any settings to adjust this behavior? 
For example blue always on, when everything is okay and red shows state of the relay.

khcnz, 2017-06-06:
Please review the LED commands on the commands page on the wiki

mr-const, 2017-06-06:
Thank you for your response, I reviewed the section with led commands on the wiki, however there is mentioned only one led, which can be on/off/blinked, while POW device has two, blue and red, so one led could be tied to relay (red for example), while another (blue) always could be on and blink-on-mqtt, or off.

arendst, 2017-06-06:
The red led is not under software control. It is hardwired to the relay control port.

The blue led can be controlled with commands from the Wiki (LedState and LedPower).

closed 2018-05-08 by stale[bot]