arendst/Tasmota #504: MQTT retain flag default value

faljse, 2017-06-05:
My OpenHab setup had inconsistent switch states after changing configuration or restarting oh server.  
As far as i understand; sending mqtt messages with retain=1 is the only way to solve this; however default values in firmware and precompiled binaries set retain to 0.

As i expect other users to suffer from this issue too - wouldn't it be better to set the default to 1 (or document the issue somewhere in the mqtt/openhab instructions)?

PS: Awesome work. Never seen a commerial device coming even close.

davidelang, 2017-06-05:
go through the various issues raised here and you will find a lot of people who 
have problems because retained was set to 1

retained being set to 1 tells the mqtt server (not the home automation server) 
to override the state of the device when it connects. This can be done in the 
home automation software if you want, and the tasmota firmware has various 
options on retaining the mode across reboots.

Having the transport layer get involved with this as well just causes more 
confusion IMHO
k

stefanbode, 2017-06-06:
Agree to David. The Retain-Flag is now only for the TASMOTO. You can also define the retain flag in openhab configuration. Therefore ENABLE in openhab and DISABLE in TASMOTO is possible. I have RETAIN also on in openhab: Here some side-effects that you may take into consideration.
1. Pulsetimers do not really work in this configuration. You need to define the OFF in openhab to "" or "none" but for sure not "OFF" and switch off after switching on to get it working.
2. If you WLAN goes ON and OFF the retain message will be interpreted on the TASMOTO like a new command. Mostly fine but not with pulsetimers. Anyhow the light switches on even if disconnected at the time when the message was fired when the light reconnects to the MQTT server.
3. Doing RETAIN=OFF in openhab required the connection of the device stable. Otherwise it could happen, that the command does not reach the client. There is no side-effect with pulsetimers because these counts still backward also after a reboot or reconnect.

faljse, 2017-06-07:
Thanks for explanation @stefanbode 

My real concern was getting the system in a consistent state after a 
(reboot/disconnect/config change/server swap).

So the suggested solution is to enable retain in openhab (+persistence, +default values)
and disable retain in tasmota.

(Ignoring implementation issues; it still feels wrong introducing persistence in openhab which only purpose is to disagree with the device state :)

stefanbode, 2017-06-07:
Yes, this is how I have it running and it works reliable over different scenarios. Here a configuration without pulsetimers that ensure the state is correct in openhab; at least after the "teleperiod" (300sec). Please be aware that my MQTT topics are a bit different to the default, but should be obvious how it works.

`Switch ESP_832E5A_Ankleidezimmer "Ankleidezimmer Licht"  <switch> ( ESP_832E5A) {mqtt=">[mybroker:hm/setting/ESP_832E5A/power:command:ON:on],>[mybroker:hm/setting/ESP_832E5A/power:command:OFF:off],<[mybroker:hm/status/ESP_832E5A/POWER:state:default],<[mybroker:hm/status/ESP_832E5A/STATE:state:JS(getValuePOWER.js)]"} `

And here the small js transformation getValuePOWER.js:
`JSON.parse(input).POWER`

closed 2018-05-08 by stale[bot]