arendst/Tasmota #505: H801 commands

kruebsam, 2017-06-05:
H801 seems (according to the possibility to select it as type 20) already to be part of TASMOTA. However I could not find any directions/hints on its usage. What are the coresponding commands to deal with? The wiki has no idea on that one too.

Staars, 2017-06-05:
Hi, this is very easy.
Read here: https://github.com/arendst/Sonoff-Tasmota/issues/387

If you have any questions, feel free to ask.

davidelang, 2017-06-05:
sensors don't have any commands to deal with them, they just put data in the 
tele/ reports once they are configured.

kruebsam, 2017-06-05:
Thanks for the Info. Is it OK if I update the WIKI accordingly by adding a section for H801 ?

osteospurnum, 2017-09-16:
An H801 entry in the Wiki would be very welcome I am trying to control through Hass and while I can send single commands manually have no real control system .

kisseler, 2017-11-15:
I would like to use the H801 WiFi controller to control a white/warm white LED strip. So I flashed Tasmota to the device, made all settings correct, plugged the strip to the ports W1, W2 and VCC.
I managed to send commands like
`cmnd/H801-01/PWM3`
with values between 0 and 1023. The result is e.g.
`stat/H801-01/RESULT = {"PWM":{"PWM1":3,"PWM2":1023,"PWM3":1023}}`
But it does not have any effect on my white color at all.

Does anyone have an idea how I can control the light temperature mixing W1 and W2 an dimm my LEDs?

Thanks very much in advance!!!

Staars, 2017-11-15:
I think You should use PWM4 and PWM5.

kisseler, 2017-11-15:
Thanks! And I did not get it right away, that I had to chose PWM4 & 5 for GPIO 4 & 14.
[image]
So issue solved. 😊

kisseler, 2017-11-17:
One more issue: I assigned the two white channels now to "R" and "G" (PWM1&2) in order to user GPIO4 and 14 as switches.
But I cannot achieve that switch1 (PGIO4) and switch2 (GPIO14) actually send out an MQTT message. I try to open/close the contact between the Ports and GND with no effect. I also tried to add another pull-up resistor.
It just won't react. I tried differect Values fpr switchMode (1) and switchTopic (1 or 2). Does anyone successfully use the GPIO4/14 for different purposes at all?

Staars, 2017-11-17:
As far as I understand there is no GPIO-functionality on these ports on the H801.
What do you try to achieve?

kisseler, 2017-11-18:
Well, I simply would like to use the "free" GPIOs to connect a normal wall switch (push button) an send signals ON/OFF via MQTT in order to control the light with an OpenHAB rule.

As far as I understand, this should be possible. Why else should one be able to select the connected device for GPIO4 (W) and GPIO14 Sensor (WW)?

kisseler, 2017-11-27:
Here is where I can select any device for GPIO4/14 on the H801
Does anyone have an idea how I can get this running?
[image]

elmicha, 2017-12-06:
The H801 has the source (input) of MOSFETs at GPIO4/14. You cannot connect a switch to the terminal blocks because they are connected to the **outputs** of the MOSFETs.

plasticbrain777, 2018-03-28:
It is possible if you're keen... If you wish to use GPIO4 and GPIO14 as inputs for switches you need to connect directly to the ESP8266 pins.  The 5 channels RGBWW connect from the ESP8266 chip to the HC245 chip. The HC245 chip then drives the individual mosfets. You could solder fine wires pins 2 and 6 of the HC245, which connect directly to the ESP8266 pins 9 and 16 (GPIO14 and GPIO4 respectively).  

Cautions: 
- Set GPIO14 and GPIO4 as inputs (switches) in Tasmota H801 configuration.
- These pins are 3.3V max input. Use appropriate input protection (4.7k series resistor and possibly a pullup/down resistor).
- The WHITE1 and WHITE2 mosfets will also conduct according to your switch positions.
- I havent actually tried this, but it should work.
- You need to be keen!

**ESP8266 to HC245:**
**ESP8266 pin 9  GPIO14	HC245 pin 6 WHITE1**
ESP8266 pin 10 GPIO12	HC245 pin 5 BLUE
ESP8266 pin 11 VDDPST
ESP8266 pin 12 GPIO13	HC245 pin 4 GREEN
ESP8266 pin 13 GPIO15	HC245 pin 3 RED
ESP8266 pin 14 GPIO2	
ESP8266 pin 15 GPIO0
**ESP8266 pin 16 GPIO4	HC245 pin 2 WHITE2**

elmicha, 2018-03-29:
Wow, that's a great investigation and great solder work! I can barely hold the multimeter probes at these tiny pins.

As nobody mentioned it: we should be able to use the RX and TX pins as inputs, shouldn't we? We can configure it as a Generic module and assign RX and TX to different duties. I didn't try it, but it should work, I think.

renne, 2018-05-18:
@arendst 
I need 3 GPIOs (I2C SCL, SDA and interrupt) for my [RoomSensorAddon ](https://github.com/renne/dominaus/projects/11). Please consider moving [wouterbaake's pull-request](https://github.com/arendst/Sonoff-Tasmota/pull/1503/commits/788d76e0a7acb0a3fbd068c6282e97518e73fed9) from development to master branch.

plasticbrain777, 2018-05-21:
You can also use the H801 module as an MQTT controlled IR remote and RGB LED strip controller.  This is good for home cinema use for example. A spare white channel drives the IR LED - just an IR LED and resistor required. No modifications required. **The H801 mosfets work at 38kHz, so no extra drive circuitry required.  In this case "GPIO14 Sensor" is set to "08 IRsend"**.  ... sorry about posting this info here - not sure where to put it.  I just want people to be aware of this functionality before any useful tasmota firmware options are removed.
[image]

renne, 2018-05-21:
@plasticbrain777 
Great idea! I suggest the [H801 Wiki-page](https://github.com/arendst/Sonoff-Tasmota/wiki/H801).

ascillato, 2018-06-13:
Hi,

If your issue was solved, please close it. Thanks :+1:

closed 2018-06-13 by ascillato2