arendst/Tasmota #507: Changing SwitchTopic<n> via MQTT doesn't allow Slashes ('/')

mirko, 2017-06-06:
While the webinterface as well as the default string in `config_user.h` allow slashes being part of the MQTT topic, setting a `SwitchTopic` via MQTT undergoes a conversion - slashes are substituted by underscores (`/` -> `_`).

```
mosquitto_pub [..] -t cmnd/DEV/SwitchTopic1 -m "foo/bar/bla/blub"
```
results in:
```
cmnd/motion_livingroom/POWER1 OFF
cmnd/DEV/SwitchTopic1 foo/bar/bla/blub
stat/DEV/RESULT {"SwitchTopic":"foo_bar_bla_blub"}
cmnd/foo_bar_bla_blub/POWER1 ON
```
which is rather annoying, especially when trying to follow some conventions involving slashes in MQTT topics (e.g. https://github.com/marvinroger/homie/).

If that substitution is not intended behaviour I consider that being a bug.
If it's intentional, why and why is it inconsistent to the "normal" MQTT topic?

arendst, 2017-06-06:
If you use the webinterface and/or the default string in ``user_config.h`` do they function as you expect?

mirko, 2017-06-06:
> If you use the webinterface and/or the default string in user_config.h do they function as you expect?

It does.

arendst, 2017-06-07:
Well suppose I would also restrict the use of slashes in the webinterface (I won't check the user_config.h defines - results are up to you), can your wish be fullfilled by playing with the FullTopic string. This allows the use of slashes.

I don't see why you want to use a slash in the %topic% string as you could make up a FullTopic of say ```%prefix%/motion/livingroom/%topic%/``` where %topic% would be replaced by either your Topic, ButtonTopic or SwitchTopic.

mirko, 2017-06-07:
In my case the "relais" in the device controls -- in that case -- the living room, while the motion sensors attached to it (via GPIOs) cover other rooms. That's why I'd like to have a topic -- sticking to your convention -- like %prefix%/motion/bathroom/%topic%.
Nomenclature of course is debatable, but (in my opinion) shouldn't be limited by arbitrary* technical limitations.

*I don't mean to be rude here - I just don't get the point in doing so (yet)

davidelang, 2017-06-07:
well, in the MQTT spec, the address is different fields seprated by a /, so 
allowing a / in a field just does not make sense.

Tasmota recently gained the ability to define the entire address layout, so you 
have the ability to create whatever you want, it just takes changing a different 
variable thatn the topic field.

closed 2018-05-08 by stale[bot]