arendst/Tasmota #508: Toggle Switch on GPIO 1(RX) or 3(TX)

amitubale, 2017-06-07:
I have been using the GPIO 14 for toggle switches(GPIO 14 and GND).
I want to use the GPIO14 for Temp/Humidity(DHT22) and i tried using gpio1 and 3 and with gnd and i was not able to toggle. I see the option of Switch1 in the configuration.  

how do i connect gpio1 or 3 to the toggle switch ? please help.

thanks

davidelang, 2017-06-07:
you need to make sure nothing else is setup to use the gpio (disable serial 
logging) and you will need to wire a pull-up resistor to the pin.

1043717432, 2017-06-20:
I also need such a setting, how to set up? The
Can you directly send a picture reference?

amitubale, 2017-06-20:
From Davidelang's tip I think it should be this way. I had drawn (attached png) this so i can verify it with folks here.
I have been using GPIO 14 and GND for my wall toogle switches. One in this case i wanted to use GPIO 14 for the DHT22 temp sensor and use one of the TX/RX pins for the wall toggle switch. I had tried it with a 10K resistor and it did not work - it could be my circuit connection but i will give it another go sometime this week. take a look at my circuit drawing.
[image]

amitubale, 2017-06-23:
Someone please comment if the above diagram is accurate?

Thanks

davidelang, 2017-06-23:
that looks right

amitubale, 2017-06-23:
so, once i get the above circuit working , and before I set GPIO 3 to '9 SWITCH1' on the webpage, 
I need to go to Console and type "SerialLog Off" ?

cryptooth, 2017-06-28:
I want To do like this setup. This switch is latching switch or push button? And how it works? 

When changing the switch state (on or off) relay output change? If state changed from remotely (like home assistant) switch mode toggled? 

I do not know did i understand?

amitubale, 2017-06-28:
latching...the standard toggle switch on the wall...that usually runs the electrical load...when you introduce the sonoff...the electrical load is handled by sonoff relay...just run GPIO&Gnd wire to the wall switch...

cryptooth, 2017-06-28:
@amitubale perfect. Thank you for prompt answer.

dawsmac, 2017-08-14:
@amitubale did you get this working, in the end, I need to also have a unit in toggle mode, would be easier for me to go with the GPIO 3 option. so just wanted to confirm you got it working?

cryptooth, 2017-08-14:
@dawsmac @amitubale i did it with sonoff and standart ESP8266-01 module (tasmota installed)
One gpio for toggle push button, one gpio for toggle latching switch.

amitubale, 2017-08-14:
@dawsmac yes the above circuit diagram wored for me. I need to use the TX, RX as i Use GPIO 4 for a Sensor Temp Humidity and use TX or RX for the wall electrical toggle switch.

dawsmac, 2017-08-15:
Thanks for the input guys, so I have added the resistor and switching looks to be fine, but my Temp has stopped GPIO14 I have also switched off serial logging but did not think this would affect this?

dawsmac, 2017-08-21:
Sorry guys me again thought i had it all working for some reason even with GPIO3 setup with the toggle switch and switchmode set to switchmode1 1 i am getting the following on a closed circuit.

8:54:43 MQTT: tele/HW/STATE = {"Time":"2017-08-21T18:54:43", "Uptime":0, "Vcc":3.204, "POWER":"OFF", "Wifi":{"AP":1, "SSID":"D******_**", "RSSI":82, "APMac":"80:2A:A8:15:8F:82"}}
18:54:43 MQTT: tele/HW/SENSOR = {"Time":"2017-08-21T18:54:43", "Switch1":"OFF"}

not sure what i have done wrong also have the serial logging set to 0

C-Duv, 2017-10-21:
Looking for the same setup: Sonoff Basic with a DHT22 sensor and a switch (momentary or toggle) to power on/off the device connected to the Sonoff Basic.

I noticed a possible typo in [your schema](https://user-images.githubusercontent.com/28624275/27345435-b08cbf58-55b7-11e7-8748-f3bc87e864f1.png): according to [wiki](https://github.com/arendst/Sonoff-Tasmota/wiki/GPIO-Locations), **RX** pin is GPIO **03** and **TX** pin is GPIO **01** (your schema places the "RX" and "TX" labels in the right order but indicates RX=01 and TX=03).
Am I wrong about this being a typo?

So you configured Tasmota (in page *Modules parameters*) as follows:
* *GPIO3 Serial In* (or *GPIO1 Serial Out* if the typo isn't one) set to: *Switch1*
* *GPIO14 Sensor* set to: *DTH22*

What resistor did you ended up using? 4.7K?

vikasbhasin, 2018-03-30:
I want to use spinoff basic with wall (toggle) switch. No other sensors, just the switch. Can I use ESPurna firmware instead of Tasmota?

hulkhaugen, 2018-04-08:
I'm a bit new to this, but i wonder if i can have a DHT22 and a push-button that doesn't operate the sonoff relay, but something else? What i'm doing is to isolate the relay to from mains so it becomes a dry contact, and the switch will actually be a magnetic switch. I will use this to control my garage door, and i know that's possible. I'm thinking adding a DHT22 should also be possible, but can a switch between Rx and GND be used as a custom command and not trigger the sonoff relay itself?

pilot1981, 2020-02-03:
Hi,
I have similar problem, I go to explain:

I have sonoff basic R2 with cutted track to control a garage door and GPIO3 for close senso.

I had many problem on GPIO3 on different HW: problem to boot with GPIO3 shortcircuited with GND and/or no answer from GPIO3.

It's possible pull-up restore can solve my problems in this scenario? I can use defective sonoff?

XxAndreroxxX, 2020-06-19:
Hi,

I'm setting up Sonoff Basic R2 as a light switch, but with a capacitive touch sensor (TTP223) as a switch. 

When I connect it to the RX (GPIO1), when starting it also presents some problem and does not execute anything, however connecting to the TX (GPIO3) it works normally.