arendst/Tasmota #510: How to use Hue emulation?

spants, 2017-06-07:
I have a number of Hue lights, Alexa, Node Red, MQTT etc etc

I see that Tasmota supports Hue emulation. Does this mean that my H801 devices (with RGBW led strips) can behave as a Hub and attached coloured light?
Does it support RGB and separate white?
Does is support the 5050 led strips or is it the ws2812 type?....

Many thanks
Tony

altelch, 2017-06-07:
Hi,

yes the ESP emulates a Hue bridge but only the minimal subset required 
to work with Alexa/Echo-Devices. I only developed  and tested with an 
WS2812 RGB type LED strip. Never tried 5050 or RGBW as I have none to 
play with.

Greetings,

     Heiko.

spants, 2017-06-08:
Many thanks!. I would like to convert my H801 with Hue emulation using PWM for RGBW. I will take a look.

closed 2017-06-08 by spants

zomars, 2018-07-06:
Are there any instructions in the wiki?