arendst/Tasmota #511: Web password lost

SzilagyiR, 2017-06-08:
How can I reset my Sonoff Touch, if I lost my password?

Thanks!

arendst, 2017-06-08:
Enter command webpassword and enter a new password. 

Command can be entered serial or via mqtt server

SzilagyiR, 2017-06-08:
I tried only HTTP command, thank I will try!

davidelang, 2017-06-08:
hold down the button for 4 seconds will reset all configs to what was compiled 
in to user_config.h

if that doesn't work, and you can't connect via mqtt to reset the password, you 
will have to reflash the device.
k

SzilagyiR, 2017-06-08:
How can I send the command via MQTT server?
I ran Mosquito on Raspberry Pi.

SzilagyiR, 2017-06-08:
I make the reset procedure but after it not connet to the wifi. Now I reflash it but nothing happen, the led blinking and not connect to the wifi.
The AP settings correct in the  user_config.h.

davidelang, 2017-06-08:
see the troubleshooting page to make sure that the settings from user_config.h 
get applied.
k

SzilagyiR, 2017-06-09:
It says to change the value of CFG_HOLDER in user_config.h.
The default is 0x20161209, can I change it to anything?

SzilagyiR, 2017-06-09:
I chang the last number to 6, reflash it and it's work fine!

Thanks for your help!

davidelang, 2017-06-09:
yes, the only thing that matters is if it matches the prior value or not.

closed 2017-06-09 by SzilagyiR

cgatnet, 2018-07-10:
Helpful thread. -->Seems there was a typo in my password, so i failed to log into after securing the webfrontend. This thread lead me to the fix.
just pushed the button for 40s and the sonoff switched on/off, after that all my flashed settings were back. 
thanks for the great wiki and the helpful threads (and this great project, off course) 👍