arendst/Tasmota #512: Error while compiling

MCmakerPro, 2017-06-08:
I get this error when compiling:

exec: "C:\\Users\\user\\Desktop\\Esp flashing IDE\\portable\\packages\\esp8266\\tools\\xtensa-lx106-elf-gcc\\1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++": file does not exist
Error compiling for board Generic ESP8266 Module.

arendst, 2017-06-08:
Try to remove spaces in folder Esp flashing IDE

knotbeerdan, 2017-07-03:
I am getting the same error, but it does not include the full path my exact error is:

exec: "/bin/xtensa-lx106-elf-g++": file does not exist
Error compiling for board Generic ESP8266 Module.

I have followed the instructions found here and created a dedicated stand alone version of Arduino IDE 1.8.3

I do not have the same issue when I use the standard unmodified stand alone version of the same IDE version.

MCmakerPro, 2017-07-25:
Hello, I dont't remember how I managed to flash my sonoff basic, maybe through platformi, but now I have my sonoff touch and the space removal didnt help.

closed 2018-05-07 by stale[bot]