arendst/Tasmota #513: Sonoff Touch unreachable when ON (Tasmota 5.1.6)

LexPrima, 2017-06-08:
When I switch the light on either by touch, web UI or Alexa, it becomes unreachable. My Router says it's connected but I can't access the web UI nor can Alexa turn it off again. Touching works though and it becomes reachable on instant. 

Downgraded to 5.1.5 as temporary fix.

arendst, 2017-06-09:
I do not observe.

Pls provide output of ``status 0``

LexPrima, 2017-06-09:
Ok, updated to 5.1.6 again. Still reproducable for me.

13:45:33 RSLT: Receive topic /status, data size 1, data 0
13:45:33 RSLT: DataCb Group 0, Index 1, Type STATUS, Data 0 (0)
13:45:33 RSLT: STATUS = {"Status":{"Module":10, "FriendlyName":"Licht", "Topic":"sonoff", "ButtonTopic":"0", "Power":0, "PowerOnState":3, "LedState":0, "SaveData":1, "SaveState":1, "ButtonRetain":0, "PowerRetain":0}}
13:45:33 RSLT: STATUS1 = {"StatusPRM":{"Baudrate":115200, "GroupTopic":"sonoffs", "OtaUrl":"http://domus1:80/api/arduino/sonoff.ino.bin", "Uptime":0, "Sleep":0, "BootCount":27, "SaveCount":264}}
13:45:33 RSLT: STATUS2 = {"StatusFWR":{"Program":"5.1.6", "BuildDateTime":"2017-06-06T21:27:52", "Boot":4, "Core":"2_3_0", "SDK":"1.5.3(aec24ac9)"}}
13:45:33 RSLT: STATUS3 = {"StatusLOG":{"Seriallog":4, "Weblog":4, "Syslog":4, "LogHost":"domus1", "SSId1":"WLAN-GTV68Z", "SSId2":"indebuurt2", "TelePeriod":300}}
13:45:33 RSLT: STATUS4 = {"StatusMEM":{"ProgramSize":470, "Free":532, "Heap":27, "ProgramFlashSize":1024, "FlashSize":1024, "FlashMode":3}}
13:45:33 RSLT: STATUS5 = {"StatusNET":{"Host":"sonoff-0522", "IP":"192.168.2.105", "Gateway":"192.168.2.1", "Subnetmask":"255.255.255.0", "DNSServer":"192.168.2.1", "Mac":"60:01:94:80:82:0A", "Webserver":2, "WifiConfig":3}}
13:45:33 RSLT: STATUS7 = {"StatusTIM":{"UTC":"Fri Jun 09 12:45:33 2017", "Local":"Fri Jun 09 13:45:33 2017", "StartDST":"Sun Mar 26 02:00:00 2017", "EndDST":"Sun Oct 29 03:00:00 2017", "Timezone":1}}
13:45:33 RSLT: STATUS10 = {"StatusSNS":{"Time":"2017-06-09T13:45:33"}}
13:45:33 RSLT: STATUS11 = {"StatusSTS":{"Time":"2017-06-09T13:45:33", "Uptime":0, "Vcc":3.211, "POWER":"OFF", "Wifi":{"AP":1, "SSID":"WLAN-GTV68Z", "RSSI":44, "APMac":"D0:65:CA:12:D6:B5"}}}

arendst, 2017-06-09:
Still do not observe...

- Installed **sonoff.ino** from release 5.1.6
- Set all logging to 4 as you did
- Disabled MQTT
- Enabled WeMo or Hue Bridge

As I said. No problem.

Suggestions:
- Disable extranous logging. If no serial connection available set seriallog to 0. If no syslog server available set syslog to 0 else 2. Set weblog to 2.
- Set timezone to 99

Try debugging:
- If you do not have a syslog server disable syslog by setting it to 0
- If you have a serial connection set seriallog to 4 else to 0.
- If no serial connection then open a webpage to your device, open console and set weblog to 4.
- if no serial connection open another webpage to device and toggle the switch.

What happens now on the console?

LexPrima, 2017-06-09:
No change. I tried to ping it but it timeouts until I physically touch to switch it off.

15:37:41 RSLT: Receive topic /status, data size 1, data 0
15:37:41 RSLT: DataCb Group 0, Index 1, Type STATUS, Data 0 (0)
15:37:41 RSLT: STATUS = {"Status":{"Module":10, "FriendlyName":"Licht", "Topic":"sonoff", "ButtonTopic":"0", "Power":0, "PowerOnState":3, "LedState":0, "SaveData":1, "SaveState":1, "ButtonRetain":0, "PowerRetain":0}}
15:37:41 RSLT: STATUS1 = {"StatusPRM":{"Baudrate":115200, "GroupTopic":"sonoffs", "OtaUrl":"http://domus1:80/api/arduino/sonoff.ino.bin", "Uptime":0, "Sleep":0, "BootCount":29, "SaveCount":278}}
15:37:41 RSLT: STATUS2 = {"StatusFWR":{"Program":"5.1.6", "BuildDateTime":"2017-06-06T21:27:52", "Boot":4, "Core":"2_3_0", "SDK":"1.5.3(aec24ac9)"}}
15:37:41 RSLT: STATUS3 = {"StatusLOG":{"Seriallog":0, "Weblog":4, "Syslog":0, "LogHost":"domus1", "SSId1":"WLAN-GTV68Z", "SSId2":"indebuurt2", "TelePeriod":300}}
15:37:41 RSLT: STATUS4 = {"StatusMEM":{"ProgramSize":470, "Free":532, "Heap":28, "ProgramFlashSize":1024, "FlashSize":1024, "FlashMode":3}}
15:37:41 RSLT: STATUS5 = {"StatusNET":{"Host":"sonoff-0522", "IP":"192.168.2.105", "Gateway":"192.168.2.1", "Subnetmask":"255.255.255.0", "DNSServer":"192.168.2.1", "Mac":"60:01:94:80:82:0A", "Webserver":2, "WifiConfig":3}}
15:37:41 RSLT: STATUS7 = {"StatusTIM":{"UTC":"Fri Jun 09 13:37:41 2017", "Local":"Fri Jun 09 15:37:41 2017", "StartDST":"Sun Mar 26 02:00:00 2017", "EndDST":"Sun Oct 29 03:00:00 2017", "Timezone":99}}
15:37:41 RSLT: STATUS10 = {"StatusSNS":{"Time":"2017-06-09T15:37:41"}}
15:37:41 RSLT: STATUS11 = {"StatusSTS":{"Time":"2017-06-09T15:37:41", "Uptime":0, "Vcc":3.209, "POWER":"OFF", "Wifi":{"AP":1, "SSID":"WLAN-GTV68Z", "RSSI":40, "APMac":"D0:65:CA:12:D6:B5"}}}

arendst, 2017-06-09:
Using the procedure above, whats on the console when you switch it on and off?

LexPrima, 2017-06-09:
15:47:54 HTTP: Handle root
15:47:54 HTTP: Handle console
15:48:00 HTTP: Handle WeMo basic event
15:48:00 RSLT: RESULT = {"POWER":"ON"}
15:48:00 RSLT: POWER = ON
15:48:01 Config: Saved configuration (1492 bytes) to flash at FB and count 289
15:48:02 APP: Multipress 1
15:48:03 RSLT: RESULT = {"POWER":"OFF"}
15:48:03 RSLT: POWER = OFF
15:48:03 Wifi: Checking connection...
15:48:03 Wifi: Connected
15:48:04 Config: Saved configuration (1492 bytes) to flash at FB and count 290

arendst, 2017-06-09:
While I can test WeMo only via Node-Red (as lack of Alexa) it still seems to work as expected:
```
15:59:29 UPnP: Wemo Response sent to 192.168.2.61:52444
15:59:35 Wifi: Checking connection...
15:59:35 Wifi: Connected
15:59:38 HTTP: Handle WeMo basic event
15:59:38 RSLT: RESULT = {"POWER":"ON"}
15:59:38 RSLT: POWER = ON
15:59:47 HTTP: Handle WeMo basic event
15:59:47 RSLT: RESULT = {"POWER":"OFF"}
15:59:47 RSLT: POWER = OFF
15:59:48 Config: Saved configuration (1492 bytes) to flash at FB and count 87
15:59:55 Wifi: Checking connection...
15:59:55 Wifi: Connected
```
If turned on I still can see the web page and am able to ping.

Other than a weak wifi connection (low RSSI) I can't explain what's going on. You might want to download the **sonoff.bin** file again from the release page and try to see if that makes sense.

LexPrima, 2017-06-09:
Already did that since I downgraded to 5.1.5 yesterday which works flawlessly. I'll just stay on 5.1.5 then.

chuxxsss, 2017-06-12:
Check the modem as I had problems with that when I started with Node-red. No idea why, but after replacing modem worked great.

LexPrima, 2017-06-16:
Problem gone with 5.1.7

closed 2017-06-16 by LexPrima