arendst/Tasmota #514: Bricked device?

kalety, 2017-06-09:
Upgrading via OTA (5.1.6) one of my sonoffs, the screen says 'this device resseting (or something) in few seconds', but were several minutes and then I closed the web page... I'd try to reflash it but I can´t put it in flashing mode, don't respond..

Bricked device? (like title ;))

kruebsam, 2017-06-13:
Have you tried disconnecting it from the mains and reconnecting it? I noticed similar behaviour ince in a while after upgrading. However, flashing it the hard way (with direct connection to the board) should bring it back to life.

kalety, 2017-06-13:
Thanks kruebsam for your help!, I tried everithing, the problem is that the led no flashing nor blinking, anything... Whe you say "hard way", you want say the normal flash through FTDI?

davidelang, 2017-06-13:
Yes, unless the chip itself is fried, you can flash it via the serial port. the 
'ground gpio0 and flash' is not something that can be changed by the software 
you load on it.

jaysuk, 2017-06-13:
Or try resetting the board using the button. See the wiki for the correct
amount of presses.

guillier, 2017-06-14:
Something similar happened to me and I thought I would have to replace the chip but didn't have to in the end.

With https://github.com/espressif/esptool can you run successfully?

esptool.py --chip esp8266 --port /dev/ttyUSB0 flash_id
and/or
esptool.py --chip esp8266 --port /dev/ttyUSB0 erase_flash

You'll probably have to press (and keep pressed) the button before applying the FTDI power, took me a few tries before it worked. Then I reprogrammed the chip using platformio (again pressing the button at boot).

Good luck!

kalety, 2017-06-14:
Thanks, I've tríed everyhing, and no success... Only 5€ / $ I've loss...

jetema, 2017-08-02:
The last chance is to replace the flash memory ( U6 on the picture ), it is not quite difficult, look this video 
https://www.youtube.com/watch?v=8mz5sCAvDAY&t=616s
[image]

davidelang, 2017-08-03:
if the esp8266 is still functioning, you should be able to get it into 
programming mode and flash it again. If it's not, changing the flash won't help 
you.

if it's retaining 'bad' settings, then you need to change CFGHOLDER or flash 
with all zeros.

These are pretty simple devices, and the flashing mechanism is built in to the 
chip, not something that can be changed in software. So the only way to 'brick' 
it is to damage the chip (say by applying 5v to it)
k

MesserMike, 2018-01-23:
i also think i have the device bricked, BUT I coincidentally found a loose solder joint of an smd component...

The primary or life side was ok, but on the secondary side (3,3 volt side) was a loose solder joint... 
iCheck the lines with an messuremeter and find it.
i use 250 degre on my solderingstation and put the solder short on every single solderside of the items...
after this work the pow works pobably again... :)

closed 2018-06-07 by stale[bot]