arendst/Tasmota #516: Counter debounce

off14, 2017-06-10:
Hi, if i configure debounce for counter let say 50 or 100,
it works ok, when my gas meter is fast enough. but when the consumption is super low, magnet can close the contact for half second and even for one second (slow rolling), or even stop at "closed" state. in this case every such event is treated AS TWO events - first when the proximity sensor closes, and the second when it opens.
making debounce too high is not an option - it can miss two events in this case.

can you build a counter option to treat only CLOSING as event (not both, closing and opening).

thanks

arendst, 2017-06-10:
You'll have to find some kind of debounce on your magnet as I'm already using only the falling edge of the signal as interrupt. I have no means to select another option.

off14, 2017-06-17:
Thanks for the explanation,
anyway - the counter right now increases on both changes - from DOWN to UP and from UP to DOWN.
why not to make this option selectable? e.g. UP only, to calculate only changes from down to UP,

Stalkk, 2017-06-18:
I can confirm the issue. 
I try to use TCRT5000 - module like this https://www.aliexpress.com/item/TCRT5000-Tracking-Module-Tracker-Infrared-Reflective-Sensor-For-Arduino-Robot-Smart-Car/32408909714.html and my counter counts always twice.

off14, 2017-06-20:
@arendst any chance you can look at this option?

arendst, 2017-06-20:
@off14 try to fiddle around with the counter interrupt routine in file ``xsns_counter.ino`` at line 81:
```
attachInterrupt(pin[GPIO_CNTR1 +i], counter_callbacks[i], FALLING);
```

Change FALLING to another nice Arduino supported value and experiment.

Make sure your counter input is ESP8266 compatible (3V3 NOT 5V).

off14, 2017-06-21:
my counter is a reed switch between GPIO14 and GND (magnetic)

hall sensor is another option - but will experiment later, not works perfectly with 3.3V

Stalkk, 2017-06-21:
I think i have fixed my issue with a hardware capacitor debouncing.

off14, 2017-06-23:
do i need any resistor (pullup or pull down) if i use standard GPIO14 ? and connect it to ground via REED switch
PS i've seen pullup necessary for re-using tx/rx pins for toggle switch

neographikal, 2018-02-01:
I have a similar problem with a sensor. On my water meter I measure the reflection. If the water meter stops at the wrong position the sensor goes mad: it keeps pulsing like mad. Now I would like to ignore the pulses if they are not X msec apart. With a classic debounce option set to 1000ms I will get a incorrectly measured pulse every second.

bubblefish12, 2018-03-30:
@neographikal 
I had this problem too with 5V

now with 3.3V no Problem

gabbermorph, 2018-10-17:
I have still the same issue on Slow rotations on my Gasmeter with the reed contact.
On Slow Rotations it counts twice, even with debounce on 3200. 

And what means Counter1n?

Its actually running on a USB adapter 550mA through a FTDI on 3.3v.

PR #8021 (2020-03-31): Enhanced counter debouncing
New commands CounterDebounceLow and CounterDebounceHigh to allow individual debounce times for low and high pulse widths to discard non valid falling edges. These are checked before legacy CounterDebounce checks distance between to valid falling edges.
Useful to have robust counter results when using e.g. an TCRT5000 optical sensor to count electromechanical "Ferraris" electricity meters.

## Description:

**Related issue (if applicable):** fixes #7488, fixes #516

## Checklist:
  - [x] The pull request is done against the latest dev branch
  - [x] Only relevant fil [... cut]