arendst/Tasmota #517: Support for bi-colour LEDs (Orvibo B25)

bruce34, 2017-06-10:
The Orvibo B25 ([banggood.com](https://www.banggood.com/ORVIBO-HomeMate-B25U-Wi-Fi-Smart-Remote-Control-Timing-Socket-USEU-Plug-p-953743.html)) can be used with a config from [here](https://blog.slange.co.uk/orvibo-b25-wifi-power-socket/) to give relay functionality.

The LED is a 2-pin red/blue LED with each pin connected to a GPIO. Therefore, 1 GPIO is driven high while the other is driven low to make it light. i.e. it has 3 states (colour1, colour2, OFF) and uses 2 GPIOs. The direction determines the colour. Is this something already in the code or, if not, is there a way it should be integrated? Given guidance on how it would be best integrated, I could create an initial PR if helpful.

davidelang, 2017-06-10:
There is currently no support for this sort of thing.

take a look in the sonoff.ino file, probably for the code around ledstate for 
how leds are currently handled.

arendst, 2017-06-10:
- Is there any sense in having control over two colors?
- What are the current colors used for?
- I suppose tasmota is at least able to signal one color at the moment? If not then I could make that work.

ins4neuk, 2017-07-21:
i'm trying to suss this out as well :(

closed 2018-05-07 by stale[bot]