arendst/Tasmota #518: mDNS not discovering Raspberry PI

Arduingo, 2017-06-11:
How the SONOFF firmware discover the MQTT broker in the network, by the broker broadcast his service or by host name?. 

In my case configuring my raspberry PI to broadcast the MQTT service availability works but configuring the .local name in the MQTT configuration instead of the IP not. My RaspPi is discoverable by hostname by any other computer in the same network.

davidelang, 2017-06-11:
are you using the short name of the fqdn name in the firmware when it doesn't 
work?

are you manually setting your IP (including DNS server) or is this set via DHCP?

Arduingo, 2017-06-11:
Hi, I´m using DHCP, and i´m entering the hostname hkpi.local in the firmware before compilation and also tried into the web configuration.

Here some logs:

00:00:04 DHT: Received 26, 00, 17, 00, 3D =? 3D
00:00:04 RTC: (UTC) Sun Jun 11 16:53:26 2017
00:00:04 RTC: (DST) Sun Mar 26 02:00:00 2017
00:00:04 RTC: (STD) Sun Oct 29 03:00:00 2017
17:53:27 MQTT: Attempting connection...
17:53:28 mDNS: Query done with 1 mqtt services found
17:53:28 mDNS: Service found on hkpi ip 10.0.1.5 port 1883
17:53:28 MQTT: Connected
17:53:28 MQTT: tele/sonoff/LWT = Online (retained)
17:53:28 MQTT: cmnd/sonoff/POWER = 
17:53:28 MQTT: tele/sonoff/INFO1 = {"Module":"Sonoff Basic", "Version":"5.1.4", "FallbackTopic":"DVES_96E42A", "GroupTopic":"sonoffs"}
17:53:29 MQTT: tele/sonoff/INFO2 = {"WebserverMode":"Admin", "Hostname":"sonoff-1066", "IPaddress":"10.0.1.4"}
17:53:29 MQTT: tele/sonoff/INFO3 = {"Started":"Software/System restart"}
17:53:29 DHT: Received 26, 00, 17, 00, 3D =? 3D
17:53:29 Config: Saved configuration (1492 bytes) to flash at FB and count 818

This log is with the mosquito server announcing its available by creating and avahi service configuration (I looked into the firmware code to get the clue), but If I don't do that the connection fail and the only way to let the SONOFF connect with the broker is by entering the IP address.

Ghostbird, 2017-08-12:
You'll have to add the avahi service on the raspberry pi (mosquitto installer does not auto-configure it).
I assume you've got avahi installed on the raspberry pi (you have if you can browse to hkpi.local)
Then read this: https://github.com/arendst/Sonoff-MQTT-OTA-Arduino/pull/115
In short, you'll have to do:
```
sudo nano /etc/avahi/services/mqtt.service
```
And write this in the file:
```
<?xml version="1.0" standalone='no'?>
<!DOCTYPE service-group SYSTEM "avahi-service.dtd">
<service-group>
 <name replace-wildcards="yes">MQTT on %h</name>
  <service>
   <type>_mqtt._tcp</type>
   <port>1883</port>
  </service>
</service-group>
```
Then run
```
sudo service avahi-daemon reload
```
By doing this you manually tell Avahi to advertise an MQTT service on port 1883 through mDNS.

Arduingo, 2017-08-14:
Thanks! Ghostbird, your suggestion works great

closed 2017-08-14 by Arduingo

JoePala, 2017-12-23:
Thank to Ghostbird - just found this solution and it worked great for me, too!