arendst/Tasmota #520: how to publish mqtt message when using a sound sensor as additional switch?

dirkclemens, 2017-06-13:
All, 
I am using the basic Sonoff with a sound sensor module attached to GPIO 14 with the intention to switch the Sonoff by clapping hands or any other noise. Works great so far, but what internal function should I use best to not only switch the relay but also send the "claps" via mqtt?
By now I am using setRelay() only. 
Is it better to use  send_button_power() or do_cmnd_power()? 
What parameter should I hand over to just switch on/off the first (single) relay? 

Thanks

```
 int soundState = digitalRead(soundPin); // GPIO 14
  if (!soundState){
    clapCount++; 
    duration = millis();
    delay(200); 
  }
  if (clapCount == 4){
    setRelay(LOW);
    clapCount = clapCount % 2; 
  }
  if (clapCount == 2){
    setRelay(HIGH);
  }
  if (millis()-duration > INTERVAL ){
    clapCount = 0; 
    duration = 0;
  }
```

davidelang, 2017-06-13:
On Mon, 12 Jun 2017, dirkc wrote:

> All,
> I am using the basic Sonoff with a sound sensor module attached to GPIO 14 with the intention to switch the Sonoff by clapping hands or any other noise. Works great so far, but what internal [...]

well, you don't need to change the source, just setting buttontopic and it would 
send mqtt messages when the module signals.

if you are adding additional state (as implied below), then you are on your own, 
what you are doing doesn't fit the existing model.
k

dirkclemens, 2017-06-13:
Thanks, but as I understand, with just setting the buttontopic there is no possibility to add some kind of debouncing, isn't it? Unfortunately the sound module does not send clear and unique on/off signals, so debouncing and double checking is necessary to prevent from continuously switching on & off.

davidelang, 2017-06-13:
In that case, you need to add a new sensor type and add all your logic in there.

PeggyFree, 2017-06-20:
dirkclemens can you give me a link to you sound sensor? I'm interested in getting one.

dirkclemens, 2017-06-23:
@PeggyFree search for "KY-038"

closed 2018-05-07 by stale[bot]

sheminasalam, 2020-06-16:
> @PeggyFree search for "KY-038"

for kY-038 which pinout did you use i,e digital out(DO) or analog (AO)? what type of sensor did you select in the tasmota consule configuration. I am having problem making this work.

bboudewijns, 2020-07-31:
> @PeggyFree search for "KY-038"

Hi there! Thanks a lot for the info! I was just looking for that. How did you integrate it within tasmota? As it doesn't show up on the supported sensor list? Thanks in advance.